openwrt/packages #23002: We are unable to use friendly name, manufacture name, etc. in miniupnpd 2.0.0.

Nandu2324, 2023-12-28:
miniupnpd to 2.0.0 version, 
Openwrt version :19.07

We are unable to use friendly name, manufacture name, etc. in miniupnpd.
In the config.h file, I have uncommented the #define ENABLE_MANUFACTURER_INFO_CONFIGURATION.
----------------------------------------------------------
During compilation, we received the error below.
----------------------------------------------------------
natpmp.o pcpserver.o upnpglobalvars.o upnpevents.o upnputils.o getconnstatus.o upnpstun.o upnppinhole.o pcplearndscp.o asyncsendto.o getifstats.o ifacewatcher.o getroute.o iptcrdr.o iptpinhole.o nfct_get.o  -lip4tc -lip6tc -lssl -lcrypto -ldl -luuid  -o miniupnpd
mips-openwrt-linux-musl-gcc -L/home/narendra/1907OWRT/openwrt/staging_dir/target-mips_24kc_musl/usr/lib -L/home/narendra/1907OWRT/openwrt/staging_dir/target-mips_24kc_musl/lib -L/home/narendra/1907OWRT/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/usr/lib -L/home/narendra/1907OWRT/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/lib -znow -zrelro -Wl,--gc-sections,--as-needed  -L/home/narendra/1907OWRT/openwrt/staging_dir/target-mips_24kc_musl/usr/lib  testupnpdescgen.o upnpdescgen.o  -lip4tc -lip6tc -lssl -lcrypto -ldl -luuid  -o testupnpdescgen
/home/narendra/1907OWRT/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/lib/gcc/mips-openwrt-linux-musl/7.5.0/../../../../mips-openwrt-linux-musl/bin/ld: /home/narendra/1907OWRT/openwrt/tmp/ccZ9qp8y.ltrans0.ltrans.o:(.data.rel.ro+0x34): undefined reference to `friendly_name'
/home/narendra/1907OWRT/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/lib/gcc/mips-openwrt-linux-musl/7.5.0/../../../../mips-openwrt-linux-musl/bin/ld: /home/narendra/1907OWRT/openwrt/tmp/ccZ9qp8y.ltrans0.ltrans.o:(.data.rel.ro+0x3c): undefined reference to `manufacturer_name'
/home/narendra/1907OWRT/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/lib/gcc/mips-openwrt-linux-musl/7.5.0/../../../../mips-openwrt-linux-musl/bin/ld: /home/narendra/1907OWRT/openwrt/tmp/ccZ9qp8y.ltrans0.ltrans.o:(.data.rel.ro+0x44): undefined reference to `manufacturer_url'
/home/narendra/1907OWRT/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/lib/gcc/mips-openwrt-linux-musl/7.5.0/../../../../mips-openwrt-linux-musl/bin/ld: /home/narendra/1907OWRT/openwrt/tmp/ccZ9qp8y.ltrans0.ltrans.o:(.data.rel.ro+0x4c): undefined reference to `model_description'
/home/narendra/1907OWRT/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/lib/gcc/mips-openwrt-linux-musl/7.5.0/../../../../mips-openwrt-linux-musl/bin/ld: /home/narendra/1907OWRT/openwrt/tmp/ccZ9qp8y.ltrans0.ltrans.o:(.data.rel.ro+0x54): undefined reference to `model_name'
/home/narendra/1907OWRT/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/lib/gcc/mips-openwrt-linux-musl/7.5.0/../../../../mips-openwrt-linux-musl/bin/ld: /home/narendra/1907OWRT/openwrt/tmp/ccZ9qp8y.ltrans0.ltrans.o:(.data.rel.ro+0x64): undefined reference to `model_url'
collect2: error: ld returned 1 exit status
make[4]: *** [<builtin>: testupnpdescgen] Error 1
make[4]: Leaving directory '/home/narendra/1907OWRT/openwrt/build_dir/target-mips_24kc_musl/miniupnpd-2.2.0'
make[3]: *** [Makefile:77: /home/narendra/1907OWRT/openwrt/build_dir/target-mips_24kc_musl/miniupnpd-2.2.0/.built] Error 2
make[3]: Leaving directory '/home/narendra/1907OWRT/openwrt/feeds/packages/net/miniupnpd'
time: package/feeds/packages/miniupnpd/compile#3.67#1.47#5.05
make[2]: *** [package/Makefile:113: package/feeds/packages/miniupnpd/compile] Error 2
make[2]: Leaving directory '/home/narendra/1907OWRT/openwrt'
make[1]: *** [package/Makefile:107: /home/narendra/1907OWRT/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/narendra/1907OWRT/openwrt'
make: *** [/home/narendra/1907OWRT/openwrt/include/toplevel.mk:227: world] Error 2
[miniupnpd-2.2.0.zip](https://github.com/openwrt/packages/files/13786418/miniupnpd-2.2.0.zip)

brada4, 2023-12-28:
you need to provide param€ter to ./configure, likely after distclean

Nandu2324, 2023-12-29:
What type of parameter i have to add in ./configure
-----------------------------------------------------------------------------------------------------------------------------
narendra@CI5W10P121407:~/1907OWRT/openwrt$ make V=s package/miniupnpd/distclean
make[1]: Entering directory '/home/narendra/1907OWRT/openwrt'
make[1]: *** No rule to make target 'package/miniupnpd/distclean'.  Stop.
make[1]: Leaving directory '/home/narendra/1907OWRT/openwrt'
make: *** [/home/narendra/1907OWRT/openwrt/include/toplevel.mk:227: package/miniupnpd/distclean] Error 2
-------------------------------------------------------------------------------------------------------------------------------
Getting the same error: after cleaning ("make V=s package/miniupnpd/distclean") and building with "make V=sc package/miniupnpd/compile"
-------------------------------------------------------------------------------------------------------------------------------
make[2]: Entering directory '/home/narendra/1907OWRT/openwrt/feeds/packages/net/miniupnpd'
mkdir -p /home/narendra/1907OWRT/openwrt/dl
SHELL= flock /home/narendra/1907OWRT/openwrt/tmp/.miniupnpd-2.2.0.tar.gz.flock -c '     /home/narendra/1907OWRT/openwrt/scripts/download.pl "/home/narendra/1907OWRT/openwrt/dl" "miniupnpd-2.2.0.tar.gz" "f105210a13ed0ebfc649f661ecc59e0a072cc547b04977851f22b5521b4cadff" "" "https://miniupnp.tuxfamily.org/files"    '
rm -f /home/narendra/1907OWRT/openwrt/build_dir/target-mips_24kc_musl/miniupnpd-2.2.0/.built
touch /home/narendra/1907OWRT/openwrt/build_dir/target-mips_24kc_musl/miniupnpd-2.2.0/.built_check
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/narendra/1907OWRT/openwrt/build_dir/target-mips_24kc_musl/miniupnpd-2.2.0:miniupnpd-2.2.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic -flto  -I/home/narendra/1907OWRT/openwrt/staging_dir/target-mips_24kc_musl/usr/include -I/home/narendra/1907OWRT/openwrt/staging_dir/target-mips_24kc_musl/include -I/home/narendra/1907OWRT/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/usr/include -I/home/narendra/1907OWRT/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include/fortify -I/home/narendra/1907OWRT/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/narendra/1907OWRT/openwrt/build_dir/target-mips_24kc_musl/miniupnpd-2.2.0:miniupnpd-2.2.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic -flto  -I/home/narendra/1907OWRT/openwrt/staging_dir/target-mips_24kc_musl/usr/include -I/home/narendra/1907OWRT/openwrt/staging_dir/target-mips_24kc_musl/include -I/home/narendra/1907OWRT/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/usr/include -I/home/narendra/1907OWRT/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include/fortify -I/home/narendra/1907OWRT/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include " LDFLAGS="-L/home/narendra/1907OWRT/openwrt/staging_dir/target-mips_24kc_musl/usr/lib -L/home/narendra/1907OWRT/openwrt/staging_dir/target-mips_24kc_musl/lib -L/home/narendra/1907OWRT/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/usr/lib -L/home/narendra/1907OWRT/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/lib -znow -zrelro -Wl,--gc-sections,--as-needed " make  -C /home/narendra/1907OWRT/openwrt/build_dir/target-mips_24kc_musl/miniupnpd-2.2.0/. AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/narendra/1907OWRT/openwrt/build_dir/target-mips_24kc_musl/miniupnpd-2.2.0:miniupnpd-2.2.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic -flto" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CROSS="mips-openwrt-linux-musl-" ARCH="mips" ;
make[3]: Entering directory '/home/narendra/1907OWRT/openwrt/build_dir/target-mips_24kc_musl/miniupnpd-2.2.0'
mips-openwrt-linux-musl-gcc -L/home/narendra/1907OWRT/openwrt/staging_dir/target-mips_24kc_musl/usr/lib -L/home/narendra/1907OWRT/openwrt/staging_dir/target-mips_24kc_musl/lib -L/home/narendra/1907OWRT/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/usr/lib -L/home/narendra/1907OWRT/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/lib -znow -zrelro -Wl,--gc-sections,--as-needed  -L/home/narendra/1907OWRT/openwrt/staging_dir/target-mips_24kc_musl/usr/lib  testupnpdescgen.o upnpdescgen.o  -lip4tc -lip6tc -lssl -lcrypto -ldl -luuid  -o testupnpdescgen
/home/narendra/1907OWRT/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/lib/gcc/mips-openwrt-linux-musl/7.5.0/../../../../mips-openwrt-linux-musl/bin/ld: /home/narendra/1907OWRT/openwrt/tmp/ccgSLh [... cut ...]
make[2]: *** [Makefile:77: /home/narendra/1907OWRT/openwrt/build_dir/target-mips_24kc_musl/miniupnpd-2.2.0/.built] Error 2
make[2]: Leaving directory '/home/narendra/1907OWRT/openwrt/feeds/packages/net/miniupnpd'
time: package/feeds/packages/miniupnpd/compile#0.35#0.21#0.63
make[1]: *** [package/Makefile:113: package/feeds/packages/miniupnpd/compile] Error 2
make[1]: Leaving directory '/home/narendra/1907OWRT/openwrt'
make: *** [/home/narendra/1907OWRT/openwrt/include/toplevel.mk:227: package/miniupnpd/compile] Error 2

BKPepe, 2023-12-29:
Can you use some formatting here? Otherwise, it is going to be mess. 

You are trying to compile OpenWrt 19.07, thus closing. Stop using the outdated and out of life branch.

closed 2023-12-29 by BKPepe