openwrt/packages #23069: openwrt/include/toplevel.mk:213: *** mixed implicit and normal rules: deprecated syntax

snakwu, 2024-01-06:
openwrt/include/toplevel.mk:213: *** mixed implicit and normal rules: deprecated syntax
[image]

snakwu, 2024-01-06:
[image]

closed 2024-01-06 by BKPepe

lengfwong, 2024-01-06:
Me too. This is different:

```
@@ -50,6 +50,7 @@
 path:=$(subst :,$(space),$(PATH))
 path:=$(filter-out .%,$(path))
 path:=$(subst $(space),:,$(path))
+export ORIG_PATH:=$(if $(ORIG_PATH),$(ORIG_PATH),$(PATH))
 export PATH:=$(path)
 export STAGING_DIR_HOST:=$(if $(STAGING_DIR),$(abspath $(STAGING_DIR)/../host),$(TOPDIR)/staging_dir/host)

@@ -209,9 +210,7 @@
        @+$(NO_TRACE_MAKE) -r -s $@

 check: .config FORCE
-       @+$(NO_TRACE_MAKE) -r -s $@ QUIET= V=s
-
-val.%: FORCE
+check val.% var.%: FORCE
        @+$(NO_TRACE_MAKE) -r -s $@ QUIET= V=s

 WARN_PARALLEL_ERROR = $(if $(BUILD_LOG),,$(and $(filter -j,$(MAKEFLAGS)),$(findstring s,$(OPENWRT_VERBOSE))))
```