openwrt/packages #23074: lib4l - v4l2-ctl segmentation fault when setting/getting any controls Openwrt 23.05.2

t3therdev, 2024-01-07:
lib4l - v4l2-ctl tool is version 1.22.1 on Openwrt 23.05.2 from the official package repository.

Maintainer: @thess @neheb  
Environment: OpenWrt 23.05.2  on openwrt-23.05.2-bcm27xx-bcm2710-rpi-3

Description:

On the lastest release of v4l and v4l2-ctl there is a segmentation fault when trying to get or set any v4l2 controls. 

the commands:
v4l2-ctl --list-devices (Properly lists all devicse))
v4l2-ctl -d /dev/video0 --list-ctrls (Properly lists all controls)

The command to get or set any control (used to work on previous Openwrt version, now does now) example:

Ouptut of control:
`User Controls

                     brightness 0x00980900 (int)    : min=0 max=240 step=1 default=115 value=115
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=115 value=115
                     saturation 0x00980902 (int)    : min=0 max=255 step=1 default=106 value=106
        white_balance_automatic 0x0098090c (bool)   : default=1 value=1
                           gain 0x00980913 (int)    : min=0 max=100 step=1 default=5 value=5
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1 value=1 (50 Hz)
      white_balance_temperature 0x0098091a (int)    : min=2600 max=6500 step=1 default=4650 value=4300 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=255 step=1 default=10 value=10

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=3 (Aperture Priority Mode)
         exposure_time_absolute 0x009a0902 (int)    : min=5 max=2500 step=1 default=100 value=100 flags=inactive
                   pan_absolute 0x009a0908 (int)    : min=-36000 max=36000 step=3600 default=0 value=0
                  tilt_absolute 0x009a0909 (int)    : min=-36000 max=36000 step=3600 default=0 value=0
                 focus_absolute 0x009a090a (int)    : min=0 max=255 step=1 default=0 value=0 flags=inactive
     focus_automatic_continuous 0x009a090c (bool)   : default=1 value=1
                  zoom_absolute 0x009a090d (int)    : min=100 max=190 step=1 default=100 value=100
`
Then try to set a control:

`v4l2-ctl -d /dev/video0 --set-ctrl=gain=12
Segmentation fault
`

Or using v4l2-ctl -C /dev/video0 --set-ctrl=gain=12
any command has a segmentation fault.

Suggest updating v4l to newer versions such as 1.24 or 1.26 (https://www.linuxtv.org/downloads/v4l-utils/)

brada4, 2024-01-07:
- Could you show sigsegv logged in `dmesg` (file name and offsets)
- `v4l2-ctl -D -d /dev/video0` (if it is uvc-anything etc)
- add at least one `-v` parameter 
-  if setting same value as existing crashes too

If you built updated package why not PR it?

t3therdev, 2024-01-07:
Hi @brada4 , we have not been able to compile a newer version yet, and were looking for assistance from the maintainers.

This is the output of one of the cameras, but they ALL segfault. This works on previous versions of openwrt and rasbian. 

`Driver Info:
        Driver name      : uvcvideo
        Card type        : VCS LL130 LL2604: VCS LL130 LL2
        Bus info         : usb-3f980000.usb-1.3
        Driver version   : 5.15.137
        Capabilities     : 0x84a00001
                Video Capture
                Metadata Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x04200001
                Video Capture
                Streaming
                Extended Pix Format
Media Driver Info:
        Driver name      : uvcvideo
        Model            : VCS LL130 LL2604: VCS LL130 LL2
        Serial           : LL2604
        Bus info         : usb-3f980000.usb-1.3
        Media version    : 5.15.137
        Hardware revision: 0x00000000 (0)
        Driver version   : 5.15.137
Interface Info:
        ID               : 0x03000002
        Type             : V4L Video
Entity Info:
        ID               : 0x00000001 (1)
        Name             : VCS LL130 LL2604: VCS LL130 LL2
        Function         : V4L2 I/O
        Flags            : default
        Pad 0x01000007   : 0: Sink
          Link 0x02000010: from remote pad 0x100000a of entity 'Extension 3' (Video Pixel Formatter): Data, Enabled, Immutable
`

Any get control also segfaults:

v4l2-ctl --device=/dev/video0 --get-ctrl=gain
Segmentation fault

Pretty sure this older version of v4l2-ctl / v4l needs to be updated. 

Log levels set to debug but I don't see anything in dmesg.

brada4, 2024-01-07:
Please format code with 3 backticks \`\`\` before and after.

brada4, 2024-01-07:
You should have space for gdb?
```
# gdb v4l2-ctl
gdb> r -v --device=/dev/video0 --get-ctrl=gain
CRASH !!!!
gdb> bt
gdb> info proc mappings
gdb> q
```

t3therdev, 2024-01-07:
@brada4 

```
Starting program: /usr/bin/v4l2-ctl -d /dev/video0 --set-ctrl=gain=12
warning: Unable to determine the number of hardware watchpoints available.
warning: Unable to determine the number of hardware breakpoints available.
warning: Unable to find dynamic linker breakpoint function.
GDB will be unable to debug shared library initializers
and track explicitly loaded dynamic code.

Program received signal SIGSEGV, Segmentation fault.
0x0000007ff7f77cb0 in ?? ()
(gdb) bt
#0  0x0000007ff7f77cb0 in ?? ()
#1  0x0000007ff7ffe000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
(gdb) info proc mappings
process 19421
Mapped address spaces:

          Start Addr           End Addr       Size     Offset  Perms  objfile
            0x400000           0x447000    0x47000        0x0  r-xp   /usr/bin/v4l2-ctl
            0x45f000           0x460000     0x1000    0x4f000  r--p   /usr/bin/v4l2-ctl
            0x460000           0x463000     0x3000    0x50000  rw-p   /usr/bin/v4l2-ctl
            0x463000           0x46f000     0xc000        0x0  rw-p   [heap]
            0x46f000           0x470000     0x1000        0x0  ---p   [heap]
            0x470000           0x471000     0x1000        0x0  rw-p   [heap]
        0x7ff7cb1000       0x7ff7cdf000    0x2e000        0x0  r-xp   /usr/lib/libv4lconvert.so.0.0.0
        0x7ff7cdf000       0x7ff7ce1000     0x2000    0x1e000  r--p   /usr/lib/libv4lconvert.so.0.0.0
        0x7ff7ce1000       0x7ff7ce2000     0x1000    0x20000  rw-p   /usr/lib/libv4lconvert.so.0.0.0
        0x7ff7ce2000       0x7ff7d34000    0x52000        0x0  rw-p
        0x7ff7d34000       0x7ff7d63000    0x2f000        0x0  r-xp   /lib/libgcc_s.so.1
        0x7ff7d63000       0x7ff7d64000     0x1000    0x1f000  r--p   /lib/libgc--Type <RET> for more, q to quit, c to continue without paging--
c_s.so.1
        0x7ff7d64000       0x7ff7d65000     0x1000    0x20000  rw-p   /lib/libgcc_s.so.1
        0x7ff7d65000       0x7ff7f26000   0x1c1000        0x0  r-xp   /usr/lib/libstdc++.so.6.0.30
        0x7ff7f26000       0x7ff7f35000     0xf000   0x1c1000  r--p   /usr/lib/libstdc++.so.6.0.30
        0x7ff7f35000       0x7ff7f36000     0x1000   0x1d0000  rw-p   /usr/lib/libstdc++.so.6.0.30
        0x7ff7f36000       0x7ff7f39000     0x3000        0x0  rw-p
        0x7ff7f39000       0x7ff7f58000    0x1f000        0x0  r-xp   /usr/lib/libv4l2.so.0.0.0
        0x7ff7f58000       0x7ff7f59000     0x1000     0xf000  r--p   /usr/lib/libv4l2.so.0.0.0
        0x7ff7f59000       0x7ff7f5d000     0x4000    0x10000  rw-p   /usr/lib/libv4l2.so.0.0.0
        0x7ff7f5d000       0x7ff7fe1000    0x84000        0x0  r-xp   /lib/libc.so
        0x7ff7fe6000       0x7ff7ff8000    0x12000        0x0  rw-p
        0x7ff7ff8000       0x7ff7ffa000     0x2000        0x0  r--p   [vvar]
        0x7ff7ffa000       0x7ff7ffb000     0x1000        0x0  r-xp   [vdso]
        0x7ff7ffb000       0x7ff7ffe000     0x3000    0x8e000  rw-p   /lib/libc.so
--Type <RET> for more, q to quit, c to continue without paging--
        0x7ff7ffe000       0x7ff8000000     0x2000        0x0  rw-p
        0x7ffffdf000       0x8000000000    0x21000        0x0  rw-p   [stack]
```

t3therdev, 2024-01-07:
```
(gdb) run --device=/dev/video0 --get-ctrl=gain
Starting program: /usr/bin/v4l2-ctl --device=/dev/video0 --get-ctrl=gain
warning: Unable to determine the number of hardware watchpoints available.
warning: Unable to determine the number of hardware breakpoints available.
warning: Unable to find dynamic linker breakpoint function.
GDB will be unable to debug shared library initializers
and track explicitly loaded dynamic code.

Program received signal SIGSEGV, Segmentation fault.
0x0000007ff7f77cb0 in ?? ()
(gdb) bt
#0  0x0000007ff7f77cb0 in ?? ()
#1  0x0000007ff7ffe000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
(gdb) info proc mappings
process 19616
Mapped address spaces:

          Start Addr           End Addr       Size     Offset  Perms  objfile
            0x400000           0x447000    0x47000        0x0  r-xp   /usr/bin/v4l2-ctl
            0x45f000           0x460000     0x1000    0x4f000  r--p   /usr/bin/v4l2-ctl
            0x460000           0x463000     0x3000    0x50000  rw-p   /usr/bin/v4l2-ctl
            0x463000           0x46f000     0xc000        0x0  rw-p   [heap]
            0x46f000           0x470000     0x1000        0x0  ---p   [heap]
            0x470000           0x471000     0x1000        0x0  rw-p   [heap]
        0x7ff7cb1000       0x7ff7cdf000    0x2e000        0x0  r-xp   /usr/lib/libv4lconvert.so.0.0.0
        0x7ff7cdf000       0x7ff7ce1000     0x2000    0x1e000  r--p   /usr/lib/libv4lconvert.so.0.0.0
        0x7ff7ce1000       0x7ff7ce2000     0x1000    0x20000  rw-p   /usr/lib/libv4lconvert.so.0.0.0
        0x7ff7ce2000       0x7ff7d34000    0x52000        0x0  rw-p
        0x7ff7d34000       0x7ff7d63000    0x2f000        0x0  r-xp   /lib/libgcc_s.so.1
        0x7ff7d63000       0x7ff7d64000     0x1000    0x1f000  r--p   /lib/libgc--Type <RET> for more, q to quit, c to continue without paging--c
c_s.so.1
        0x7ff7d64000       0x7ff7d65000     0x1000    0x20000  rw-p   /lib/libgcc_s.so.1
        0x7ff7d65000       0x7ff7f26000   0x1c1000        0x0  r-xp   /usr/lib/libstdc++.so.6.0.30
        0x7ff7f26000       0x7ff7f35000     0xf000   0x1c1000  r--p   /usr/lib/libstdc++.so.6.0.30
        0x7ff7f35000       0x7ff7f36000     0x1000   0x1d0000  rw-p   /usr/lib/libstdc++.so.6.0.30
        0x7ff7f36000       0x7ff7f39000     0x3000        0x0  rw-p
        0x7ff7f39000       0x7ff7f58000    0x1f000        0x0  r-xp   /usr/lib/libv4l2.so.0.0.0
        0x7ff7f58000       0x7ff7f59000     0x1000     0xf000  r--p   /usr/lib/libv4l2.so.0.0.0
        0x7ff7f59000       0x7ff7f5d000     0x4000    0x10000  rw-p   /usr/lib/libv4l2.so.0.0.0
        0x7ff7f5d000       0x7ff7fe1000    0x84000        0x0  r-xp   /lib/libc.so
        0x7ff7fe6000       0x7ff7ff8000    0x12000        0x0  rw-p
        0x7ff7ff8000       0x7ff7ffa000     0x2000        0x0  r--p   [vvar]
        0x7ff7ffa000       0x7ff7ffb000     0x1000        0x0  r-xp   [vdso]
        0x7ff7ffb000       0x7ff7ffe000     0x3000    0x8e000  rw-p   /lib/libc.so
        0x7ff7ffe000       0x7ff8000000     0x2000        0x0  rw-p
        0x7ffffdf000       0x8000000000    0x21000        0x0  rw-p   [stack]
```

brada4, 2024-01-07:
(1) #0  0x0000007ff7f77cb0 in ?? () falls into text (code) of libc, 
        0x7ff7f5d000       0x7ff7fe1000    0x84000        0x0  r-xp   /lib/libc.so
(2) backstep makes no sense.

Given (2) I would ask to build individual package (?libv4l?) with LTO disabled and try again. Unless something very adverse happens everything should get backtraced to main() in main executable.

t3therdev, 2024-01-08:
@brada4 when compiling with 

PKG_BUILD_FLAGS: and including no-lto , it did not resolve the issue.

is it posible to update to the next version?

t3therdev, 2024-01-09:
@brada4 are you able to help get this moved to 1.2.4 or 1.2.6?
@thess @neheb

neheb, 2024-06-08:
@t3therdev you must build with
```
TARGET_CFLAGS:= $(filter-out -O%,$(TARGET_CFLAGS)) -Og
```

and disable stripping.

closed 2024-07-31 by BKPepe