openwrt/packages #23088: ksmbd-server: Shared files/directories are not visible in root folder of the share

sppmasterspp, 2024-01-11:
Ksmbd-server Version 3.4.8-1

Maintainer: @\common packages maintainers

```
ubus call system board
{
        "kernel": "6.1.71",
        "hostname": "QNAP",
        "system": "ARMv8 Processor rev 4",
        "model": "QNAP 301w",
        "board_name": "qnap,301w",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "SNAPSHOT",
                "revision": "r24887+2-c43e96d28f",
                "target": "qualcommax/ipq807x",
                "description": "OpenWrt SNAPSHOT r24887+2-c43e96d28f"
        }
}

```

Description:
I saw there were a lot of ksmbd changes (commits) in [kernel 6.1.71 2](https://cdn.kernel.org/pub/linux/kernel/v6.x/ChangeLog-6.1.71).
Having compiled a new build with 6.1.71 I cannot access the root directory of shared disk. Actually it opens but is shown as empty on all devices.
image
[image893×241 13.4 KB](https://forum.openwrt.org/uploads/default/original/3X/9/8/981f66beb39846b25ae20ff56cb1f3593ff2c532.png)

The files are still there
image
[image1344×795 37.6 KB](https://forum.openwrt.org/uploads/default/original/3X/4/2/420b7c0b6ee21f0938f6cf1988fb0a6c3646068f.png)

but not visible in file explorers on different devices.
Writing to the root shared folder is not possible either. I can write to the down directories if I access them directly via address bar.
It worked normally with kernel 6.1.69.
I can view and access the mounted device directories with cd and ls commands via router SSH console and I can open some shared directories down the root directory from Windows File Explorer typing the address in the address bar.

The [discussion on OpenWrt forum](https://forum.openwrt.org/t/ksmbd-samba3-4-alternative-ex-cifsd-smbd-package-support-thread/51695/329?u=sppmaster).

sppmasterspp, 2024-01-13:
The developer said this was known issue resolved in newer versions.

egc112, 2024-01-13:
New kernel version is out and avalaible : https://github.com/openwrt/openwrt/pull/14383 
Maybe this solves your problem?

sppmasterspp, 2024-01-14:
Yes, recompiled this morning with latest changes and all is OK.
The issue is fixed with https://github.com/openwrt/packages/pull/23095. Thanks a lot guys.

sppmasterspp, 2024-01-14:
Fixed.

closed 2024-01-14 by sppmasterspp