openwrt/packages #23092: coova-chilli: xt_coova module doesn't work on OpenWRT 23.05

xsokolikx, 2024-01-12:
Maintainer: @f00b4r0 (?)
Environment: ipq807x,  Xiaomi AX36000, OpenWrt 23.05

Description: After nftables being used as default firewall framework for OpenWRT, I can't use the _xt_coova_ module. The module is built without errors, but the related libxt library (_libxt_coova.so_), doesn't work as expected.

If you try to load any iptables rule containing a _coova_ match, it fails with the following error:

```
root@OpenWRT:~# iptables -I FORWARD -i br-vlan1 -m coova --name chilli -j ACCEPT
iptables v1.8.8 (nf_tables): Couldn't load match `coova':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.

```
Also, if you try to print the help message, it also fails:
```
root@OpenWRT:~# iptables -m coova -h
iptables v1.8.8 (nf_tables): Couldn't load match `coova':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.
```

I already checked that _xt_coova.ko_ module is loaded and I can see the _coova_ entry on _/proc/net/ip_tables_matches_

Any hints?
Thanks!

P.D: related discussion on forum: https://forum.openwrt.org/t/coova-chilli-on-openwrt-22-03-nftables/136941/12

f00b4r0, 2024-01-12:
I'm not the maintainer for coova-chilli and in fact I'm developing a replacement: https://github.com/f00b4r0/uspot (which is now packaged in OpenWrt 23.05). HTH

brada4, 2024-01-12:
It will work with iptables-legacy only.
you need to replace nft conntrack with ipt conntrack so that nft-based fw4 uses ipt modules via xtables bridge

brada4, 2024-01-12:
nft-conntrack - ipt-conntrack-extra
nft-nat - ipt-nat-extra ipt-nat6
nft-offload - ipt-offload
nft-compat - stays, it is THE BRIDGE
iptables-nft - iptables-zz-legacy

xsokolikx, 2024-01-15:
Hi @brada4 ! I'm currently on a custom build, but these are the related packages installed:

```
coova-chilli - 1.6-9.1
firewall4 - 2023-09-01-598d9fbb-1
iptables-mod-conntrack-extra - 1.8.8-2
iptables-mod-extra - 1.8.8-2
iptables-mod-hashlimit - 1.8.8-2
iptables-mod-ipaddr - 3.24-1
iptables-mod-ipopt - 1.8.8-2
iptables-mod-iprange - 1.8.8-2
iptables-mod-ipsec - 1.8.8-2
iptables-mod-nat-extra - 1.8.8-2
iptables-nft - 1.8.8-2
kmod-ip6tables - 5.15.137-1
kmod-ipt-compat-xtables - 5.15.137+3.24-1
kmod-ipt-conntrack - 5.15.137-1
kmod-ipt-conntrack-extra - 5.15.137-1
kmod-ipt-coova - 5.15.137+1.6-9.1
kmod-ipt-core - 5.15.137-1
kmod-ipt-extra - 5.15.137-1
kmod-ipt-hashlimit - 5.15.137-1
kmod-ipt-ipaddr - 5.15.137+3.24-1
kmod-ipt-ipopt - 5.15.137-1
kmod-ipt-iprange - 5.15.137-1
kmod-ipt-ipsec - 5.15.137-1
kmod-ipt-nat - 5.15.137-1
kmod-ipt-nat-extra - 5.15.137-1
kmod-nf-conncount - 5.15.137-1
kmod-nf-conntrack - 5.15.137-1
kmod-nf-conntrack6 - 5.15.137-1
kmod-nf-flow - 5.15.137-1
kmod-nf-ipt - 5.15.137-1
kmod-nf-ipt6 - 5.15.137-1
kmod-nf-log - 5.15.137-1
kmod-nf-log6 - 5.15.137-1
kmod-nf-nat - 5.15.137-1
kmod-nf-reject - 5.15.137-1
kmod-nf-reject6 - 5.15.137-1
kmod-nfnetlink - 5.15.137-1
kmod-nft-bridge - 5.15.137-1
kmod-nft-compat - 5.15.137-1
kmod-nft-core - 5.15.137-1
kmod-nft-fib - 5.15.137-1
kmod-nft-nat - 5.15.137-1
kmod-nft-offload - 5.15.137-1
libxtables12 - 1.8.8-2
nftables-json - 1.0.8-1
xtables-nft - 1.8.8-2
```

brada4, 2024-01-15:
Gradually remove nft modules confirming with `nft list ruleset` after reboot that particular function is offloaded to xt.
Remove iptables-nft and install iptables-zzz-legacy.
You need to offload all nft via xt bridge to have fw4 interact properly with old engine. Then you can program off-tree module without fears to break something.
start with nft-conntrack to build self confidence.

brada4, 2024-01-15:
(you might need to block modules in modules.d if dependencies do not permit uninstall)

Neustradamus, 2024-01-26:
@f00b4r0: Interesting!

pparent76, 2024-01-29:
Any solution for this? I'm having the same problem on snapshot.

brada4, 2024-01-29:
The dependency on x-tables kmod is a compile time option. Simpler to install but slower in day to day operation would be to build coova-chili without kmod and make kmod package an empty boilerplate. Then it will work with either of 2 iptables. It adds individual NAT rules per client MAC.

pparent76, 2024-01-29:
`"The dependency on x-tables kmod is a compile time option"`

Compilation of what package are we talking about here? Cos I'm recompiling coova-chilli and kmod-ipt-coova from sdk, and it does not seem to work.

brada4, 2024-01-29:
You need to deselect kmod thewn coova will not catch option and built in clean iptables mode.

pparent76, 2024-01-29:
Yes, sure it can operate without kmod. If you remove "kname chilli" from the config file it will not use the module anyway (without having to recompile).

But if the module is not used it is way more CPU consuming right? And on embedded hardware with low-spec CPU, the CPU charge can be a bottleneck no? ( At least it was the case on previous versions, I don't know if things changed on newer coova-chilli versions ).

brada4, 2024-01-29:
It uses iptables to insert rules that can be offloaded to flowtables (or xt offload), it may not be that slow in the end.

pparent76, 2024-01-29:
Well I will do some testing on the last version to check that out. But I'm sure that on older version on a MT7620 CPU you would not get a throughput higher that 20Mbits through coova if you did not use the kmod-ipt-coova (and this using 100% cpu).

brada4, 2024-01-29:
Watch out - there is config option - if module is built even empty it adds module requirement to main executable. You need to rip out that line

pparent76, 2024-01-30:
But where did you see that "It uses iptables command to insert rules that can be offloaded to flowtables (or xt offload)"? I've never seen chilli process automatically inserting iptables rules when a client get authenticated....

But it makes me think that maybe it could be done quite easily to add a per client firewall allowance within the conup.sh and condown.sh scripts.

pparent76, 2024-01-30:
I've done some tests and I can get something somewhat working and not consuming much CPU by inserting per customer iptables from conup.sh script:

`iptables -I FORWARD -s $FRAMED_IP_ADDRESS -i br-lan -j ACCEPT`

And deleting it from condown. But it's really a hack and I'm not sure about the reliability of this.

pparent76, 2024-01-30:
Thing is I'm trying to investigate why the libxt_coova refuses to load, because from what I understand now, there is no actual reason for this to happen. Because other match modules succeed in loading without problem with last version of iptables-legacy.

I guess there might be something to adapt in here: https://github.com/coova/coova-chilli/blob/master/src/linux/libxt_coova.c

brada4, 2024-01-30:
iptables v1.8.8 (nf_tables) -> this emulates most common -legacy modules and emits native nft rules. You need real legacy iptables to interact with other modules. 
xt_coova module just maintains simple list of IP4 and IP6 addresses to forward, and is programmed via procfs. nftables or ipsets does same by large.

brada4, 2024-01-30:
You can load it from these tables https://github.com/coova/coova-chilli/blob/7d18bd6b24fd20a4d8cde2a2f9b34a214327f07c/src/linux/xt_coova.c#L598

pparent76, 2024-01-30:
```

root@xxxxxx$ iptables -I FORWARD -m coova  --help
iptables v1.8.8 (legacy): Couldn't load match `coova':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.
root@xxxxxx$ iptables -I PREROUTING -t nat -m coova  --help
iptables v1.8.8 (legacy): Couldn't load match `coova':No such file or directory

```

brada4, 2024-01-30:
Is it published in /proc/net/ip_tables_matches?

brada4, 2024-01-31:
The userspace libxt_coova library is not even packaged or installed.

brada4, 2024-01-31:
It is certainly a packaging bug that /usr/lib/iptables/libxt_coova.so is not installed. Anyone may recall when?

pparent76, 2024-01-31:
Well actually if you inspect this package you will see that the file /usr/lib/iptables/libxt_coova.so  is there:

https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/coova-chilli_1.6-10_aarch64_cortex-a53.ipk

brada4, 2024-01-31:
xt_coova module cannot load in iptables useless without coova-chilli full package. Sounds reasonable.

pparent76, 2024-02-12:
Well in any case I guess we're stuck here.

What I would like to try is to recompile the package with a modified version of libxt_coova.c to try to understand what's going wrong when loading the match library.
https://github.com/coova/coova-chilli/blob/master/src/linux/libxt_coova.c

Any indication on how to perform that with SDK?

brada4, 2024-02-12:
Last change in that file is 10 years ago. It has mandatory sections - like register_match, parameters and help text...  It is installed in correct place, by another package , but in correct place. I cannot think on what needs to be corrected in the file itself.

pparent76, 2024-02-12:
I agree with you, but I cannot think of other tests to do, in order to understand why this match library fails to load, while other succeed.

On of the tests I would like to do is replace the code with the code of libxt_string (while keeping the name of libxt_coova, to see at least if it loads then:

https://android.googlesource.com/platform/external/iptables/+/master/extensions/libxt_string.c

Other test could also include adding debug prints of course.

brada4, 2024-02-12:
The string blocks seem re-ordered, not certain by LTO.

`PKG_BUILD_FLAGS:=no-lto`

pparent76, 2024-02-12:
> In my case, libxt_coova.so is also installed and I see how iptables load the library correctly using strace.

By the way for anyone interested here is the strace

We can see that it first tries to load "/usr/lib/iptables/libipt_coova.so" which fails (I guess it's normal) and then falls back to loading "/usr/lib/iptables/libxt_coova.so" which seems to load sucesfully.

```
strace iptables -I FORWARD -m coova -j ACCEPT
execve("/usr/sbin/iptables", ["iptables", "-I", "FORWARD", "-m", "coova", "-j", "ACCEPT"], 0xffffd4ede430 /* 13 vars */) = 0
set_tid_address(0xffffa34e9258)         = 13372
brk(NULL)                               = 0x1bc4d000
brk(0x1bc4f000)                         = 0x1bc4f000
mmap(0x1bc4d000, 4096, PROT_NONE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x1bc4d000
openat(AT_FDCWD, "/etc/ld-musl-aarch64.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/libiptext.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/lib/libiptext.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/libiptext.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
fstat(3, {st_mode=S_IFREG|0755, st_size=138418, ...}) = 0
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
mmap(NULL, 204800, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xffffa3415000
mmap(0xffffa3443000, 16384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x1e000) = 0xffffa3443000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libiptext4.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/lib/libiptext4.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/libiptext4.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
fstat(3, {st_mode=S_IFREG|0755, st_size=66490, ...}) = 0
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
mmap(NULL, 135168, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xffffa33f4000
mmap(0xffffa3413000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0xf000) = 0xffffa3413000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libiptext6.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/lib/libiptext6.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/libiptext6.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
fstat(3, {st_mode=S_IFREG|0755, st_size=66490, ...}) = 0
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
mmap(NULL, 135168, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xffffa33d3000
mmap(0xffffa33f2000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0xf000) = 0xffffa33f2000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libip4tc.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/lib/libip4tc.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/libip4tc.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
fstat(3, {st_mode=S_IFREG|0755, st_size=66066, ...}) = 0
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
mmap(NULL, 135168, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xffffa33b2000
mmap(0xffffa33d1000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0xf000) = 0xffffa33d1000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libip6tc.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/lib/libip6tc.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/libip6tc.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
fstat(3, {st_mode=S_IFREG|0755, st_size=66066, ...}) = 0
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
mmap(NULL, 135168, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xffffa3391000
mmap(0xffffa33b0000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0xf000) = 0xffffa33b0000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libxtables.so.12", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/lib/libxtables.so.12", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/libxtables.so.12", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
fstat(3, {st_mode=S_IFREG|0755, st_size=65548, ...}) = 0
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
mmap(NULL, 139264, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xffffa336f000
mmap(0xffffa338e000, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0xf000) = 0xffffa338e000
mmap(0xffffa3390000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xffffa3390000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
[... 13 lines cut ...]
[... cut ...]
writev(2, [{iov_base="iptables v1.8.8 (legacy): ", iov_len=26}, {iov_base=NULL, iov_len=0}], 2iptables v1.8.8 (legacy): ) = 26
writev(2, [{iov_base="Couldn't load match `coova':No s"..., iov_len=54}, {iov_base=NULL, iov_len=0}], 2Couldn't load match `coova':No such file or directory
) = 54
writev(2, [{iov_base="", iov_len=0}, {iov_base="\n", iov_len=1}], 2
) = 1
writev(2, [{iov_base="Try `iptables -h' or 'iptables -"..., iov_len=61}, {iov_base=NULL, iov_len=0}], 2Try `iptables -h' or 'iptables --help' for more information.
) = 61
exit_group(2)                           = ?
+++ exited with 2 +++
```

pparent76, 2024-02-12:
> PKG_BUILD_FLAGS:=no-lto

I'm not sure to follow you here. What is this no-lto?
Do you think adding this option in the SDK could solve the problem?

brada4, 2024-02-12:
Adding to coova-chilli  package

pparent76, 2024-02-12:
I recompiled the coova-chilli package adding the option PKG_BUILD_FLAGS:=no-lto in the makefile  ( feeds/packages/net/coova-chilli/makefile ), it did not change anything.

brada4, 2024-02-12:
You sound  right.... 
https://git.netfilter.org/iptables/tree/extensions/libxt_tcp.c#n472
v22 is broken too it seems.

pparent76, 2024-02-12:
From my understanding what happens is a that a library called crti.o (seemingly linked by default and probably part of the C library because it is located at /lib/gcc/arc-openwrt-linux-gnu/11.3.0/crti.o ), already implements _init() , which enters in conflict with the _init() of libxt_coova.c, and cause the whole thing to fail to compile.

In order to solve that the Patch that was added changes the name of the _init() function in libxt_coova.c , so that it gets to compile. But it does not feel to be the right solution because then the match library cannot get loaded from iptables.

To my guess the solution would be: to remove the patch, and to find a way to specify to the compiler via an option that we don't want to link against crti.o. For now I don't know how to do the later.

brada4, 2024-02-12:
```
libxt_DSCP.so:libxt_DSCP_init
libxt_TOS.so:libxt_TOS_init
libxt_coova.so:xtables_init <---------------- 
libxt_dscp.so:libxt_dscp_init
libxt_ecn.so:libxt_ecn_init
```

brada4, 2024-02-12:
Maybe rename _init to libxt_coova_init ?

pparent76, 2024-02-12:
Yes that's what I'm currently trying.

pparent76, 2024-02-12:
It works!!

brada4, 2024-02-12:
fork repo
clone yours with ssh
branch
change
add 
commit
make pull request

pparent76, 2024-02-12:
Yes, I was going to do that!

Thank you so much for the help by the way!

brada4, 2024-02-12:
yw, thanks for brilliant research.

brada4, 2024-02-12:
Fixes: https://github.com/openwrt/packages/commit/55297e45c85d0ab3ba31e413c0c58729cddbf72f

pparent76, 2024-02-12:
> yw, thanks for brilliant research.

Thank's a lot!

Here is the pull request: https://github.com/openwrt/packages/pull/23386

( Hope that I've respected approximately the guidelines of the project to make the PR )

xsokolikx, 2024-02-27:
Thanks for you work, @pparent76 and @brada4 ! 

I think we can close this issue now.

closed 2024-02-27 by xsokolikx

brada4, 2024-02-27:
@neheb  patch should be backported to stable branches as in OP?

raheelhirani35, 2024-11-14:
@brada4  Hi, 

I am trying to run coovachilli with and without xt_coova on openwrt 23.05 i tried your procedure to unload nft modules but still its not redirecting to splash page. Though iptables rules looks fine. Please suggest what am i missing. 
this is my list of package installed:
base-files - 1545-r23497-6637af95aa
busybox - 1.36.1-1
ca-bundle - 20230311-1
cgi-io - 2022-08-10-901b0f04-21
coova-chilli - 1.6-10
dnsmasq - 2.89-4
dropbear - 2022.82-5
firewall4 - 2023-03-23-04a06bd7-1
fstools - 2023-02-28-bfe882d5-1
fwtool - 2019-11-12-8f7fe925-1
getrandom - 2022-08-13-4c7b720b-2
haserl - 0.9.36-1
hostapd-common - 2023-09-08-e5ccbfc6-4
iptables-zz-legacy - 1.8.8-1
iw - 5.19-1
iwinfo - 2023-07-01-ca79f641-1
jansson4 - 2.14-3
jshn - 2023-05-23-75a3b870-1
jsonfilter - 2018-02-04-c7e938d6-1
kernel - 5.15.134-1-d9fd917b385e2572dbbf5ec32e92c76d
kmod-cfg80211 - 5.15.134+6.1.24-3
kmod-crypto-acompress - 5.15.134-1
kmod-crypto-aead - 5.15.134-1
kmod-crypto-ccm - 5.15.134-1
kmod-crypto-cmac - 5.15.134-1
kmod-crypto-crc32c - 5.15.134-1
kmod-crypto-ctr - 5.15.134-1
kmod-crypto-gcm - 5.15.134-1
kmod-crypto-gf128 - 5.15.134-1
kmod-crypto-ghash - 5.15.134-1
kmod-crypto-hash - 5.15.134-1
kmod-crypto-hmac - 5.15.134-1
kmod-crypto-manager - 5.15.134-1
kmod-crypto-null - 5.15.134-1
kmod-crypto-rng - 5.15.134-1
kmod-crypto-seqiv - 5.15.134-1
kmod-crypto-sha512 - 5.15.134-1
kmod-gpio-button-hotplug - 5.15.134-3
kmod-ip6tables - 5.15.134-1
kmod-ipt-conntrack - 5.15.134-1
kmod-ipt-conntrack-extra - 5.15.134-1
kmod-ipt-coova - 5.15.134+1.6-10
kmod-ipt-core - 5.15.134-1
kmod-ipt-nat - 5.15.134-1
kmod-ipt-nat-extra - 5.15.134-1
kmod-ipt-nat6 - 5.15.134-1
kmod-ipt-offload - 5.15.134-1
kmod-leds-gpio - 5.15.134-1
kmod-lib-crc-ccitt - 5.15.134-1
kmod-lib-crc32c - 5.15.134-1
kmod-lib-lzo - 5.15.134-1
kmod-mac80211 - 5.15.134+6.1.24-3
kmod-mmc - 5.15.134-1
kmod-mt76-core - 5.15.134+2023-08-14-b14c2351-1
kmod-mt7603 - 5.15.134+2023-08-14-b14c2351-1
kmod-mt76x02-common - 5.15.134+2023-08-14-b14c2351-1
kmod-mt76x2 - 5.15.134+2023-08-14-b14c2351-1
kmod-mt76x2-common - 5.15.134+2023-08-14-b14c2351-1
kmod-nf-conncount - 5.15.134-1
kmod-nf-conntrack - 5.15.134-1
kmod-nf-conntrack6 - 5.15.134-1
kmod-nf-flow - 5.15.134-1
kmod-nf-ipt - 5.15.134-1
kmod-nf-ipt6 - 5.15.134-1
kmod-nf-log - 5.15.134-1
kmod-nf-log6 - 5.15.134-1
[... 325 lines cut ...]

	chain mangle_postrouting {
		type filter hook postrouting priority mangle; policy accept;
	}

	chain mangle_input {
		type filter hook input priority mangle; policy accept;
	}

	chain mangle_output {
		type route hook output priority mangle; policy accept;
	}

	chain mangle_forward {
		type filter hook forward priority mangle; policy accept;
		iifname "wan" tcp flags syn tcp option maxseg size set rt mtu comment "!fw4: Zone wan IPv4/IPv6 ingress MTU fixing"
		oifname "wan" tcp flags syn tcp option maxseg size set rt mtu comment "!fw4: Zone wan IPv4/IPv6 egress MTU fixing"
	}
}

Please suggest what I am missing in this?

@pparent76  please help if you are manage to run coovachilli with xtcoova in 23.05.

Many thanks

raheelhirani35, 2024-11-14:
Thank you for your reply @brada4.  

For Opt1 do we have to change anything in coovachilli code or in up.sh or down.sh script for iptable rules? Also we donot need any kmod-ipt for this option?

For opt3 can we compile 23.05 without fw4 and only with fw3 + iptables-zz+kmod-ipt? If yes can you please provide guide because i saw fw4 is by default in 23.05 

Thanks once again

brada4, 2024-11-14:
Opt1: ipt modules will be installed as a dependecy of iptables-nft
Opt3: you do not have to compile anything, remove firewall4 add firewall in e.g imagebuilder

raheelhirani35, 2024-11-14:
Hi @brada4 

Thank you for your response. I tried opt1 with fw4 and iptables-nft keeping ipt and nft modules in there but it is not redirecting user to page on connect with coovachilli. I have just used fw4, iprables-nft and kmod_nft and ipt and did not remove anything and run coovachilli as it is without any change. Am i missing something?

Also for opt3 i checked using makemenuconfig i cannot remove firewall4. How to add firewall3 and remove firewall4 during image builder a little guide will be helpful. Thank you.

Thanks once again.

brada4, 2024-11-14:
Check `iptables --version` -> should say nft
Then `iptables-save` -> should contain 2 coova rules
Then `nft list ruleset` -> should contain xt match coova or something like that twice

raheelhirani35, 2024-11-14:
Hi @brada4 , thank you your reply and sorry to bother you but i just need a way as i saw discussion here that it will work.

iptables --version:
iptables v1.8.8 (nf_tables): no command specified

iptables--save:
*mangle
:PREROUTING ACCEPT [0:0]
:INPUT ACCEPT [0:0]
:FORWARD ACCEPT [0:0]
:OUTPUT ACCEPT [0:0]
:POSTROUTING ACCEPT [0:0]
-A FORWARD -p tcp -m tcp --tcp-flags SYN,RST SYN -j TCPMSS --clamp-mss-to-pmtu
COMMIT
# Completed on Thu Nov 14 16:53:39 2024
# Generated by iptables-save v1.8.8 (nf_tables) on Thu Nov 14 16:53:39 2024
*filter
:INPUT ACCEPT [0:0]
:FORWARD ACCEPT [0:0]
:OUTPUT ACCEPT [0:0]
-A INPUT -i br-lan -j DROP
-A INPUT -d 192.168.182.1/32 -i tun0 -p icmp -j ACCEPT
-A INPUT -d 192.168.182.1/32 -i tun0 -p udp -m udp --dport 53 -j ACCEPT
-A INPUT -d 192.168.182.1/32 -i tun0 -p udp -m udp --dport 67:68 -j ACCEPT
-A INPUT -d 255.255.255.255/32 -i tun0 -p udp -m udp --dport 67:68 -j ACCEPT
-A INPUT -d 192.168.182.1/32 -i tun0 -p tcp -m tcp --dport 4990 -j ACCEPT
-A INPUT -d 192.168.182.1/32 -i tun0 -p tcp -m tcp --dport 3990 -j ACCEPT
-A INPUT -d 192.168.182.1/32 -i tun0 -j DROP
-A FORWARD -i tun0 -o wan -j ACCEPT
-A FORWARD -i tun0 ! -o wan -j DROP
-A FORWARD -p tcp -m tcp --tcp-flags SYN,RST SYN -j TCPMSS --clamp-mss-to-pmtu
-A FORWARD -o tun0 -j ACCEPT
-A FORWARD -i tun0 -j ACCEPT
-A FORWARD -o br-lan -j DROP
-A FORWARD -i br-lan -j DROP
COMMIT

nft list ruleset has this iptables-nft based rules. (i am currently testing it without xt_coova for it to work atleast)

# Warning: table ip filter is managed by iptables-nft, do not touch!
table ip filter {
	chain INPUT {
		type filter hook input priority filter; policy accept;
		iifname "br-lan" counter packets 331 bytes 35098 drop
		iifname "tun0" meta l4proto icmp ip daddr 192.168.182.1 counter packets 0 bytes 0 accept
		iifname "tun0" ip daddr 192.168.182.1 udp dport 53 counter packets 0 bytes 0 accept
		iifname "tun0" ip daddr 192.168.182.1 udp dport 67-68 counter packets 0 bytes 0 accept
		iifname "tun0" ip daddr 255.255.255.255 udp dport 67-68 counter packets 0 bytes 0 accept
		iifname "tun0" ip daddr 192.168.182.1 tcp dport 4990 counter packets 0 bytes 0 accept
		iifname "tun0" ip daddr 192.168.182.1 tcp dport 3990 counter packets 0 bytes 0 accept
		iifname "tun0" ip daddr 192.168.182.1 counter packets 0 bytes 0 drop
	}

	chain FORWARD {
		type filter hook forward priority filter; policy accept;
		iifname "tun0" oifname "wan" counter packets 0 bytes 0 accept
		iifname "tun0" oifname != "wan" counter packets 0 bytes 0 drop
		tcp flags syn / syn,rst counter packets 1012 bytes 60720 xt target "TCPMSS"
		oifname "tun0" counter packets 0 bytes 0 accept
		iifname "tun0" counter packets 0 bytes 0 accept
		oifname "br-lan" counter packets 0 bytes 0 drop
		iifname "br-lan" counter packets 1307 bytes 149363 drop
	}
}
# Warning: table ip mangle is managed by iptables-nft, do not touch!
table ip mangle {
	chain FORWARD {
		type filter hook forward priority mangle; policy accept;
		tcp flags syn / syn,rst counter packets 1012 bytes 60720 xt target "TCPMSS"
	}

Please guide if you see any issue in here.

Thanks

brada4, 2024-11-14:
Strange xt target tcpmss, shoould use native tcpmss 
```
iptables-translate  -A FORWARD -p tcp -m tcp --tcp-flags SYN,RST SYN -j TCPMSS --clamp-mss-to-pmtu
nft add rule ip filter FORWARD tcp flags syn / syn,rst counter tcp option maxseg size set rt mtu
```

raheelhirani35, 2024-11-14:
Hi @brada4,

thanks for reply.

I am now using xt coova:
this updated one for iptables-save:

*mangle
:PREROUTING ACCEPT [0:0]
:INPUT ACCEPT [0:0]
:FORWARD ACCEPT [0:0]
:OUTPUT ACCEPT [0:0]
:POSTROUTING ACCEPT [0:0]
COMMIT
# Completed on Thu Nov 14 18:00:18 2024
# Generated by iptables-save v1.8.8 (nf_tables) on Thu Nov 14 18:00:18 2024
*filter
:INPUT ACCEPT [0:0]
:FORWARD ACCEPT [0:0]
:OUTPUT ACCEPT [0:0]
:INPUT_tun0 - [0:0]
:INPUT_tun1 - [0:0]
-A INPUT -i tun1 -j INPUT_tun1
-A INPUT -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT
-A INPUT -d 11.1.0.1/32 -i br-lan2 -p tcp -m tcp --dport 3990 -m coova--name chilli --source  -j ACCEPT
-A INPUT -i tun13 -j ACCEPT
-A INPUT -j DROP
-A FORWARD -o tun1 -j ACCEPT
-A FORWARD -i tun1 -j ACCEPT
-A FORWARD -o br-lan2 -m coova--name chilli --dest  -j ACCEPT
-A FORWARD -i br-lan2 -m coova--name chilli --source  -j ACCEPT
-A FORWARD -i tun1 -o wan -j ACCEPT
-A FORWARD -i wan -o tun1 -j ACCEPT
-A FORWARD -i wan -o br-lan2 -m coova--name chilli --dest  -j ACCEPT
-A FORWARD -i br-lan2 -o wan -m coova--name chilli --source  -j ACCEPT
-A FORWARD -i wan -o br-lan2 -m coova--name chilli --dest  -j ACCEPT
-A FORWARD -i tun0 -o wan -j ACCEPT
-A FORWARD -i wan -o tun0 -j ACCEPT
-A INPUT_tun1 -d 11.1.0.1/32 -i tun1 -p tcp -m tcp --dport 3990 -j ACCEPT
-A INPUT_tun1 -j RETURN
COMMIT

this is nft list rule set :

# Warning: table ip filter is managed by iptables-nft, do not touch!
table ip filter {
	chain INPUT {
		type filter hook input priority filter; policy accept;
		iifname "tun1" counter packets 0 bytes 0 jump INPUT_tun1
		xt match "conntrack" counter packets 118 bytes 7860 accept
		iifname "br-lan2" ip daddr 11.1.0.1 tcp dport 3990 xt match "coova" counter packets 0 bytes 0 accept
		iifname "tun13" counter packets 0 bytes 0 accept
		counter packets 23 bytes 7608 drop
	}

	chain FORWARD {
		type filter hook forward priority filter; policy accept;
		oifname "tun1" counter packets 0 bytes 0 accept
		iifname "tun1" counter packets 0 bytes 0 accept
		oifname "br-lan2" xt match "coova" counter packets 0 bytes 0 accept
		iifname "br-lan2" xt match "coova" counter packets 0 bytes 0 accept
		iifname "tun1" oifname "wan" counter packets 0 bytes 0 accept
		iifname "wan" oifname "tun1" counter packets 0 bytes 0 accept
		iifname "wan" oifname "br-lan2" xt match "coova" counter packets 0 bytes 0 accept
		iifname "br-lan2" oifname "wan" xt match "coova" counter packets 0 bytes 0 accept
		iifname "wan" oifname "br-lan2" xt match "coova" counter packets 0 bytes 0 accept
		iifname "tun0" oifname "wan" counter packets 0 bytes 0 accept
		iifname "wan" oifname "tun0" counter packets 0 bytes 0 accept
	}

	chain INPUT_tun0 {
	}

	chain INPUT_tun1 {
		iifname "tun1" ip daddr 11.1.0.1 tcp dport 3990 counter packets 0 bytes 0 accept
		counter packets 0 bytes 0 return
	}
}
table ip mangle {
	chain FORWARD {
		type filter hook forward priority mangle; policy accept;
	}
}

This tcpmss now removed from mangle but still not working not redirecting to uamserver

brada4, 2024-11-14:
Please go to forums, and learn about formatting buttons `</>` and `<>`
Counters are incomplete, they show just existing connections continuing, you can flush conntrack with `conntrack -D` or wait 2h for all states to expire.

aswanthk07, 2025-01-07:
Hi @raheelhirani35 , Did you figure how to make coova redirect using fw4 and xt_coova? Please let me know if you having any working solution. Thanks