openwrt/packages #23099: ocserv: UCI option to use alternative SLScertificate instead of self-signed one

jacklovell, 2024-01-14:
Maintainer: @\nmav (find it by checking history of the package Makefile)
Environment: mips_24kc, BT Home Hub 5 (lantiq/xrx200), OpenWrt 23.05.2 r23630-842932a63d

Description:

It is useful to be able to use a Let's Encrypt SSL certificate for ocserv. Currently the package generates a self-signed certificate and private key on first run, which then needs to be imported into all clients which can be inconvenient/cumbsersome. The `acme` package can automatically manage Let's Encrypt certificates placed in /etc/ssl/acme: it would be nice to have a UCI option to tell ocserv to use one of these certificates instead of its self-signed one.

```
Format code blocks by wrapping them with pairs of ```
```

nmav, 2024-01-14:
That would be great. Would you like to submit a patch?

brada4, 2024-01-14:
You can change -f ...pem to -e ...pem in init script and replace cert location with symlink to acme products.

jacklovell, 2024-01-15:
I'm happy to give this a go, but I can't dedicate any time to it for a few weeks yet.