openwrt/packages #23102: Zabbix Agent 6.4.x Fails to compile

ddubnyk, 2024-01-14:
Maintainer: @champtar 
Environment: (For Linksys EA8450, Mediatek MT7622, Master branch, build machine=Gentoo or Debian)

Description:
When building with zabbix agent 6.4.x, i'm getting an error related to libevent. Tried 6.4.6 all the way up to 6.4.9.

I can successfully build with any version of 6.2, (currently went with 6.2.9) however even on a fresh build environment, 6.4 always fails with this error..

```
checking for libevent support... configure: error: Unable to use libevent (libevent check failed)
make[3]: *** [Makefile:445: /mnt/build/openwrt/build_dir/target-aarch64_cortex-a53_musl/zabbix-nossl/zabbix-6.4.7/.configured_fd514688267783f347afaec29bc4c495] Error 1
make[3]: Leaving directory '/mnt/build/openwrt/feeds/packages/admin/zabbix'
time: package/feeds/packages/zabbix/nossl/compile#3.07#1.83#4.45
    ERROR: package/feeds/packages/zabbix failed to build (build variant: nossl).
make[2]: *** [package/Makefile:128: package/feeds/packages/zabbix/compile] Error 1

```

brada4, 2024-01-14:
Does proxy-nossl build? It has libevent dependency in place. If so just copy section to agent and 'aybe pr

ddubnyk, 2024-01-15:
> Does proxy-nossl build? It has libevent dependency in place. If so just copy section to agent and maybe pr

Adding in the proxy fixed it thank you!

brada4, 2024-01-16:
Maybe wait for maintainer to clise with patch?

ddubnyk, 2024-01-16:
Alright I'll reopen this. 

As an FYI this solution also is working with 6.4.10 release of the agent.

ddubnyk, 2024-10-02:
As an update, this is broken on anything newer than 6.4.7.  Same error as initially.
I've added in the proxy and same error persists.  It was fine up until sometime recently.  I know I had compiled up to 6.4.15 perfectly fine so long as I had the proxy selected as well.  But now nothing newer than 6.4.7 works.

danielfdickinson, 2026-01-19:
@systemcrash @BKPepe @GeorgeSapkin @hnyman This can be closed as well: https://github.com/openwrt/packages/commit/1f3251545df7f7ad05c6684d5c60e1a7e4487882 updates Zabbix in master to 7.0.22. I can also grab the 7.0.21 updates for 25.12 and 24.10 (25.12 I intend to cherry-pick the 7.0.22 update).