openwrt/packages #23171: modemmanager: netifd giving up on enabling a modem device

yegorich, 2024-01-26:
Maintainer: @mips171 / @aleksander0m
Environment:  ARMv7, TI Sitara AM335x, OpenWrt v23.05.2

Description:

In the following log though the modem could be found, netifd is giving up and doens't try to get it up once more. Restarting the interface via the web interface leads to a successful conncetion.

```
Fri Jan 26 16:58:12 2024 daemon.notice netifd: Mobile (3482): stopping network
Fri Jan 26 16:58:12 2024 daemon.notice netifd: Mobile (3482): error: couldn't find modem
Fri Jan 26 16:58:12 2024 daemon.notice netifd: Mobile (3482): couldn't load bearer path: disconnecting anyway
Fri Jan 26 16:58:13 2024 daemon.notice netifd: Interface 'Mobile' is now down
Fri Jan 26 16:58:19 2024 daemon.notice [2957]: <msg> [device /sys/devices/platform/ocp/47400000.target-module/47401400.usb/musb-hdrc.0/usb1/1-1/1-1.3] creating modem with plugin 'simtech' and '7' ports
Fri Jan 26 16:58:19 2024 daemon.info [2957]: <inf> [modem0] tty/ttyUSB2: at (primary)
Fri Jan 26 16:58:19 2024 daemon.info [2957]: <inf> [modem0] tty/ttyUSB3: at (secondary)
Fri Jan 26 16:58:19 2024 daemon.info [2957]: <inf> [modem0] tty/ttyUSB2: at (data primary)
Fri Jan 26 16:58:19 2024 daemon.info [2957]: <inf> [modem0] net/wwan0: net (data)
Fri Jan 26 16:58:19 2024 daemon.info [2957]: <inf> [modem0] tty/ttyUSB1: gps (nmea)
Fri Jan 26 16:58:19 2024 daemon.info [2957]: <inf> [modem0] tty/ttyUSB4: audio
Fri Jan 26 16:58:19 2024 daemon.info [2957]: <inf> [modem0] usbmisc/cdc-wdm0: qmi
Fri Jan 26 16:58:19 2024 daemon.notice [2957]: <msg> [base-manager] modem for device '/sys/devices/platform/ocp/47400000.target-module/47401400.usb/musb-hdrc.0/usb1/1-1/1-1.3' successfully created
Fri Jan 26 16:58:21 2024 daemon.info [2957]: <inf> [modem0] SIM hot swap setup succeeded
Fri Jan 26 16:58:21 2024 daemon.notice [2957]: <msg> [modem0] state changed (unknown -> locked)
Fri Jan 26 16:58:21 2024 daemon.warn [2957]: <wrn> [modem0] error initializing: Modem is currently locked, cannot fully initialize
Fri Jan 26 16:58:21 2024 daemon.info ModemManager[2925]: hotplug: interface 'Mobile' set '/sys/devices/platform/ocp/47400000.target-module/47401400.usb/musb-hdrc.0/usb1/1-1/1-1.3' state available
Fri Jan 26 16:58:21 2024 daemon.notice netifd: Interface 'Mobile' is setting up now
Fri Jan 26 16:58:22 2024 daemon.notice netifd: Mobile (3906): modem available at /org/freedesktop/ModemManager1/Modem/0
Fri Jan 26 16:58:24 2024 user.notice root: Found modem: 0
Fri Jan 26 16:58:24 2024 user.notice root: 868020033875112
Fri Jan 26 16:58:24 2024 user.notice nitz: error: modem not enabled yet
Fri Jan 26 16:58:24 2024 daemon.info procd: - init complete -
Fri Jan 26 16:58:25 2024 daemon.info [2957]: <inf> [modem0] SIM is ready, and no need for the after SIM unlock step...
Fri Jan 26 16:58:25 2024 daemon.notice netifd: Mobile (3906): successfully sent PIN code to the SIM
Fri Jan 26 16:58:25 2024 daemon.notice [2957]: <msg> [modem0] state changed (locked -> initializing)
Fri Jan 26 16:58:25 2024 daemon.notice netifd: Mobile (3906): setting allowed mode to 'any'
Fri Jan 26 16:58:25 2024 daemon.info [2957]: <inf> [modem0] SIM is ready, and no need for the after SIM unlock step...
Fri Jan 26 16:58:25 2024 daemon.info [2957]: <inf> [modem0/sim0] loaded IMSI: ###
Fri Jan 26 16:58:25 2024 daemon.info [2957]: <inf> [modem0/sim0] loaded operator identifier: 46692
Fri Jan 26 16:58:25 2024 daemon.notice netifd: Mobile (3906): error: couldn't set current modes: 'GDBus.Error:org.freedesktop.ModemManager1.Error.Core.WrongState: modem in initializing state'
Fri Jan 26 16:58:25 2024 daemon.info [2957]: <inf> [modem0/sim0] loaded GID1: FFFFFFFFFFFFFFFF
Fri Jan 26 16:58:25 2024 daemon.info [2957]: <inf> [modem0/sim0] loaded GID2: FFFFFFFFFFFFFFFF
Fri Jan 26 16:58:25 2024 daemon.notice netifd: Mobile (3906): error: couldn't enable the modem: 'GDBus.Error:org.freedesktop.ModemManager1.Error.Core.WrongState: modem in initializing state'
Fri Jan 26 16:58:26 2024 daemon.notice netifd: Mobile (4140): stopping network
Fri Jan 26 16:58:26 2024 daemon.warn [2957]: <wrn> [modem0] couldn't load supported assistance data types: LOC service: general failure
Fri Jan 26 16:58:26 2024 daemon.notice netifd: Mobile (4140): couldn't load bearer path: disconnecting anyway
Fri Jan 26 16:58:26 2024 daemon.notice [2957]: <msg> [modem0] state changed (initializing -> disabled)
Fri Jan 26 16:58:26 2024 daemon.info [2957]: <inf> [modem0] processing user request to disconnect modem: all bearers
Fri Jan 26 16:58:26 2024 daemon.info [2957]: <inf> [modem0] all requested bearers disconnected
Fri Jan 26 16:58:26 2024 daemon.notice netifd: Interface 'Mobile' is now down
```

How to force netifd to try the initialization again?

feckert, 2024-01-26:
What OpenWrt version are you using?
I made some changes recently, that should fix this issue on boot.
https://github.com/openwrt/packages/commits/master/net/modemmanager

yegorich, 2024-01-26:
OpenWrt: r24436-0f283ab4c9, packages: f9c90fce0aad9ebec04e09b10843da3571c09f83 and two patches from @osedlbauer.

feckert, 2024-01-26:
Ok, so you already have the latest changes. Normally, the new monitor service is doing this task.

When the modemmanager has recognized the modem, then the monitor service [marks](https://github.com/openwrt/packages/blob/master/net/modemmanager/files/usr/sbin/ModemManager-monitor#L86) the uci interface section as avialable. And only then the netifd should start a connection attempt.

Is your log the output when booting?

yegorich, 2024-01-27:
Yes, it is the bootlog. I have added more output.

yegorich, 2024-02-09:
@feckert have already seen this [thread](https://lists.freedesktop.org/archives/modemmanager-devel/2024-February/009818.html)? Seems to be related.

feckert, 2024-02-12:
I think that could solve the problem. This change has the effect that the proto setup should be tried again and again as long as ifdown <iface> is not entered. The only question now is whether we should introduce a new option `keep_retry` for this, as it changes the current behavior. @yegorich Can you test this?
```diff
diff --git a/net/modemmanager/files/modemmanager.proto b/net/modemmanager/files/modemmanager.proto
index e67a84cf2..849e66709 100755
--- a/net/modemmanager/files/modemmanager.proto
+++ b/net/modemmanager/files/modemmanager.proto
@@ -587,8 +587,8 @@ proto_modemmanager_setup() {
        append_param "${password:+password=${password}}"

        mmcli --modem="${device}" --timeout 120 --simple-connect="${connectargs}" || {
-               proto_notify_error "${interface}" MM_CONNECT_FAILED
-               proto_block_restart "${interface}"
+               echo "connection failed -> attempting restart"
+               proto_notify_error "${interface}" MM_CONNECT_RESTART
                return 1
        }
```

yegorich, 2024-02-13:
@feckert will do. Though, it will take some time.

feckert, 2024-06-12:
The first part with the reconnect has been fixed https://github.com/openwrt/packages/pull/24370. Thanks to @osedlbauer. What is still missing is that the modem is not switched off during a reconnect in order to improve the reconnect time.