openwrt/packages #23185: openconnect: "Write error on DTLS session: Error in the push function." after established connection & no RX

gohai, 2024-01-29:
Maintainer: @nmav
Environment: ipq806x/generic, Netgear R7800, 22.03.6 & 23.05.2

Description:

After a successful connection with a Cisco AnyConnect VPN server gets established, the error `Write error on DTLS session: Error in the push function.` gets written to the logs, and the RX counter of the interface remains at zero (no connectivity).

* OpenWrt 22.03.6 & "bring up on boot": this *always happens* for me
* OpenWrt 22.03.6 & manual bringup after boot: this *does not* occur (works fine)
* OpenWrt 23.05.2 & "bring up on boot": this *always happens* for me
* OpenWrt 23.05.2 & manual bringup after boot: this *always happens* for me

I'd gladly try to help figure out what's going on with logs, or experiments. Please let me know what I can do.

---

\<speculation>I believe this might be timing dependent, perhaps an interaction with the firewall, which gets reloaded after the interface is up?</speculation>

nmav, 2024-01-29:
Very strange error. What else is on the logs? You can disable DTLS if that causes a problem in your case. The issue is UDP-related - there is an error in the UDP/DTLS socket, but what it is is unknown. The rest of the logs may help as well as the incoming UDP/ICMP packets during that message.

gohai, 2024-01-29:
Thank you @nmav. I uploaded the output of logread for you [here](https://gist.github.com/gohai/b3e47ae107f8f060921c59126adfefcd). I will look into the other suggestions now.

gohai, 2024-01-29:
I had the same symptoms with `-no-dtls`. No connectivity. RX counter on the VPN interface remains at zero. Log [here](https://gist.github.com/gohai/99554d7465498aa14a8ef0b7c0de63aa).

gohai, 2024-01-29:
I have two short package captures, one broken (bring up on boot) - and one working (manual bringup after boot). Could I share those with you privately @nmav?

I'm a novice looking at this, but in the broken one I see lots of STP RST packets from my ISP router and TCP retransmissions from the VPN server after the DTLSv1.2 handshake. In the working one, there's an uninterrupted exchange of DTLSv1.2 application data.

nmav, 2024-01-30:
If the reset is from your ISP, then I would suggest trying the router in a different environment (i.e., different ISP) before we dig too hard here. I cannot think of why an ISP would try to kill VPNs connections, but it can happen.

brada4, 2024-02-02:
> Not using vhost-net due to low queue length 

should be 256 by default...

gohai, 2024-02-03:
@nmav I haven't made any changes to my setup since my earlier messages, but suddenly it started working randomly at some reboots.

When I diff the logs one thing jumps out: When it does work I see
```
openconnect: initializing...
openconnect: adding host dependency for vpn.shanghai.nyu.edu at vpn_nyush
user.notice openconnect: adding host dependency for 103.242.128.100 at vpn_nyush
user.notice openconnect: executing 'openconnect 'vpn.shanghai.nyu.edu' ...
```

Whereas when it doesn't, the line about adding the resolved IP of the VPN server is absent from the logs. 

```
openconnect: initializing...
openconnect: adding host dependency for vpn.shanghai.nyu.edu at vpn_nyush
user.notice openconnect: executing 'openconnect 'vpn.shanghai.nyu.edu' ...
```

... which is then later always followed by

```
daemon.notice openconnect[]: Write error on DTLS session: Error in the push function.
daemon.notice openconnect[]: Failed to send DPD request. Expect disconnect
daemon.notice openconnect[]: Write error on DTLS session: The specified session has been invalidated for some reason.
```

I don't quite know (yet) what the implication of "host dependencies" are - but is it possible that the absence of it explains the symptoms I am seeing?

gohai, 2024-02-03:
When I manually call `proto_add_host_dependency` in the event the `resolveip` doesn't yield any IPs, this indeed appears to make it work for me.

I also noticed that my timestamp jumped forward around the time when this happened. My router doesn't have a RTC. Is it possible that `resolveip` runs into a timeout prematurely when the system clock changes?

```
Sat Feb  3 12:56:17 2024 user.notice openconnect: adding host dependency for vpn.somewhere.edu at vpn_somewhere
(...)
Sat Feb  3 12:56:18 2024 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Sat Feb  3 14:31:34 2024 user.notice openconnect: Unable to resolve VPN server IP, adding manually
```

gohai, 2024-02-03:
Turns out `resolveip` returns with `2`, which means that `getaddrinfo()` returned with an error code. I am seeing some _Address not available_ errors by other services around this time in the logs as well - so I am guessing that this will be the cause.

Would it make sense to make `openconnect.sh` retry when `resolveip` returns an error code?

brada4, 2024-02-03:
it uses alarm() for timeout and posix says it is from realtime clock, not interval timer.

maxflutehd, 2024-02-10:
I have the same problem. I am trying to connect to a university vpn with openwrt 23.05.02.
It appears on "bring up on boot" and manual starting.
Coneccting via android app works fine in my network.
I attached my log.
Is it the same issue or another?
[log_blurred.log](https://github.com/openwrt/packages/files/14227416/log_blurred.log)

nmav, 2024-02-10:
Please check https://github.com/openwrt/packages/pull/23348

gohai, 2024-02-11:
Thank you @nmav - I will test (w/ 22.03.06) on Thursday and report back. (Confident that it will fix the issue.)

gohai, 2024-02-15:
@nmav #23348 works for me (after fixing the `-t` argument) on 22.03.06. I am seeing one or more retries before it ultimately succeeds. Thank you very much.

Note that `resolveip` always returned error code `2` for me when it failed, which means that my issue was rather `getaddrinfo()` failing with an error code than the timeout occurring.

commit 20ea7260 (2024-02-19): openconnect: make host dependency more resilient

Retry when resolveip fails as it seems to be causing issues
on startup depending on various unpredictable parameters.

Resolves: #23185
closed 2024-02-19 by nmav