openwrt/packages #23188: freeradius3-default: radiusd does not start

odi79, 2024-01-29:
Maintainer: none?
Environment: tested with openwrt-23.05.2-x86-64-generic (and also with ath79-generic-glinet_gl-ar750)

Description:
**radiusd fails to start after installing the basic freeradius3 packages using "opkg install freeradius3-default".**
Back in October 2023 (at times of openwrt-23.05.0-rc4), everything was working.

Currently the following steps/workarounds are needed to get started:

**1)**
```
opkg install libopenssl-legacy
```
to avoid
```
# radiusd -X
[...]
(TLS) Failed loading legacy provider
```
This is issue #21428.

**2)**
Comment out some lines in ```/etc/freeradius3/sites-available/default``` (e.g. by using ```sed -i '/^\tif (EAP-Key-Name/,+4 s/^/#/' /etc/freeradius3/sites-available/default```)
```
1054,1058c1054,1058
<       if (EAP-Key-Name && &reply:EAP-Session-Id) {
<               update reply {
<                       &EAP-Key-Name := &reply:EAP-Session-Id
<               }
<       }
---
> #     if (EAP-Key-Name && &reply:EAP-Session-Id) {
> #             update reply {
> #                     &EAP-Key-Name := &reply:EAP-Session-Id
> #             }
> #     }
```
to avoid
```
# radiusd -X
[...]
/etc/freeradius3/sites-enabled/default[1054]: Parse error in condition
/etc/freeradius3/sites-enabled/default[1054]: (EAP-Key-Name && &reply:EAP-Session-Id) {
/etc/freeradius3/sites-enabled/default[1054]:  ^ Expected a module return code
Errors reading or parsing /etc/freeradius3/radiusd.conf
```
Note that this is just a quick workaround to get rid of the message and not a solution. What is the root cause of this error?

**3)**
Add missing file ```/etc/freeradius3/mods-config/attr_filter/coa``` to avoid
```
# radiusd -X
[...]
reading pairlist file /etc/freeradius3/mods-config/attr_filter/coa
Couldn't open /etc/freeradius3/mods-config/attr_filter/coa for reading: No such file or directory
Errors reading /etc/freeradius3/mods-config/attr_filter/coa
/etc/freeradius3/mods-enabled/attr_filter[58]: Instantiation failed for module "attr_filter.coa"
```
This was already fixed by @predators46 in repository (master branch) with 104098c38ba5c603a2a78b478d349e62b8072ed2 but is not yet available in the downloadable package.

With 1) to 3), ```radiusd -X``` starts again.

I don't know whether it is related, but freeradius3 issues started for me with https://forum.openwrt.org/t/freeradius3-packages-missing/176271/2 caused by #22006 and further tracked in #22532. However, #22670/ #22666 made the package compile/build again. 
Currently issue #22576 is still open. It might improve something...?

In my opinion, "freeradius3" is a pretty basic package and should work properly out-of-the-box in the currently released OpenWrt version. It is unfortunate that it does not work for weeks now. Happy for everybody who can help fix it!

brada4, 2024-01-30:
PCRE issues are not related, now it is built against libc RE support.

odi79, 2024-02-05:
Note that the file ``/etc/freeradius3/sites-available/default`` is the same on Alpine Linux and does not yield any errors there. Thus the root cause of 2) is somewhere else.

morales-alex, 2024-03-20:
Thanks, @odi79, for your helpful post: I can't reach to fix the third issue. Do I need to download the version which has that commit version you said?

odi79, 2024-03-20:
Glad that I'm not the only one attempting to get freeradius working under OpenWrt, @morales-alex .

The missing file ``/etc/freeradius3/mods-config/attr_filter/coa`` is taken from upstream [https://github.com/FreeRADIUS/freeradius-server/releases/](https://github.com/FreeRADIUS/freeradius-server/releases/) by the OpenWrt Makefile (net/freeradius3/Makefile). For convenience, here is the content:
```
#
#	Configuration file for the rlm_attr_filter module.
#	Please see rlm_attr_filter(5) manpage for more information.
#
#	$Id: 89cea2ea97dea10b82a8146cfeeeb1d7dd33b2f8 $
#
#	This configuration file is used to remove attributes From an
#	CoA-Request or Disconnect-Request message.  We have specified
#	a sample list here.  This will have to be modified to add
#	attributes needed by your local configuration.
#
DEFAULT
	User-Name =* ANY,
	NAS-IP-Address =* ANY,
	NAS-IPv6-Address =* ANY,
	NAS-Port =* ANY,
	NAS-Identifier =* ANY,
	NAS-Port-Type =* ANY,
	Calling-Station-Id =* ANY,
	State =* ANY,
	Message-Authenticator =* ANY,
	Proxy-State =* ANY
```

Does anybody know the root cause of the second issue and can fix it? The content of ``/etc/freeradius3/sites-available/default`` is working on Alpine Linux but yields the mentioned parse error on OpenWrt...

odi79, 2024-03-27:
Situation unchanged with OpenWrt 23.05.3  :-(

trippleflux, 2024-03-28:
Found out most probably second issue solution : 
```
--- a/raddb/mods-available/eap
+++ b/raddb/mods-available/eap
@@ -705,7 +705,7 @@
 			#  deleted by the server when the command
 			#  returns.
 			#
-		#	client = "/path/to/openssl verify -CApath ${..ca_path} %{TLS-Client-Cert-Filename}"
+		#	client = "/usr/bin/openssl verify -CApath ${..ca_path} %{TLS-Client-Cert-Filename}"
 		}
 
 		#  OCSP Configuration

```

or edit /etc/freeradius3/mods-available/eap on running OpenWrt , this is found after digging through the alpinelinux's [freeradius package](https://git.alpinelinux.org/aports/tree/main/freeradius/default-config.patch) ,  not really sure on why commenting in "client" variable is still being evaluated by freeradius. On short test, no more error with second issue.

odi79, 2024-05-08:
Thanks for looking into it @trippleflux and posting your findings!
Can anybody confirm that the posted config change works to get rid of the second issue? Unfortunately, it does not work for me.

trippleflux, 2024-05-10:
@odi79
Forgot to mention that I am also using this version bump [PR](https://github.com/openwrt/packages/pull/22576). Also I don't know to test freeradius3 functionalities wise, my test was only running radiusd binary until it reach "Ready to process requests" .

sqrammi, 2024-06-07:
Found out why the first parsing error occurs. The rfc4072 is incorrectly not included in the release.

@@ -536,6 +537,7 @@ PKG_DICTIONARIES:= \
        compat \
        freeradius freeradius.internal \
        rfc2865 rfc2866 rfc2867 rfc2868 rfc2869 rfc3162 rfc3576 rfc3580 \
+       rfc4072 \
        rfc4372 rfc4675 rfc4679 rfc5580\
        microsoft \
        wispr \

Neustradamus, 2024-10-26:
Have you progressed on this ticket?

odi79, 2024-10-26:
> Have you progressed on this ticket?
When downloading a current OpenWRT 23.05.5 and installing freeradius3-default on it, one still faces the same issues as at the beginning of this year. However, I hope that we now collected all needed steps/resolutions in this ticket to allow one of the maintainers/developers to fix the issues.

qzhwang, 2024-11-16:
Same issue. Can we bump this up? Been a while.

qzhwang, 2024-11-16:
@sqrammi Can we have an updated version that includes that RFC? Or is there a way to build it by ourselves, with complete details on how to build one? Thanks!

sqrammi, 2024-11-16:
@qzhwang Here is a better-formatted patch. You patch the freeradius Makefile, like this:
```
--- a/net/freeradius3/Makefile
+++ b/net/freeradius3/Makefile
@@ -131,6 +131,7 @@ define Package/freeradius3-mod-attr-filter/conffiles
 /etc/freeradius3/mods-config/attr_filter/access_challenge
 /etc/freeradius3/mods-config/attr_filter/access_reject
 /etc/freeradius3/mods-config/attr_filter/accounting_response
+/etc/freeradius3/mods-config/attr_filter/coa
 /etc/freeradius3/mods-config/attr_filter/post-proxy
 /etc/freeradius3/mods-config/attr_filter/pre-proxy
 endef
@@ -536,6 +537,7 @@ PKG_DICTIONARIES:= \
        compat \
        freeradius freeradius.internal \
        rfc2865 rfc2866 rfc2867 rfc2868 rfc2869 rfc3162 rfc3576 rfc3580 \
+       rfc4072 \
        rfc4372 rfc4675 rfc4679 rfc5580\
        microsoft \
        wispr \
```

qzhwang, 2024-11-17:
Thanks @sqrammi for the diff! But I've been kinda trying not to build it myself. I mean if the build is broken 'coz of this, can we patch the Makefile and create a new package that can be installed by opkg?

qzhwang, 2024-11-19:
@sqrammi ohh, I see. Thought you were a maintainer. Never mind. Thanks for the help anyways!

Neustradamus, 2024-11-19:
Dear @LucileQ, @dangowrt, @tohojo, @champtar, @hauke, @cotequeiroz, @Robby-, @yangfl, @audricschiltknecht, @ja-pa, @jefferyto, @uedvt359, @hnyman, @ematsu, @odmdas, @neheb, @zhaojh329, @crza, @predators46, @esaaprillia,

You have contributed in the past about freeradius in OpenWrt, can you look this ticket?

Thanks in advance.

uedvt359, 2024-11-19:
it's pretty rude to randomly tag people in unrelated tickets.

predators46, 2024-11-19:
@Neustradamus 

as explained above overcomes this problem. I have tried quickly radiusd -X everything works as desired. i have added perl module to freeradius

```
root@OpenWrt:/etc/freeradius3# radiusd -X
FreeRADIUS Version 3.2.6
Copyright (C) 1999-2023 The FreeRADIUS server project and contributors
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE
You may redistribute copies of FreeRADIUS under the terms of the
GNU General Public License
For more information about these matters, see the file named COPYRIGHT
Starting - reading configuration files ...
including dictionary file /usr/share/freeradius3/dictionary
including dictionary file /etc/freeradius3/dictionary
including configuration file /etc/freeradius3/radiusd.conf
including configuration file /etc/freeradius3/proxy.conf
including configuration file /etc/freeradius3/clients.conf
including files in directory /etc/freeradius3/mods-enabled/
including configuration file /etc/freeradius3/mods-enabled/always
including configuration file /etc/freeradius3/mods-enabled/attr_filter
including configuration file /etc/freeradius3/mods-enabled/chap
including configuration file /etc/freeradius3/mods-enabled/detail
including configuration file /etc/freeradius3/mods-enabled/digest
including configuration file /etc/freeradius3/mods-enabled/eap
including configuration file /etc/freeradius3/mods-enabled/mschap
including configuration file /etc/freeradius3/mods-enabled/exec
including configuration file /etc/freeradius3/mods-enabled/expiration
including configuration file /etc/freeradius3/mods-enabled/expr
including configuration file /etc/freeradius3/mods-enabled/files
including configuration file /etc/freeradius3/mods-enabled/logintime
including configuration file /etc/freeradius3/mods-enabled/pap
including configuration file /etc/freeradius3/mods-enabled/preprocess
including configuration file /etc/freeradius3/mods-enabled/radutmp
including configuration file /etc/freeradius3/mods-enabled/sradutmp
including configuration file /etc/freeradius3/mods-enabled/realm
including configuration file /etc/freeradius3/mods-enabled/unix
including configuration file /etc/freeradius3/mods-enabled/passwd
including configuration file /etc/freeradius3/mods-enabled/perl
including files in directory /etc/freeradius3/policy.d/
including configuration file /etc/freeradius3/policy.d/accounting
including configuration file /etc/freeradius3/policy.d/filter
including configuration file /etc/freeradius3/policy.d/eap
including files in directory /etc/freeradius3/sites-enabled/
including configuration file /etc/freeradius3/sites-enabled/default
including configuration file /etc/freeradius3/sites-enabled/inner-tunnel
main {
 security {
        allow_core_dumps = no
 }
        name = "radiusd"
        prefix = "/usr"
        localstatedir = "/var"
        logdir = "/var/log"
        run_dir = "/var/run/radiusd"
}
main {
        name = "radiusd"
        prefix = "/usr"
        localstatedir = "/var"
        sbindir = "/usr/sbin"
        logdir = "/var/log"
        run_dir = "/var/run/radiusd"
        libdir = "/usr/lib/freeradius3"
        radacctdir = "/var/db/radacct"
        hostname_lookups = no
        max_request_time = 30
        proxy_dedup_window = 1
        cleanup_delay = 5
[... 552 lines cut ...]
}
listen {
        type = "acct"
        ipv6addr = ::
        port = 0
   limit {
        max_connections = 16
        lifetime = 0
        idle_timeout = 30
   }
}
listen {
        type = "auth"
        ipaddr = 127.0.0.1
        port = 18120
}
Listening on auth address * port 1812 bound to server default
Listening on acct address * port 1813 bound to server default
Listening on auth address :: port 1812 bound to server default
Listening on acct address :: port 1813 bound to server default
Listening on auth address 127.0.0.1 port 18120 bound to server inner-tunnel
Listening on proxy address * port 56717
Listening on proxy address :: port 52916
Ready to process requests
```

qzhwang, 2024-11-20:
@predators46 Could you help post some details on how you get 3.2.6 working on OpenWrt? I'm only getting `3.0.26` from `openwrt-23.05.5`. Also tried to compile it from the source but could even get the build tools to work properly on openwrt. Some more details would be much appreciated! Thanks!

PS, not sure how this is related to Perl...

predators46, 2024-11-20:
@qzhwang 

If you use OpenWRT version 21 you don't need libopenssl-legacy.

but if you use openwrt version 23 you need libopenssl-legacy.

but if you want to use pcre in freeradius you can build it in openwrt version 21 and don't need libopenssl-legacy.

If you want to use pcre2 in freeradius you need to wait for freeradius4 to be released in the stable branch

how to test freeradius3 is running as desired you need to install freeradius3-default and use the radiusd -X command in terminal/putty

notes :

perl has nothing to do with this problem

qzhwang, 2024-11-20:
Thanks @predators46 . I just installed `libopenssl-legacy` and `libpcre`, since I'm on openwrt 23. `radiusd -X` still generates the same error as the original poster had. (I removed and re-installed `freeradius3-default`.) :-/

qzhwang, 2024-11-20:
Again, I'm getting `FreeRADIUS Version 3.0.26`. How did you get `3.2.6`? Did you compile it from source? If so, could yo share some details on how to do that? I can't even get the build tools to work on OpenWRT...

dannutu, 2024-12-28:
> I'm only getting `3.0.26` from `openwrt-23.05.5`. 
same here as of 28 Dec