openwrt/packages #23193: emailrelay package 2.4.1-1 is old, appreciate maintainer update to current version 2.5.2

aon33, 2024-01-30:
Maintainer: @stokito  @BKPepe
Environment: mips GNU/Linux, OpenWrt 23.05

Description:

Appreciate any maintainer can update the emailrelay package to latest version 2.5.2 (it fixed dnsbl options), thank you.

stokito, 2024-01-30:
The dnsbl was broken in v2.5.0 but the current openwrt version is 2.3. You may use `extra_cmdline` to set options manually.

We have a PR #18536 with update to 2.5 so I'll update it.
Also given that there was a lot of discussion I'll probably close the PR and create a new one.

aon33, 2024-01-30:
Thanks for the fast reply.  I have been using the 'extra_cmdline' but I just keep getting errors despite trying various '-dnsbl' parameters.  Then I saw the changlog mentioned a fix so was hoping a updated version will fix it.

Some attempts below, any hints as to the correct format it expects?
```
'--dnsbl=dnsbl-2.uceprotect.net'
emailrelay: error: invalid dnsbl configuration: not enough comma-sparated fields

'--dnsbl=dnsbl-2.uceprotect.net,1,1000,8.8.8.8:53'
emailrelay: error: invalid dnsbl configuration: address error: no port separator: invalid network address: dnsbl-2.uceprotect.net

```

stokito, 2024-01-30:
Check documentation:
https://emailrelay.sourceforge.net/index.html#reference_md_Connection_blocking

```
For backwards compatibility the comma-separated fields can be reversed:

--dnsbl 1.1.1.1:53,500,1,spam.example.com,block.example.com
```

So in your case try:
`--dnsbl=8.8.8.8:53,1,1000,dnsbl-2.uceprotect.net`

stokito, 2024-01-30:
cool, then please close the issue. The emailrelay will be updated later

aon33, 2024-01-30:
Really appreciate your help, thanks again.

closed 2024-01-30 by aon33