openwrt/packages #23196: tailscale: update from 1.32.0 to 1.58.2

cyberqueenmeg, 2024-01-30:
# Issue template (remove lines from top till here)

Maintainer: @mochaaP @BKPepe
Environment: amd64, Raspberry Pi 4, current OpenWRT/opkg

Description:

```
Tailscale package needs to be updated in opkg from 1.32.0 to 1.58.2 (or whatever the current version is at time of fix.

Current tailscale version in opkg
<img width="555" alt="Screenshot 2024-01-30 at 11 12 04 AM" src="https://github.com/openwrt/packages/assets/62813469/d97dbee9-130d-41a6-99ef-06e1e1929625">

Current released tailscale version
<img width="878" alt="Screenshot 2024-01-30 at 11 12 34 AM" src="https://github.com/openwrt/packages/assets/62813469/82cadba3-f430-409c-b953-d9e0ae4efde8">

```

mochaaP, 2024-01-30:
tailscale is already 1.58.2 in the master branch.

cyberqueenmeg, 2024-01-30:
Must be my system then. Tried upgrading the firmware and opkg and wouldn't pull the newest version. Sorry to disturb.

closed 2024-01-30 by cyberqueenmeg