openwrt/packages #23206: openthread-br cp: cannot overwrite non-directory failed to build

outbackdingo, 2024-01-31:
```
mkdir -p /home/dingo/openwrt-r4/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp
SHELL= flock /home/dingo/openwrt-r4/tmp/.root-copy.flock -c 'cp -fpR /home/dingo/openwrt-r4/build_dir/target-aarch64_cortex-a53_musl/openthread-br-2023-08-01-1738d8cd/.pkgdir/openthread-br/. /home/dingo/openwrt-r4/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/'
cp: cannot overwrite non-directory '/home/dingo/openwrt-r4/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/./var' with directory '/home/dingo/openwrt-r4/build_dir/target-aarch64_cortex-a53_musl/openthread-br-2023-08-01-1738d8cd/.pkgdir/openthread-br/./var'
make[3]: *** [Makefile:109: /home/dingo/openwrt-r4/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp/.openthread-br_installed] Error 1
make[3]: Leaving directory '/home/dingo/openwrt-r4/feeds/packages/net/openthread-br'
time: package/feeds/packages/openthread-br/compile#91.78#9.31#102.11
    ERROR: package/feeds/packages/openthread-br failed to build.
make[2]: *** [package/Makefile:128: package/feeds/packages/openthread-br/compile] Error 1
make[2]: Leaving directory '/home/dingo/openwrt-r4'
make[1]: *** [package/Makefile:122: /home/dingo/openwrt-r4/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/dingo/openwrt-r4'
make: *** [/home/dingo/openwrt-r4/include/toplevel.mk:232: world] Error 2
```

brada4, 2024-01-31:
openthread-br Makefile is 107 lines long, could you help with changes you made in your package tree?

outbackdingo, 2024-01-31:
this is on a stock fresh openwrt trunk build, we have not changed or modified any packages

kaizirlewagen, 2024-02-09:
@outbackdingo i've got the same error. Randomly and really nothing changed at code. 

```make package/openthread-br/compile``` works, but make the complete firmware does end with error 2 cp: cannot overwrite non-directory. I try a fresh .config, check makefile of openthread-br. But if i can compile the package standalone, i am confused about the error.

Do you found a solution for this error?

dhewg, 2024-02-09:
```
diff --git a/net/openthread-br/Makefile b/net/openthread-br/Makefile
index 1b42cceee..8a4a90743 100644
--- a/net/openthread-br/Makefile
+++ b/net/openthread-br/Makefile
@@ -94,10 +94,8 @@ endef
 
 define Package/openthread-br/install
    $(INSTALL_DIR) \
-       $(1)/etc/init.d \
        $(1)/lib/netifd/proto \
-       $(1)/usr/sbin \
-       $(1)/var/lib/thread
+       $(1)/usr/sbin
    $(INSTALL_BIN) ./files/openthread-proto.sh $(1)/lib/netifd/proto/openthread.sh
    $(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/sbin/* $(1)/usr/sbin
 endef
```

kaizirlewagen, 2024-02-12:
@dhewg works perfect. Thx for this patch.

closed 2024-02-27 by stintel