openwrt/packages #23275: https-dns-proxy canary domains logic backwards ?

professor-jonny, 2024-02-04:
Maintainer: @stangri

Description:

```
It seems the settings for ''let local devices use icloud private relay'' and  ''let local devices use mozilla private relay'' have their logic inverted, they are actually blocked when those settings are enabled unless I am mistaken.

```

stangri, 2024-02-05:
How do you test?

professor-jonny, 2024-02-05:
[image]
[image]

It puts the domains in the DNS forwarding section alongside my DNS servers, I would expect them to be removed when the Domains are blocked.

stangri, 2024-02-05:
> I would expect them to be removed when the Domains are blocked.

Why?

professor-jonny, 2024-02-06:
Those canary domains are used for the private relay and having them listed would then use them as valid DNS forwarders.
unless I'm missing something, it seems to me, they are back too front as when i select to block them they appear.
resolving these domains with NXDOMAIN in your network tells browsers to disable DoH from my understanding.

I would of thought the way to block them would be to add them to the "adb_list_overall" in the DNSMASQ config directory in tmp and remove them from the DNS forwarder section of DNSMASQ.

egc112, 2024-02-07:
https://support.mozilla.org/en-US/kb/canary-domain-use-application-dnsnet
`A negative result will be a signal to disable application DNS, (i.e., DoH).`
So adding this to DNSMasq will result in the client **not** using DoH: `address=/use-application-dns.net/`

stangri, 2024-02-07:
> Those canary domains are used for the private relay and having them listed would then use them as valid DNS forwarders. unless I'm missing something, it seems to me, they are back too front as when [...]

They are not listed as servers, they are in the dhcp config to make sure they don't resolve and NXDOMAIN is returned is a result for them.

professor-jonny, 2024-02-07:
Ok I must have something configured wrong then as those domains resolve unless I add them to AdBlock but it looks like it is all configured ok.

stangri, 2024-02-08:
> Ok I must have something configured wrong then as those domains resolve unless I add them to AdBlock but it looks like it is all configured ok.

I can't reproduce:
```
xg-135r3 in ~ # nslookup mask.icloud.com
Server:		127.0.0.1
Address:	127.0.0.1:53

** server can't find mask.icloud.com: NXDOMAIN

** server can't find mask.icloud.com: NXDOMAIN

xg-135r3 in ~ # nslookup mask-h2.icloud.com
Server:		127.0.0.1
Address:	127.0.0.1:53

** server can't find mask-h2.icloud.com: NXDOMAIN

** server can't find mask-h2.icloud.com: NXDOMAIN
```

closed 2024-02-08 by stangri