openwrt/packages #23297: ddns: Updater services not started when PPPoE connection is established

schuettecarsten, 2024-02-05:
On latest snapshot, ddns updaters are not started when PPPoE connection is established, regardless if I choose then pppoe-wan interface or the @wan or @wan6 aliases as event networks. I cannot make it work and have to restart ddns services by hand after PPPoE connects. I can see serveral notifications from netifd about wan and wan6 coming up, but nothing about ddns at all.

Looks like the hotplug.d scripts are not executed or do not behave as expected.

edrikk, 2024-02-06:
Can confirm this behavior.
I used to be on Cable Internet (DHCP) until a month ago and there were no issues with respect to DDNS.  As soon as I switched over to PPPOE based ISP DDNS behaved as described here.  I have added the below to my local startup to workaround the issue:

```
### Workaround for ddns not starting properly
{ sleep 120; /etc/init.d/ddns restart; }  &
```

feckert, 2024-02-07:
What exactly is happing?
Does it not start on boot?
What exactly do you have to do to make it work again.

There were no changes recently that explain this issue in the [ddns-scripts](https://github.com/openwrt/packages/commits/master/net/ddns-scripts).

schuettecarsten, 2024-02-07:
@feckert - It does not start the "service processes" when the PPPoE connection is up. Workaround is to click the "Reload" buttons after the PPPoE connection was established or to use @edrikk's workaround.

There have been no changes to the ddns-scripts, but there have been changes to netifd and others and maybe the hotplug scripts now also need to be updated?

feckert, 2024-02-09:
If you have a setup that doesn't work! Can you check whether the netifd calls the script?
https://github.com/openwrt/packages/blob/master/net/ddns-scripts/files/etc/hotplug.d/iface/ddns

Please add some debug log messages to see what happens

schuettecarsten, 2024-02-11:
I've added some logging to the hotplug script 

```
#!/bin/sh

logger -t ddns "$ACTION of $INTERFACE ($DEVICE)"

# there are other ACTIONs like ifupdate we don't need
case "$ACTION" in
        ifup)                                   # OpenWrt is giving a network not phys. Interface
                /etc/init.d/ddns enabled && /usr/lib/ddns/dynamic_dns_updater.sh -n "$INTERFACE" -- start
                logger -t ddns "Starting ddns due to $ACTION of $INTERFACE ($DEVICE)"
                ;;
        ifdown)
                /usr/lib/ddns/dynamic_dns_updater.sh -n "$INTERFACE" -- stop
                logger -t ddns "Stopping ddns due to $ACTION of $INTERFACE ($DEVICE)"
                ;;
esac
```

Now Ican see these lines:

```
Sun Feb 11 16:53:50 2024 user.notice ddns: ifup of wan (pppoe-wan)
Sun Feb 11 16:53:50 2024 user.notice ddns: Starting ddns due to ifup of wan (pppoe-wan)
[...]
Sun Feb 11 16:53:58 2024 user.notice ddns: ifup of wan6 (pppoe-wan)
Sun Feb 11 16:53:58 2024 user.notice ddns: Starting ddns due to ifup of wan6 (pppoe-wan)
```

Here are the relevant parts of the ddns config:

```
 8 config service 'ipv6'
10         option use_ipv6 '1'
11         option enabled '1'
21         option interface '@wan6'
26         option bind_network 'wan6'
27
28 config service 'ipv4'
30         option use_ipv6 '0'
31         option enabled '1'
41         option interface '@wan'
46         option bind_network 'wan'
```

schuettecarsten, 2024-02-11:
Just a quick idea, the hotplug script notified about wan and wan6, but the ddns LuCI extension writes @wan and @wan6 to the config...?

jackjinke, 2024-10-10:
I have the same issue that DDNS daemons does not start after a reboot, and I’m using PPPoE. Additionally, I’m using the IP from the `pppoe-wan` interface directly, and configured all this via LuCI.

Inspired by @schuettecarsten ‘s comment:
> Just a quick idea, the hotplug script notified about wan and wan6, but the ddns LuCI extension writes @wan and @wan6 to the config...?

I checked my /etc/config/ddns file and found my old config has set `option interface 'pppoe-wan'`. I have then changed the config to use `wan` or `wan_6` here, similar to the following, and now DDNS daemons can start normally after a reboot:
```
config service 'myddns_ipv4'
    option enabled '1'
    option interface 'wan'
    option ip_source 'interface'
    option use_ipv6 '0'
    option force_ipversion '1'
    option ip_interface 'pppoe-wan'

config service 'myddns_ipv6`
    option enabled '1'
    option interface 'wan_6'
    option ip_source 'interface'
    option use_ipv6 '1'
    option force_ipversion '1'
    option ip_interface 'pppoe-wan'
```

Note that:
- Do not include the @ before `wan` or `wan_6`. LuCI seems to do this, and you’ll need to manually update the config file to change it.
- I’m using IP from interface, so I have the `option ip_source 'pppoe-wan'` config. Despite the name, it’s actually expecting an device name instead of interface name for this field, so you cannot use `wan` or `wan_6` here.