openwrt/packages #23364: Openconnect: camouflage + non 443 port. Incorrect executing

itdoginfo, 2024-02-11:
Maintainer: @neheb @vgaetera @dhewg @rsalvaterra (Sorry guys, you're the people who have been committing for the last year)

Environment: OpenWrt 23.05, OpenWrt SNAPSHOT r25115-7181eb9f81 (11.02.2024)

Description:
I use openconnect server with camouflage and 8443 port. And running a client on openwrt.

Config client:
```
config interface 'vpn0'
	option proto 'openconnect'
	option server 'test.com/?NUqVl3YLuX'
	option port '8443'
	option username 'test'
	option password 'uxNCieYKh6'
```

Log:
```
Sun Feb 11 09:12:25 2024 user.notice openconnect: executing 'openconnect 'test.com/?NUqVl3YLuX:8443' '-i' 'vpn-vpn0' '--non-inter' '--syslog' '--script' '/lib/netifd/vpnc-script' '-u' 'test' '--passwd-on-stdin''
```

As you can see, the port is substituted after the URI. And in this case, there is no connection.
I tested this on the last stable and dev version.

It's good working on port 443.

Thanks.

nmav, 2024-02-11:
Adding anything beyond the server in the server option will not work. We will need to change the openconnect scripts and possibly to the lua interface for this to work.

An option that is backwards compatible is to allow an https URI on the server field and if found to skip the server:port construction.

nmav, 2024-05-10:
@itdoginfo does https://github.com/openwrt/packages/pull/24109 work for you?

closed 2024-05-13 by nmav

root149, 2025-02-19:
24.10  option uri 'test.com:8433/?NUqVl3YLuX' 
notwork

PR #24109 (2024-05-13): openconnect: introduced URI parameter
Maintainer: me
Compile tested: arm_cortex-a9+vfpv3-d16_musl_eabi
Run tested: -

Description:
This introduces the uri configuration option in openconnect to replace the 'server' and 'port' legacy options. This allows specifying a camouflage string in ocserv.

Fixes: #23364
commit ab8adb7d (2024-05-13): openconnect: introduced URI parameter

This allows specifying a camouflage string in ocserv.

Fixes: #23364
commit a14bb12d (2024-05-18): openconnect: introduced URI parameter

This allows specifying a camouflage string in ocserv.

Fixes: #23364