openwrt/packages #23368: netifd: not bringing up interface with NO_DEVICE after edit

danog, 2024-02-11:
Environment: u7621-06, 256m, openwrt 23.05.0, netifd 2024-01-04-c18cc79d-1

Description:

Changing some configuration parameters of a 6in4 or wireguard interface leads to netifd not being able to bring up the interface when running ifup, with ifstatus returning NO_DEVICE.

Config:
```
option proto '6in4'
        option peeraddr 'xxx.xxx.xxx.xxx'
        option ip6addr '2a0e:97c0:38f:fffd::1/64'
        list ip6prefix '2a0e:97c0:38f::/64'
        option ip6assign '64'
        option mtu '1472'
        option ipaddr 'yyy.yyy.yyy.yyy'
        option ttl '255'
```

Debug config in /etc/network/interfaces:
```
start_service() {
        init_switch

        procd_open_instance
        procd_set_param command /sbin/netifd -l 5 -d 15
        procd_set_param stdout 1
        procd_set_param stderr 1
        procd_set_param respawn
        procd_set_param watch network.interface
        [ -e /proc/sys/kernel/core_pattern ] && {
                procd_set_param limits core="unlimited"
        }
        procd_close_instance
}
```

Logs (no errors or debug-level logs strangely):
```
Sun Feb 11 17:42:59 2024 daemon.err netifd[1493]: device_apply_config(1188): Device 'br-lan': config applied
Sun Feb 11 17:42:59 2024 daemon.err netifd[1493]: interface_update(1402): Update interface 'loopback'
Sun Feb 11 17:42:59 2024 daemon.err netifd[1493]: interface_update(1402): Update interface 'lan'
Sun Feb 11 17:42:59 2024 daemon.err netifd[1493]: interface_add_dns_server(1435): Add IPv4 DNS server: 192.168.69.1
Sun Feb 11 17:42:59 2024 daemon.err netifd[1493]: interface_update(1402): Update interface 'wan'
Sun Feb 11 17:42:59 2024 daemon.err netifd[1493]: interface_update(1402): Update interface 'wan6'
Sun Feb 11 17:42:59 2024 daemon.err netifd[1493]: interface_update(1402): Update interface 'wg_securebit'
Sun Feb 11 17:42:59 2024 daemon.err netifd[1493]: wdev_update(754): Update wireless device 'radio0'
Sun Feb 11 17:42:59 2024 daemon.err netifd[1493]: wdev_update(754): Update wireless device 'radio1'
```

brada4, 2024-02-11:
You miss "config interface whatever" before first option setting proto 6in4.

brada4, 2024-02-11:
Please clean ia fuss and  report to a proper place:
https://openwrt.org/bugs

brada4, 2024-02-12:
Literarily it links to other tracker for base packages.

danog, 2024-02-12:
Done! :)

closed 2024-02-12 by danog