openwrt/packages #23370: openvpn: ovpn config doesn't come up with latest changes in master

dibdot, 2024-02-11:
Maintainer: @feckert 
Environment: mediatek/filelogic, GL.iNet GL-MT3000, OpenWrt SNAPSHOT r25115-7181eb9f81

Description:
For testing I've added one mullvad config via LuCI frontend, config as follows:

```
root@2go:/etc/init.d$ ls -la /etc/openvpn

drwxr-xr-x    1 root     root           304 Jan  1  1970 .
drwxr-xr-x    1 root     root          1912 Feb 11 18:34 ..
-rw-r--r--    1 root     root            19 Feb  2 21:40 mullvad_se.auth
-rw-r--r--    1 root     root          2634 Feb  2 21:47 mullvad_se.ovpn
```
```
root@2go:/etc/init.d$ cat /etc/config/openvpn

config openvpn 'mullvad'
	option enabled '1'
	option config '/etc/openvpn/mullvad_se.ovpn'
```

After router boot I found tons of these messages in the log:
```
Sun Feb 11 19:37:15 2024 daemon.err openvpn(mullvad_se)[7476]: Options error: In [CMD-LINE]:1: Error opening configuration file: mullvad_se.conf
Sun Feb 11 19:37:15 2024 daemon.warn openvpn(mullvad_se)[7476]: Use --help for more information.
Sun Feb 11 19:37:18 2024 user.info trm-vpn  [7320]: openvpn client connection enabled 'ovpn/mullvad_se'
Sun Feb 11 19:37:20 2024 daemon.err openvpn(mullvad_se)[7691]: Options error: In [CMD-LINE]:1: Error opening configuration file: mullvad_se.conf
Sun Feb 11 19:37:20 2024 daemon.warn openvpn(mullvad_se)[7691]: Use --help for more information.
[...]
```

For debugging I've added some debug prints to the openvpn init script:
```
	if [ -n "$instance" ]; then
		if [ "$instance_found" -gt 0 ]; then
			logger -t openvpn "DEBUG ::: START_UCI_INSTANCE" # DEBUG
			start_uci_instance "$instance"
		else
			logger -t openvpn "DEBUG ::: START_PATH_INSTANCE" # DEBUG
			start_path_instance "$instance"
		fi
	else
		logger -t openvpn "DEBUG ::: NO INSTANCE" # DEBUG
		config_foreach start_uci_instance 'openvpn'

		auto="$(uci_get openvpn globals autostart 1)"
		if [ "$auto" = "1" ]; then
			logger -t openvpn "DEBUG ::: START_PATH_INSTANCES" # DEBUG
			start_path_instances
[...]
```

Another router reboot prints the following messages:
```
root@2go:~# logread -e "openvpn"
Sun Feb 11 19:00:52 2024 user.notice openvpn: DEBUG ::: NO INSTANCE
Sun Feb 11 19:00:53 2024 user.notice openvpn: DEBUG ::: START_PATH_INSTANCES
Sun Feb 11 19:00:53 2024 daemon.notice openvpn(mullvad)[3328]: OpenVPN 2.6.8 aarch64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]

un Feb 11 19:35:13 2024 daemon.notice openvpn(mullvad)[3328]: /usr/libexec/openvpn-hotplug up mullvad tun0 1500 0 10.9.0.3 255.255.0.0 init
Sun Feb 11 19:35:13 2024 daemon.notice openvpn(mullvad)[3328]: net_route_v4_add: 185.65.135.83/32 via 10.168.30.1 dev [NULL] table 0 metric -1
Sun Feb 11 19:35:13 2024 daemon.notice openvpn(mullvad)[3328]: net_route_v4_add: 0.0.0.0/1 via 10.9.0.1 dev [NULL] table 0 metric -1
Sun Feb 11 19:35:13 2024 daemon.notice openvpn(mullvad)[3328]: net_route_v4_add: 128.0.0.0/1 via 10.9.0.1 dev [NULL] table 0 metric -1
Sun Feb 11 19:35:13 2024 daemon.notice openvpn(mullvad)[3328]: add_route_ipv6(::/2 -> fdda:d0d0:cafe:1195:: metric -1) dev tun0
Sun Feb 11 19:35:13 2024 daemon.notice openvpn(mullvad)[3328]: net_route_v6_add: ::/2 via :: dev tun0 table 0 metric -1
Sun Feb 11 19:35:13 2024 daemon.notice openvpn(mullvad)[3328]: add_route_ipv6(4000::/2 -> fdda:d0d0:cafe:1195:: metric -1) dev tun0
Sun Feb 11 19:35:13 2024 daemon.notice openvpn(mullvad)[3328]: net_route_v6_add: 4000::/2 via :: dev tun0 table 0 metric -1
Sun Feb 11 19:35:13 2024 daemon.notice openvpn(mullvad)[3328]: add_route_ipv6(8000::/2 -> fdda:d0d0:cafe:1195:: metric -1) dev tun0
Sun Feb 11 19:35:13 2024 daemon.notice openvpn(mullvad)[3328]: net_route_v6_add: 8000::/2 via :: dev tun0 table 0 metric -1
Sun Feb 11 19:35:13 2024 daemon.notice openvpn(mullvad)[3328]: add_route_ipv6(c000::/2 -> fdda:d0d0:cafe:1195:: metric -1) dev tun0
Sun Feb 11 19:35:13 2024 daemon.notice openvpn(mullvad)[3328]: net_route_v6_add: c000::/2 via :: dev tun0 table 0 metric -1
Sun Feb 11 19:35:13 2024 daemon.notice openvpn(mullvad)[3328]: Initialization Sequence Completed
Sun Feb 11 19:35:13 2024 daemon.notice openvpn(mullvad)[3328]: Data Channel: cipher 'AES-256-GCM', peer-id: 8
Sun Feb 11 19:35:13 2024 daemon.notice openvpn(mullvad)[3328]: Timers: ping 10, ping-restart 60
Sun Feb 11 19:35:13 2024 daemon.notice openvpn(mullvad)[3328]: Protocol options: protocol-flags cc-exit tls-ekm dyn-tls-crypt
Sun Feb 11 19:37:15 2024 user.notice openvpn: DEBUG ::: START_PATH_INSTANCE
Sun Feb 11 19:37:15 2024 daemon.err openvpn(mullvad_se)[7476]: Options error: In [CMD-LINE]:1: Error opening configuration file: mullvad_se.conf
Sun Feb 11 19:37:15 2024 daemon.warn openvpn(mullvad_se)[7476]: Use --help for more information.
Sun Feb 11 19:37:18 2024 user.info trm-vpn  [7320]: openvpn client connection enabled 'ovpn/mullvad_se'
Sun Feb 11 19:37:20 2024 daemon.err openvpn(mullvad_se)[7691]: Options error: In [CMD-LINE]:1: Error opening configuration file: mullvad_se.conf
Sun Feb 11 19:37:20 2024 daemon.warn openvpn(mullvad_se)[7691]: Use --help for more information.
```

As you can see, the init script finally try to start "START_PATH_INSTANCE", which looks wrong to me.
A restart of the init script works fine (only "START_PATH_INSTANCES" get started) and the openvpn connection comes up.

I'm too lazy to get the root cause of this issue, as I'm using wireguard usually.

Thanks for looking into this!

feckert, 2024-02-12:
@dibdot Sorry I could not reproduce this on my system!
I have used a native OpenVpn configuration file and placed it under `/etc/openvpn/` and named it `mullvad_se.ovpn`.
After that I added a OpenVpn UCI in `/etc/config/openvpn` configuration with the following content.
```
config openvpn 'mullvad'
	option enabled '1'
	option config '/etc/openvpn/mullvad_se.ovpn'
```

And the I restarted the system. Everything works as expected! The OpenVpn configuration gets connected on boot.

What is the output of the command `/etc/init.d/openvpn info`?

dibdot, 2024-02-12:
Faulty info after router reboot:
```
root@2go:~# /etc/init.d/openvpn info
{
	"openvpn": {
		"instances": {
			"mullvad": {
				"running": true,
				"pid": 3271,
				"command": [
					"/usr/sbin/openvpn",
					"--syslog",
					"openvpn(mullvad)",
					"--status",
					"/var/run/openvpn.mullvad.status",
					"--cd",
					"/etc/openvpn",
					"--config",
					"mullvad_se.ovpn",
					"--up",
					"/usr/libexec/openvpn-hotplug up mullvad",
					"--down",
					"/usr/libexec/openvpn-hotplug down mullvad",
					"--route-up",
					"/usr/libexec/openvpn-hotplug route-up mullvad",
					"--route-pre-down",
					"/usr/libexec/openvpn-hotplug route-pre-down mullvad",
					"--ipchange",
					"/usr/libexec/openvpn-hotplug ipchange mullvad",
					"--script-security",
					"2"
				],
				"term_timeout": 15,
				"respawn": {
					"threshold": 3600,
					"timeout": 5,
					"retry": -1
				}
			},
			"mullvad_se": {
				"running": false,
				"command": [
					"/usr/sbin/openvpn",
					"--syslog",
					"openvpn(mullvad_se)",
					"--status",
					"/var/run/openvpn.mullvad_se.status",
					"--cd",
					"/etc/openvpn",
					"--config",
					"mullvad_se.conf",
					"--up",
					"/usr/libexec/openvpn-hotplug up mullvad_se",
					"--down",
					"/usr/libexec/openvpn-hotplug down mullvad_se",
					"--route-up",
					"/usr/libexec/openvpn-hotplug route-up mullvad_se",
					"--route-pre-down",
					"/usr/libexec/openvpn-hotplug route-pre-down mullvad_se",
					"--script-security",
					"2"
				],
				"term_timeout": 15,
				"exit_code": 1,
				"respawn": {
					"threshold": 3600,
					"timeout": 5,
					"retry": -1
				}
			}
		},
		"triggers": [
			[
				"config.change",
				[
					"if",
					[
						"eq",
						"package",
						"openvpn"
					],
					[
						"run_script",
						"/etc/init.d/openvpn",
						"reload"
					]
				],
				1000
			]
		]
	}
}

```

Correct info after openvpn restart:
```
root@2go:~# /etc/init.d/openvpn info
{
	"openvpn": {
		"instances": {
			"mullvad": {
				"running": true,
				"pid": 7446,
				"command": [
					"/usr/sbin/openvpn",
					"--syslog",
					"openvpn(mullvad)",
					"--status",
					"/var/run/openvpn.mullvad.status",
					"--cd",
					"/etc/openvpn",
					"--config",
					"mullvad_se.ovpn",
					"--up",
					"/usr/libexec/openvpn-hotplug up mullvad",
					"--down",
					"/usr/libexec/openvpn-hotplug down mullvad",
					"--route-up",
					"/usr/libexec/openvpn-hotplug route-up mullvad",
					"--route-pre-down",
					"/usr/libexec/openvpn-hotplug route-pre-down mullvad",
					"--ipchange",
					"/usr/libexec/openvpn-hotplug ipchange mullvad",
					"--script-security",
					"2"
				],
				"term_timeout": 15,
				"respawn": {
					"threshold": 3600,
					"timeout": 5,
					"retry": -1
				}
			}
		},
		"triggers": [
			[
				"config.change",
				[
					"if",
					[
						"eq",
						"package",
						"openvpn"
					],
					[
						"run_script",
						"/etc/init.d/openvpn",
						"reload"
					]
				],
				1000
			]
		]
	}
}
```

feckert, 2024-02-13:
If I understand your output correctly, then the OpenVpn configuration is started twice.
* From the uci section `config openvpn 'mullvad'`
* From the native OpenVpn configuration file located under `//etc/openvpn/mullvad_se.ovpn`

The second one (native OpenVpn configuration) should not get started because int the function [start_path_instances](https://github.com/openwrt/packages/blob/master/net/openvpn/files/openvpn.init#L292)  there is a check that only [*.conf](https://github.com/openwrt/packages/blob/master/net/openvpn/files/openvpn.init#L226) files should start.
Your extension for the native OpenVpn configuration ist `*.ovpn`

So I do not know what exactly is happening one your system.

Could you please add the following uci section to your openvpn configuration.

```
config globals 'globals'
        option autostart '0'
```

This will not start any native OpenVpn configuration that are located unter `/etc/openvpn/*.conf` during boot and restart of the openvpn init script restart.

This is my output:
```
root@VR2-101845 ~ # /etc/init.d/openvpn info
{
        "openvpn": {
                "instances": {
                        "mullvad": {
                                "running": true,
                                "pid": 6856,
                                "command": [
                                        "/usr/sbin/openvpn",
                                        "--syslog",
                                        "openvpn(mullvad)",
                                        "--status",
                                        "/var/run/openvpn.mullvad.status",
                                        "--cd",
                                        "/etc/openvpn",
                                        "--config",
                                        "mullvad_se.ovpn",
                                        "--up",
                                        "/usr/libexec/openvpn-hotplug up mullvad",
                                        "--down",
                                        "/usr/libexec/openvpn-hotplug down mullvad",
                                        "--route-up",
                                        "/usr/libexec/openvpn-hotplug route-up mullvad",
                                        "--route-pre-down",
                                        "/usr/libexec/openvpn-hotplug route-pre-down mullvad",
                                        "--ipchange",
                                        "/usr/libexec/openvpn-hotplug ipchange mullvad",
                                        "--script-security",
                                        "2"
                                ],
                                "term_timeout": 15,
                                "respawn": {
                                        "threshold": 3600,
                                        "timeout": 5,
                                        "retry": -1
                                }
                        }
                },
                "triggers": [
                        [
                                "config.change",
                                [
                                        "if",
                                        [
                                                "eq",
                                                "package",
                                                "openvpn"
                                        ],
                                        [
                                                "run_script",
                                                "/etc/init.d/openvpn",
                                                "reload"
                                        ]
                                ],
                                1000
                        ]
                ]
        }
}
```

egc112, 2024-02-14:
Interesting, I have done the same (DL-WRX36 OpenWrt SNAPSHOT r24930-0c45d2cfc6)
I added my `mullvad_ro.ovpn`  and `mullvad_ro.auth` to /etc/openvpn and added to /etc/config/openvpn:
```
config openvpn 'mullvad_ro'
	option config '/etc/openvpn/mullvad_ro.ovpn'
	option enabled '1'
```
and I could also not reproduce it.

Perhaps check that there is no `*.conf` file in your /etc/openvpn ?

dibdot, 2024-02-16:
It turns out, that these errors are triggered by travelmate. I've sent a PR #23462 to fix these init errors.

feckert, 2024-02-20:
Fixed with commit https://github.com/openwrt/packages/commit/fe736b2f8c415e6d590931f1f1f504e1d18c579b

closed 2024-02-20 by feckert