openwrt/packages #23385: <ddns-scripts> Cloudflare DDNS Updater script

Grommish, 2024-02-12:
On the following Description, new information needs to be added https://github.com/openwrt/packages/blob/fa69fdc2b081abb57fc05c8d16356b16bdfdec5c/net/ddns-scripts/Makefile#L88

If possible, adding it to the `lUCI` configuration editing dialog to avoid having to edit the `/etc/config/ddns` file directly.

- The API need to be generated on the Cloudflare Dashboard at https://dash.cloudflare.com/profile/api-tokens` and should be a `Zone DNS` with `Edit` Abilities
- `Username` needs to be `Bearer`
- `Password` needs to be the created above Token
- In `/etc/config/ddns`, two options need to be added and set to the `Zone ID` as shown on the Domain Page (right side)
     - `option id`
     - `option zone-id`

brada4, 2024-02-12:
Why? API returns zone-id based on zone name.

Grommish, 2024-02-13:
From the link I provided:

[image]

It is required to be sent.  I couldn't get a successful update until I added both `option id` and `option zone-id` to the config file.

Perhaps something changed?  I'm not sure what to tell you.  The scripts grabs those `options` from the config to put in place of `__ID` and `__ZONE_ID` (I believe, I'm not looking at the script itself at the moment)

brada4, 2024-02-13:
You just describe what script is already doing.
 >      # extract zone id

There is no point in posting more random links.

Grommish, 2024-02-13:
I'm not asking for a script update, I'm asking for a _description_ update
in the Makefile because there is no mention of what needs to be manually
altered and cannot be set via the luci interface.  The script can (and
does) extract the value, as long as it's actually there in the config to be
pulled.  This is a documentation issue, which is why the OP is for the line
it is, in the file it is.

brada4, 2024-02-13:
id is email and zone-id is queried from API. You do not need extra parameters.