openwrt/packages #23395: Modemmanager since openwrt snapshot r25132 is not working anymore

5gfreak, 2024-02-13:
Modemmanager 1.22

there are problems with the modemmanager 1.22 on the new snapshots on device Z8102AX 

Tue Feb 13 06:38:32 2024 daemon.err ModemManager[5662]: hotplug: couldn't report initial kernel events: ModemManager not running
Tue Feb 13 06:38:32 2024 daemon.info ModemManager[5670]: hotplug: Checking if ModemManager is available...
Tue Feb 13 06:38:32 2024 daemon.info ModemManager[5670]: hotplug: ModemManager not yet available
Tue Feb 13 06:38:32 2024 daemon.err ModemManager[5670]: hotplug: couldn't report initial kernel events: ModemManager not running
Tue Feb 13 06:38:37 2024 daemon.notice [6409]: <msg> ModemManager (version 1.22.0) starting in system bus...
Tue Feb 13 06:38:37 2024 daemon.warn [6409]: <wrn> could not acquire the 'org.freedesktop.ModemManager1' service name
Tue Feb 13 06:38:37 2024 daemon.notice [6409]: <msg> ModemManager is shut down

feckert, 2024-02-13:
Thanks for reporting.
Yes I can confirm this.
```
Tue Feb 13 08:43:48 2024 daemon.info ModemManager[9985]: hotplug: ModemManager not yet available                                      
Tue Feb 13 08:43:48 2024 daemon.info ModemManager[9984]: hotplug: checking if ModemManager is available...
```
There were no changes recently in the ModemManager packages.
This could be related to the glib2 [update](https://github.com/openwrt/packages/commit/08c7b0dfcae48114176762e93aa1b4ce5d42f8ad) 4 days ago.

5gfreak, 2024-02-13:
Thanks for replying :-)

Could you please forward this issue to the right guy?

Beste Grüsse

suyuan168, 2024-02-15:
Same problem ~ looking forward to solving it.

feckert, 2024-02-23:
Fixed by reverting glib2 bump [back to 2.74.0](https://github.com/openwrt/packages/commit/7b33b1503208dbf00f10f748252ddbc393082970)

closed 2024-02-23 by feckert