openwrt/packages #23399: rpcsvc-proto: compile failure

oskarirauta, 2024-02-13:
Maintainer: @Andy2244 @neheb @ffontaine 
Environment:
x86_64
latest git

Description:

Would seem like a issue with musl maybe?
Adding `#define stat64 stat` makes it build without problem.

build log:
```
Format code blocks by wrapping them with pairs of ```
/usr/src/openwrt/staging_dir/host/bin/gcc -DHAVE_CONFIG_H -I. -I..   -I/usr/src/openwrt/staging_dir/host/include -I/usr/src/openwrt/staging_dir/hostpkg/include -I/usr/src/openwrt/staging_dir/target-x86_64_musl/host/include  -O2 -I/usr/src/openwrt/staging_dir/host/include -I/usr/src/openwrt/staging_dir/hostpkg/include -I/usr/src/openwrt/staging_dir/target-x86_64_musl/host/include -c -o rpc_main.o rpc_main.c
cc1: note: someone does not honour COPTS correctly, passed 0 times
rpc_main.c: In function 'find_cpp':
rpc_main.c:340:17: error: storage size of 'buf' isn't known
  340 |   struct stat64 buf;
      |                 ^~~
rpc_main.c:342:7: warning: implicit declaration of function 'stat64'; did you mean 'stat'? [-Wimplicit-function-declaration]
  342 |   if (stat64 (CPP, &buf) == 0)
      |       ^~~~~~
      |       stat
rpc_main.c: In function 'checkfiles':
rpc_main.c:1128:17: error: storage size of 'buf' isn't known
 1128 |   struct stat64 buf;
      |                 ^~~
make[6]: *** [Makefile:438: rpc_main.o] Error 1
make[6]: Leaving directory '/usr/src/openwrt/build_dir/hostpkg/rpcsvc-proto-1.4.3/rpcgen'
make[5]: *** [Makefile:389: all-recursive] Error 1
make[5]: Leaving directory '/usr/src/openwrt/build_dir/hostpkg/rpcsvc-proto-1.4.3'
make[4]: *** [Makefile:330: all] Error 2
make[4]: Leaving directory '/usr/src/openwrt/build_dir/hostpkg/rpcsvc-proto-1.4.3'
make[3]: *** [Makefile:51: /usr/src/openwrt/build_dir/hostpkg/rpcsvc-proto-1.4.3/.built] Error 2
make[3]: Leaving directory '/usr/src/openwrt/feeds/packages/libs/rpcsvc-proto'
time: package/feeds/packages/rpcsvc-proto/host-compile#4.67#0.60#6.25
    ERROR: package/feeds/packages/rpcsvc-proto [host] failed to build.
make[2]: *** [package/Makefile:126: package/feeds/packages/rpcsvc-proto/host/compile] Error 1
make[2]: Leaving directory '/usr/src/openwrt'
make[1]: *** [package/Makefile:122: /usr/src/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/usr/src/openwrt'
make: *** [/usr/src/openwrt/include/toplevel.mk:232: world] Error 2
```

neheb, 2024-02-13:
musl had a change where stat64 needs LARGEFILE64 defined or something like that.

oskarirauta, 2024-02-14:
That would probably be better solution instead of patching it :) Had forgotten all about it, since I've been pretty occupied elsewhere lately.

oskarirauta, 2024-02-15:
It's often used to check that if building system with musl instead of glibc, then apply -DLARGEFILE64_SOURCE to cflags; problem is, that it won't work this time- for me at least, since my build host is also using musl and this package is built for both, host and target :/ so I'd need to apply it to host_cflags also, but there's no way to detect host musl (actually there exists a working script, but it's not included with default build system)

even though musl attempts to mimic standards; they should had made something like `#ifdef __MUSL` exist..