openwrt/packages #23410: <watchcat v1-17> - Network interface reset doesn't work as expected on a Wireguard VPN interface

ako82, 2024-02-14:
Maintainer: @kofec , @jtkohl
Environment:

Modell Raspberry Pi 3 Model B Rev 1.2
Architektur ARMv8 Processor rev 4
Platform bcm27xx/bcm2710
Firmware-Version OpenWrt 23.05.2 r23630-842932a63d / LuCI openwrt-23.05 branch git-24.006.68745-9128656
Kernel-Version 5.15.137

Description:

These two commands to bring down and back up an interface don't work properly on a Wireguard VPN interface. The link status of the interface changes to DOWN, but Wireguard interfaces seem to be missing the UP status. When they're actually up, the link status always is UNKNOWN.
```
watchcat_restart_network_iface() {
	logger -p daemon.info -t "watchcat[$$]" "Restarting network interface: \"$1\"."
	ip link set "$1" down
	ip link set "$1" up
}
```
After changing the commands in the watchcat.sh script to this, my Wireguard VPN interface gets properly restarted.
```
watchcat_restart_network_iface() {
        logger -p daemon.info -t "watchcat[$$]" "Restarting network interface: \"$1\"."
        ifdown "$1"
        ifup "$1"
}
```

jtkohl, 2024-02-17:
`ifup` does both a down and an up itself, so you only really need the `ifup` line in that function.
Looks good to me...but I don't have commit privileges.  May I suggest you create a pull request for this and then we can get someone to formally review/approve/commit?

franganghi, 2024-10-29:
I have the same problem and following your advice, i managed to make it work but it can really be misleading for a person who's dealing with it for the first time. Also, watchcat logs are not that explicit with regards of what is going on in the background.

Did you create that pull request?

lrussell887, 2025-05-27:
Yes, I can also confirm that swapping `ip link set "$1" down; ip link set "$1" up` with just `ifup "$1"` in the `watchcat.sh` script allows my WireGuard tunnel to restart properly. Doing this and using LuCI to configure Watchcat is the least complex way I've found to make sure my tunnel comes up after a reboot and stays working. Startup of my tunnel can be finicky for some reason even knowing the clock is set correctly, so this helps.

rondoval, 2025-08-15:
it also somehow removes IPv6 addresses from the Wireguard interface. Implementing the patch proposed resolves the issue.