openwrt/packages #23413: mariadb missing libfmt dependency

oskarirauta, 2024-02-14:
I came up with a strange issue with mariadb, build failed on the beginning due to missing libfmt. I checked and even re-built my libfmt, and still it wasn't able to find it. To solve the issue, I made it to use bundled libfmt.

Not sure, but might be related to this: https://jira.mariadb.org/browse/MDEV-27965

Maintainer: @mhei @neheb
Environment: x86_64, latest git

oskarirauta, 2024-02-14:
well, that fixed it on my end; but don't we want to utilise system's libfmt instead? Can't help with that, since I didn't see clear reason why it didn't work, but current package tries to use libfmt from system.

neheb, 2024-02-14:
I started failing after libfmt was updated. seems mariadb doesn't work with newer fmt.

oskarirauta, 2024-02-15:
Okay.. Maybe I'll then PR it, unless you already did..

krant, 2024-02-15:
It seems https://github.com/fmtlib/fmt/issues/3835 is the reason

mhei, 2024-02-24:
AS noted, I think we can close this since #23380 was merged. Please re-open if issue still persists.

closed 2024-02-24 by mhei