openwrt/packages #23417: travelmate: Fails in environments with lots of APs

hackerb9, 2024-02-14:
# Issue template (remove lines from top till here)

Maintainer: @dibdot
Environment: (put here arch, model, OpenWrt version)
```bash
# cat /etc/openwrt_release  
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.3'
DISTRIB_REVISION='r20028-43d71ad93e'
DISTRIB_TARGET='ath79/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt 22.03.3 r20028-43d71ad93e'
DISTRIB_TAINTS=''
```

Description:

Travelmate uses `trm_maxscan` to limit the results of a scan to the strongest Access Points with no way to disable it. 

This breaks where I am, in a dense urban environment where there are dozens and dozens of strong AP signals but the top 30 are not relevant. This causes Travelmate to never connect to the correct AP; it just shows the spinning icon as if it is scanning. 

The fix is easy. Allow the user to set maxscan to 0 and use that to mean "unlimited". The code that handles maxscan in /usr/bin/travelmate looks like this:

```bash
... scanlist | sort -rn | head -qn "${trm_maxscan}"
```

Therefore, simply setting `trm_maxscan=-0` (negative zero) will cause head to output the entire list. That can be done by adding these three lines anywhere after `config_load travelmate` — I suggest putting them right before `config_load wireless`:

```bash
	if [ "${trm_maxscan}" -lt 1 ]; then
		trm_maxscan=-0				# Negative zero means unlimited
	fi
```

____

_Sidenote: I do not know why trm_maxscan exists since Travelmate already has a minimum signal quality threshold. It seems like an unnecessary optimization that can only cause mysterious problems in the future as WiFi APs become more and more common. I think the best course is to remove the limit completely._

brada4, 2024-02-15:
sorting uses 2x memory before trimming list... i doubt is optimisation, you can safely remove head cmd in that place.
If you mean to sort by "goodness" you might want to parse 40/70 looking occupancy info element too.

dibdot, 2024-02-16:
Thanks for your input. This branch is no longer supported by me and it's only a LuCI frontend limitation. Just raise the trm_maxscan value in the uci file directly. Maybe I'll remove this option with the next update.

closed 2024-02-16 by dibdot