openwrt/packages #23460: php8: build failure, apr-1-config searched from wrong place

oskarirauta, 2024-02-16:
While attempting to build php8 I get a failure due to apr-1-config not found; I do have apr-1-config-
it is in staging_dir/target-x86_64_musl/usr/bin but php8 configure is trying to find it from staging_dir/target-x86_64_musl - from root of target staging_dir. Easy to quick fix, I just linked _apr-1-config_ to there and all good..
But something definetly fails there.. might be that usr/bin/apxs is culprit, it's executed right before this issue, should it get patch to apr-1-config from that?

This is my php8 config:
```
#
# PHP8
#
CONFIG_PACKAGE_php8=m
CONFIG_PHP8_LIBXML=y
CONFIG_PHP8_SYSTEMTZDATA=y
CONFIG_PACKAGE_php8-cgi=m
CONFIG_PACKAGE_php8-cli=m
CONFIG_PACKAGE_php8-fastcgi=m
CONFIG_PACKAGE_php8-fpm=m
CONFIG_PACKAGE_php8-mod-bcmath=m
CONFIG_PACKAGE_php8-mod-calendar=m
CONFIG_PACKAGE_php8-mod-ctype=m
CONFIG_PACKAGE_php8-mod-curl=m
CONFIG_PACKAGE_php8-mod-dom=m
CONFIG_PACKAGE_php8-mod-exif=m
CONFIG_PACKAGE_php8-mod-fileinfo=m
CONFIG_PACKAGE_php8-mod-filter=m
CONFIG_PACKAGE_php8-mod-ftp=m
CONFIG_PACKAGE_php8-mod-gd=m
CONFIG_PACKAGE_php8-mod-gettext=m
CONFIG_PACKAGE_php8-mod-gmp=m
CONFIG_PACKAGE_php8-mod-iconv=m
CONFIG_PACKAGE_php8-mod-imap=m
CONFIG_PACKAGE_php8-mod-intl=m
CONFIG_PHP8_FULLICUDATA=y
CONFIG_PACKAGE_php8-mod-ldap=m
CONFIG_PACKAGE_php8-mod-mbstring=m
CONFIG_PACKAGE_php8-mod-mysqli=m
CONFIG_PACKAGE_php8-mod-mysqlnd=m
CONFIG_PACKAGE_php8-mod-opcache=m
CONFIG_PACKAGE_php8-mod-openssl=m
CONFIG_PACKAGE_php8-mod-pcntl=m
CONFIG_PACKAGE_php8-mod-pdo=m
CONFIG_PACKAGE_php8-mod-pdo-mysql=m
CONFIG_PACKAGE_php8-mod-pdo-pgsql=m
CONFIG_PACKAGE_php8-mod-pdo-sqlite=m
CONFIG_PACKAGE_php8-mod-pgsql=m
CONFIG_PACKAGE_php8-mod-phar=m
CONFIG_PACKAGE_php8-mod-session=m
CONFIG_PACKAGE_php8-mod-shmop=m
CONFIG_PACKAGE_php8-mod-simplexml=m
CONFIG_PACKAGE_php8-mod-snmp=m
CONFIG_PACKAGE_php8-mod-soap=m
CONFIG_PACKAGE_php8-mod-sockets=m
CONFIG_PACKAGE_php8-mod-sodium=m
CONFIG_PACKAGE_php8-mod-sqlite3=m
CONFIG_PACKAGE_php8-mod-sysvmsg=m
CONFIG_PACKAGE_php8-mod-sysvsem=m
CONFIG_PACKAGE_php8-mod-sysvshm=m
CONFIG_PACKAGE_php8-mod-tokenizer=m
CONFIG_PACKAGE_php8-mod-xml=m
CONFIG_PACKAGE_php8-mod-xmlreader=m
CONFIG_PACKAGE_php8-mod-xmlwriter=m
CONFIG_PACKAGE_php8-mod-zip=m
CONFIG_PACKAGE_php8-pecl-dio=m
CONFIG_PACKAGE_php8-pecl-http=m
CONFIG_PACKAGE_php8-pecl-imagick=m
CONFIG_PACKAGE_php8-pecl-krb5=m
CONFIG_PACKAGE_php8-pecl-mcrypt=m
CONFIG_PACKAGE_php8-pecl-raphf=m
CONFIG_PACKAGE_php8-pecl-redis=m
CONFIG_PACKAGE_php8-pecl-xdebug=m
# end of PHP8
```

Maintainer: @mhei 
Environment: x86_64, recent git

brada4, 2024-02-16:
Add apr here https://github.com/openwrt/packages/blob/dc888dea24f8c218796c9365bd6bfda297a2ae78/lang/php8/Makefile#L155
or better unselect modphp and go with fpm.

mhei, 2024-02-16:
This looks like a regression introduced with the latest apr update. Maybe you can try the patch in the referenced PR.

PR #23463 (2024-02-19): apr/subversion: fix subversion build and apache-mod-php8 build regres…
…sion (fixes #23460)

The recent upgrade of apr included a change with should fix the subversion build.

Unfortunately, this fix resulted in a build regression of apache-mod-php8.

The new approach is to pass the locations of the apr config helpers to configure via parameter.

Fixes: 68dd7b7cf632 ("apr: update to 1.7.4")

Maintainer: @heil
Compile tested: x86-64
Run tested: -
commit 84ac0263 (2024-02-19): apr/subversion: fix subversion build and apache-mod-php8 build regression (fixes #23460)

The recent upgrade of apr included a change with should fix the subversion build.

Unfortunately, this fix resulted in a build regression of apache-mod-php8.

The new approach is to pass the locations of the apr config helpers
to configure via parameter.

Fixes: 68dd7b7cf632 ("apr: update to 1.7.4")
closed 2024-02-19 by mhei