openwrt/packages #23469: https-dns-proxy: Multiple issues with running configuration and backup configuration

Shine-, 2024-02-17:
Maintainer: @stangri
Environment: at least 23.05.2, likely all targets, tested with multiple ath79 and mt7621 devices
Package version: 2023-10-25-1 and 2023-10-25-5

# Description:

I started playing with this package recently out of interest, and noticed that it's not quite working properly. It somehow fails to configure the dnsmasq settings correctly. The first config after installation is always fine. The problems start e.g. after reboot (-> dnsmasq settings lost) or after multiple changes e.g. using LuCI (-> backup config overwritten with running config).

Let me explain in more detail.

All these examples are done off a fresh, vanilla 23.05.2 or recent 23.05-SNAPSHOT setup, no special config, only `curl` and `https-dns-proxy` installed on top.

# Issue 1: dnsmasq config lost after reboot

This one is easy to reproduce. Install on 23.05.2 using `opkg` or LuCI, don't change any settings, verify that the config got set up properly, e.g.:
```
# uci show dhcp
[...]
dhcp.@dnsmasq[0].doh_backup_noresolv='-1'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].doh_backup_server=''
dhcp.@dnsmasq[0].server='/mask.icloud.com/' '/mask-h2.icloud.com/' '/use-application-dns.net/' '127.0.0.1#5053' '127.0.0.1#5054'
dhcp.@dnsmasq[0].doh_server='127.0.0.1#5053' '127.0.0.1#5054'
[...]
```
Reboot the device. After reboot, check whether `https-dns-proxy` is running (e.g. via GUI or `service https-dns-proxy status`), then check config again - it's gone, ie. back to pre-package-installation config:
```
# service https-dns-proxy status
running
# uci show dhcp
[...] <--(none of the above config properties exists)
```
Restart service and check again, now it's there:
```
# service https-dns-proxy restart
Starting https-dns-proxy 2023-10-25-1 instances ✓✓
Updating dnsmasq config ✓
Restarting dnsmasq on_config_update ✓
# uci show dhcp
[...]
dhcp.@dnsmasq[0].doh_backup_noresolv='-1'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].doh_backup_server=''
dhcp.@dnsmasq[0].server='/mask.icloud.com/' '/mask-h2.icloud.com/' '/use-application-dns.net/' '127.0.0.1#5053' '127.0.0.1#5054'
dhcp.@dnsmasq[0].doh_server='127.0.0.1#5053' '127.0.0.1#5054'
[...]
```

# Issue 2: `dnsmasq` backup configuration contains running configuration after some config changes

This takes some configuration changes to reproduce, but it will happen eventually. For example, try adding a third DoH provider using LuCI, and change its settings a couple of times, press `Apply & Save` after each change. It won't take long, maybe 2 or 3 changes, and you'll see something like the following:
```
# uci show dhcp
[...]
dhcp.@dnsmasq[0].doh_backup_noresolv='-1'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].doh_backup_server='' '/mask.icloud.com/' '/mask-h2.icloud.com/' '/use-application-dns.net/' '127.0.0.1#5053' '127.0.0.1#5054'
dhcp.@dnsmasq[0].server='/mask.icloud.com/' '/mask-h2.icloud.com/' '/use-application-dns.net/' '127.0.0.1#5053' '127.0.0.1#5054' '127.0.0.1#5055'
dhcp.@dnsmasq[0].doh_server='127.0.0.1#5053' '127.0.0.1#5054' '127.0.0.1#5055'
[...]
```
Note how the `.doh_backup_server` attribute (which should be just '') will be appended with the current running configuration. This also happens if the original `server` attribute already contained one or more forwarders.

# Issue 3: The package will mess with `dnsmasq` saved config, even without any settings change, at every service start/stop

Since most devices running OpenWrt are flash based, this will add to the flash wear, ie. reduce lifetime, which is **bad-bad-bad**. The package should not, ever, alter the saved `dnsmasq` config. It should do its job using the running config, which is kept in RAM (`/tmp/run`), not in flash.

Also, _one_ package (https-dns-proxy) should never add new attributes into _another_ package's config (dnsmasq), e.g. `dhcp.@dnsmasq[0].doh_[...]`. This is bad coding for one. And it's prone to settings loss, when the original package (ie. `dnsmasq`) or for example LuCI cleans invalid settings off the config.

`https-dns-proxy` should save its non-volatile settings **only** to its **own** UCI config. All volatile settings (like whatever the package is doing to the dnsmasq running config) should only be altered in RAM.

stangri, 2024-02-18:
1. Can't reproduce.
2. Please include exact steps to reproduce.
3. Incorrect, you'd benefit from checking out the README.
4. I don't feel it warrants crowding WebUI.

Shine-, 2024-02-18:
1. Can reproduce every time.
2. Steps already given, can also be reproduced quite reliably.
3. It **is** true, you can check file time stamps in the overlay file system - /etc/config/dhcp **does** get written at every service start/stop (when the `dnsmasq_config_update` option is set, which is set by default); the rest is also true.
4. Just a suggestion, but it's your package after all.

I was just trying to help by opening this issue, you can use it or ignore it. I'm leaving it at this.
Cheers and bye.

NorthP8, 2024-02-18:
I have the exact same issue. After rebooting the router https-dns-proxy does not work. After service restart https-dns-proxy works again.
But I have suspicions that the problem is in the OpenWrt itself.
"luci-app-wifischedule" also not fully functional.
These problems appeared after 23.05.2 installation.
Tp-Link Archer c6u (mt7621)
So the problem actually exists.

stangri, 2024-02-18:
> I have the exact same issue. After rebooting the router https-dns-proxy does not work. 

Not the same issue, @Shine- claims that dnsmasq servers and noresolv settings are wiped on reboot. You may want to create a separate issue if https-dns-proxy doesn't start on boot, as I will need some logs to try to address that.

stangri, 2024-02-18:
Given that I can't act on `This takes some configuration changes to reproduce, but it will happen eventually.`, closing this.

closed 2024-02-18 by stangri

teohhanhui, 2024-07-23:
FWIW, I'm experiencing something similar to issue (2) mentioned by OP.

`https-dns-proxy` would correctly change the DNS forwards list to only `127.0.0.1#5053` on start. But after some time, DNS resolving would stop working, and I'd observe `127.0.0.1#5054` in addition to `127.0.0.1#5053` in the DNS forwards list.

I'm not sure about the exact steps to reproduce, but it does happen reliably every day. The fix is always for me to remove all DNS forwards, and restart `https-dns-proxy`. (Just restarting `https-dns-proxy` doesn't fix the problem, as it'd still leave the incorrect forwards in the list...)

stangri, 2024-07-23:
> FWIW, I'm experiencing something similar to issue (2) mentioned by OP.
>
> `https-dns-proxy` would correctly change the DNS forwards list to only `127.0.0.1#5053` on start. But after some time, DNS [...]

I'd advise you create a new issue with details, including your `https-dns-proxy` config, as well as `dhcp` config before and after you experience issues and if anything has been done to your OpenWrt leading to the issue.

Shine-, 2024-07-24:
@teohhanhui and others who may experience similar issues.

I almost forgot about this, since I stopped using the package soon after experiencing the author's not very friendly replies here.

After getting tired how this package constantly messed with my saved configuration, even when I told it to not touch the `dnsmasq` config, here's how I worked around it:

1. To help with preventing my Issue 2+3 from happening, set `https-dns-proxy.config.dnsmasq_config_update='-'` and never use the LuCI GUI to change any configuration - only use `uci`, and only while the `https-dns-proxy` service is stopped. Always make sure that `https-dns-proxy.config.dnsmasq_config_update` is set to "-" **before** restarting the service (the setting might get randomly reset to "*", similar to my Issue 1...).
2. To prevent Issue 1 from happening (or rather, to auto-fix it when it happens, because it **will** happen eventually!), create a script script that checks:
- the setting from (1): `https-dns-proxy.config.dnsmasq_config_update` (should be "-")
- your desired `dhcp.@dnsmasq[0].server` configuration
- the existence of `dhcp.@dnsmasq[0].doh_backup_noresolv` and `dhcp.@dnsmasq[0].doh_backup_server` (neither should exist)

This, after some trial and error, worked for me as a well-enough indication whether the package **again** messed with my configuration.

3. If **any** of the above didn't match:
- Stop the service `https-dns-proxy`
- Remove `dhcp.@dnsmasq[0].doh_backup_*`
- Recreate from scratch the full configuration that this package might have messed with (ie. `dhcp.@dnsmasq[0].noresolv`, `dhcp.@dnsmasq[0].server` and `https-dns-proxy.*` in my case)
- Commit the dhcp configuration and the restart both `https-dns-proxy` and `dnsmasq` (in this order).
4. call your script from `rc.local`, but **with a 1-2 minute delay**, otherwise it won't work reliably.

If you're not worried about flash wear, you could also skip step (2) and directly proceed with (3) everytime the script is called.

In my case, the package broke my `dnsmasq` configuration only after a reboot, sysupgrade or after changing `https-dns-proxy` settings in LuCI. So my approach worked for me. If I understand correctly, @teohhanhui experiences this also during runtime, without changing any settings, restarting services or rebooting? In this case, my approach won't help, I suppose...

I'm aware that you'll need some shell scripting skills to follow this approach. Unfortunately, I haven't saved my script after I stopped using this package, so I can't provide a ready-to-use solution at this time :-(

As you can see, it's quite some work to constantly fix the erratic behavior of this package. Which is why I stopped using it in the end.