openwrt/packages #23470: adblock-fast: unnecessary force-refreshes when travelmate switches networks

ckuethe, 2024-02-17:
Maintainer: @stangri 
Environment:
```
Model: GL-MT300N-V2
Architecture: MediaTek MT7628AN ver:1 eco:2
Target Platform: ramips/mt76x8
Firmware Version: OpenWrt 23.05.0 r23497-6637af95aa / LuCI openwrt-23.05 branch git-24.040.70644-45b9e79
Kernel Version: 5.15.134
```

And since there are userland packages involved:
```
# opkg list-installed | grep -E "travelmate|adblock"
adblock-fast - 1.1.1-3
luci-app-adblock-fast - 1.1.1-3
luci-app-travelmate - git-24.006.72299-11a1a43
travelmate - 2.1.2-2
```

Description:

I've experience this a lot during my last trip somewhere, but I just happened to have logread running to catch it this time. In https://github.com/openwrt/packages/issues/20207 @dibdot said that travelmate wasn't explicitly forcing adblock or unbound to reload and that they should should somehow ignore travelmate's `ifup` events.

Anyhoo, I've got adblock-fast set to use a compressed cache because it takes about 5 minutes to download and process the ad lists to my liking; I try to use the cache as long as absolutely possible. It seems like the cache usability check isn't doing the right thing.

```
Fri Feb 16 05:55:35 2024 daemon.warn odhcpd[1425]: No default route present, overriding ra_lifetime!
Fri Feb 16 05:59:33 2024 daemon.notice netifd: travelmate (7591): udhcpc: sending renew to server 10.27.88.201
Fri Feb 16 05:59:33 2024 daemon.notice netifd: travelmate (7591): udhcpc: lease of 172.31.129.117 obtained from 10.27.88.201, lease time 14400
Fri Feb 16 05:59:33 2024 user.notice firewall: Reloading firewall due to ifupdate of travelmate (phy0-sta0)
Fri Feb 16 05:59:40 2024 user.notice adblock-fast [19855]: Force-reloading adblock-fast 1.1.1-3...
Fri Feb 16 05:59:53 2024 user.notice adblock-fast [19855]: [DL] Blocked File: gist.githubusercontent.com (domains) [✓]
Fri Feb 16 05:59:59 2024 user.notice adblock-fast [19855]: [DL] Blocked File: raw.githubusercontent.com (domains) [✓]
Fri Feb 16 05:59:59 2024 user.notice adblock-fast [19855]: [DL] Blocked File: gist.githubusercontent.com (domains) [✓]
Fri Feb 16 05:59:59 2024 user.notice adblock-fast [19855]: [DL] Blocked File: v.firebog.net (domains) [✓]
Fri Feb 16 05:59:59 2024 user.notice adblock-fast [19855]: [DL] Blocked File: raw.githubusercontent.com (domains) [✓]
Fri Feb 16 06:00:11 2024 user.notice adblock-fast [19855]: [DL] Blocked File: sysctl.org (hosts) [✓]
Fri Feb 16 06:00:18 2024 user.notice adblock-fast [19855]: [DL] Blocked File: raw.githubusercontent.com (domains) [✓]
Fri Feb 16 06:00:19 2024 user.notice adblock-fast [19855]: [DL] Blocked File: raw.githubusercontent.com (hosts) [✓]
Fri Feb 16 06:00:46 2024 user.notice adblock-fast [19855]: [DL] Blocked File: cdn.jsdelivr.net (domains) [✓]
Fri Feb 16 06:00:59 2024 user.notice ucitrack: Setting up /etc/config/dhcp reload dependency on /etc/config/network
Fri Feb 16 06:00:59 2024 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/wireless
...
```

Unfortunately my adblock-fast debug log is about 53MB and I haven't yet found exactly what's going wrong

ckuethe, 2024-02-17:
full logs available on request

stangri, 2024-02-18:
Please update to 1.1.1-5 and re-test.

ckuethe, 2024-02-19:
Alas, still seeing this

```
# opkg list-installed | grep -E "travelmate|adblock-fast"
adblock-fast - 1.1.1-5
luci-app-adblock-fast - 1.1.1-5
luci-app-travelmate - git-24.006.72299-11a1a43
travelmate - 2.1.2-2
```

No surprise to see that the lists failed to download, because I hadn't yet logged into the captive portal. Still, my compressed cache got trashed.

```
Service Status
Version 1.1.1-5 - Active.
Blocking 486 domains (with dnsmasq.conf).
Compressed cache file created.
Force DNS ports: 53 853.

Service Errors
Failed to download https://gist.githubusercontent.com/ckuethe/15a532850744682fb0c2a668bee0b07f/raw/intercrap-block-list.txt!
Failed to download https://raw.githubusercontent.com/AdguardTeam/cname-trackers/master/data/combined_disguised_trackers_justdomains.txt!
Failed to download https://cdn.jsdelivr.net/gh/bongochong/CombinedPrivacyBlockLists/NoFormatting/cpbl-ctld.txt!
Failed to download http://sysctl.org/cameleon/hosts!
Failed to download https://raw.githubusercontent.com/AdguardTeam/cname-trackers/master/data/combined_disguised_clickthroughs_justdomains.txt!
Failed to download https://gist.githubusercontent.com/ckuethe/15a532850744682fb0c2a668bee0b07f/raw/!
Failed to download https://raw.githubusercontent.com/craiu/mobiletrackers/master/list.txt!
Failed to download https://v.firebog.net/hosts/Admiral.txt!
Failed to download https://raw.githubusercontent.com/jkrejcha/AdmiraList/master/AdmiraList.txt!
Failed to process allow-list!
Errors encountered, please check the [README](https://docs.openwrt.melmac.net/adblock-fast/)!
```

I copied my backup of my compressed cache and triggered a restart - not a reload - and adblock still tried to download again.

```
root@computer:/etc# /etc/init.d/adblock-fast restart
Stopping adblock-fast 1.1.1-5... [✓]
Force-reloading adblock-fast 1.1.1-5...
[DL] Blocked File: gist.githubusercontent.com (domains) [✓]
[DL] Blocked File: raw.githubusercontent.com (domains) [✓]
[DL] Blocked File: gist.githubusercontent.com (domains) [✓]
[DL] Blocked File: raw.githubusercontent.com (domains) [✓]
[DL] Blocked File: v.firebog.net (domains) [✓]
[DL] Blocked File: sysctl.org (hosts) [✓]
[DL] Blocked File: raw.githubusercontent.com (domains) [✓]
[DL] Blocked File: raw.githubusercontent.com (hosts) [✓]
[DL] Blocked File: cdn.jsdelivr.net (domains) [✓]
...
```

stangri, 2024-02-19:
@ckuethe thank you for additional testing. if you can run the "compressed cache trashing scenario" again and capture the following before running the `service adblock-fast restart`, I can try to replicate:
1. Location/filename of compressed cache file
2. content of `/etc/config/adblock-fast`
3. output of `service adblock-fast info`
4. output of `cat /dev/shm/adblock-fast.*`

Thanks!

ckuethe, 2024-02-19:
Logs sent via email

stangri, 2024-02-20:
What's up with `"adblock-fast.orig"` from the service info call?

ckuethe, 2024-02-20:
I'm trying to fix it, so I've got adblock-fast.orig and adblock-fast.ckuethe so I'm flipping a symlink back and forth to use either the unmodified version or my patched version

stangri, 2024-02-20:
> I'm trying to fix it, so I've got adblock-fast.orig and adblock-fast.ckuethe so I'm flipping a symlink back and forth to use either the unmodified version or my patched version

Don't do that, some parts may depend on the information from ubus under the proper project name.

Regarding removal of compressed cache -- if the previous start of the service led to any errors (reported in the /dev/shm/adblock-fast.* files) then upon next start the force-redownload is initiated. The logic is that even if the compressed cache was created on a previous start, it cannot be trusted to be complete if there were errors discovered. Also, I did not expect manual backup/restoration of the compressed cache by the user.

I believe that if you delete the `/dev/shm/adblock-fast.*` files when copying the compressed cache to the router, the service should pick up the compressed cache on the next start.

If you expect to have the service started/restarted when you may not have connectivity, I'd definitely try to keep the compressed cache on the router so it's used and only trigger force-download manually to update the compressed cache.

ckuethe, 2024-02-20:
I definitely expect to be (re)?starting without proper connectivity. My typical use case is to power up my travel router when I'm in hotels, airports, and airplanes - all situations where a captive portal gets in my way, possibly making me re-authenticate as I roam between access points.

My compressed cache is at least 10kB and has at least 10000 entries. A size check would be a good indication that the download is valid. And if the cache shrinks by some amount, that's probably a download failure. 

I do keep the cache on the router as `/etc/adblock-fast.dnsmasq.conf.gz` (generated by adblock-fast) and `/etc/adblock-fast.dnsmasq.conf.gz.bak` (my on device backup). I'm just not sure why adblock-fast didn't see I had a valid cache on disk and trust it. Something I do not understand seems to be triggering a force download.

stangri, 2024-02-20:
Does adblock-fast pick the compressed cache on boot?

ckuethe, 2024-02-20:
Sometimes. I just rebooted; my router connected to the same AP and network where it was previously connected and the compressed cache was used.

I've not been able to tell what was going on when it didn't pick up the cache, but I'll be passing through another airport in a few hours so we'll see what happens then.

ckuethe, 2024-03-09:
Took a while to get back to an airport where I could test; the compressed cache did get picked up just now.

stangri, 2024-03-09:
Feel free to reopen with logs if you can reproduce the issue in the future, I'm closing this for now.

closed 2024-03-09 by stangri