openwrt/packages #23477: travelmate: when does random mac generation happen

ckuethe, 2024-02-20:
Maintainer: @dibdot 
Environment:
```
Model: GL-MT300N-V2
Architecture: MediaTek MT7628AN ver:1 eco:2
Target Platform: ramips/mt76x8
Firmware Version: OpenWrt 23.05.0 r23497-6637af95aa / LuCI openwrt-23.05 branch git-24.048.61449-c66fc92
Kernel Version: 5.15.134

# opkg list-installed | grep travelmate
luci-app-travelmate - git-24.006.72299-11a1a43
travelmate - 2.1.2-2
```

Description:

In the [README](https://github.com/openwrt/packages/blob/master/net/travelmate/files/README.md#travelmate-config-options), the `trm_randomize` option is documented as "generat[ing] a random unicast MAC address for each uplink connection".

It's not clear under what conditions this randomization is done. When I first add the uplink network? Every time I reconnect? Had a couple of moments this evening where I got kicked off my connection because my outgoing MAC seems to have changed.

Travelmate logs (~275KB) at https://gist.github.com/ckuethe/36a87142368d0a1f21905f170e192773 - trm_uplink18 is the active uplink network.

brada4, 2024-02-20:
If you read the script it sets random MAC once per added wifi network:
https://github.com/openwrt/packages/blob/187b52c95afc826213ba3e204e43ed009d8e8908/net/travelmate/files/travelmate.sh#L271
Your log shows just one connection with one MAC address
```
Tue Feb 20 03:27:36 2024 user.info trm-2.1.2-2[2568]: connected to uplink 'radio0/Unitedwifi.com/-' with mac '9E:A0:82:4E:8D:61' (1/30)
```
And it is random ie not IEEE allocated.

To diagnose:
Double log size
Once you suspect something wrong:
`logread -e with.mac`

dibdot, 2024-02-21:
This is an issue tracker, not an additional support channel. Please ask such questions in the forum.

closed 2024-02-21 by dibdot