openwrt/packages #23480: mwan3: Potential race condition between mwan3 and tailscale

qdrop17, 2024-02-20:
Maintainer: @neheb / @mochaaP / @BKPepe
Environment: 
- x86/64 / OpenWrt 23.05.2 r23630-842932a63d / LuCI openwrt-23.05 branch git-23.306.39416-c86c256 
- mwan3 2.11.8-1
- tailscale 1.56.1-2

### Description:

We have set up mwan3 and tailscale in accordance with our specific requirements, and have included the configuration details for your reference.

However, we are encountering an issue where the routing from an internal subnet, which is behind OpenWRT, to the subnets behind tailscale is inconsistent upon rebooting the appliance. Sometimes it functions correctly, while at other times it does not. We suspect that this inconsistency may be due to a race condition between mwan3 and tailscale. Interestingly, once we restart tailscale, the system begins to operate as expected without any hitches.

Here are the conditions before restarting tailscale, when we are unable to connect to the tailscale mesh:

```
Priority	Rule
0	from all lookup local
1001	from all iif eth1 lookup 1
1003	from all iif eth2 lookup 3
2001	from all fwmark 0x100/0x3f00 lookup 1
2003	from all fwmark 0x300/0x3f00 lookup 3
2061	from all fwmark 0x3d00/0x3f00 blackhole
2062	from all fwmark 0x3e00/0x3f00 unreachable
3001	from all fwmark 0x100/0x3f00 unreachable
3003	from all fwmark 0x300/0x3f00 unreachable
5210	from all fwmark 0x80000/0xff0000 lookup main
5230	from all fwmark 0x80000/0xff0000 lookup default
5250	from all fwmark 0x80000/0xff0000 unreachable
5270	from all lookup 52
32766	from all lookup main
32767	from all lookup default
```

After restarting tailscale (or when it initially works properly):

```
Priority	Rule
0	from all lookup local
1001	from all iif eth1 lookup 1
1003	from all iif eth2 lookup 3
1310	from all fwmark 0x80000/0xff0000 lookup main
1330	from all fwmark 0x80000/0xff0000 lookup default
1350	from all fwmark 0x80000/0xff0000 unreachable
1370	from all lookup 52
2001	from all fwmark 0x100/0x3f00 lookup 1
2003	from all fwmark 0x300/0x3f00 lookup 3
2061	from all fwmark 0x3d00/0x3f00 blackhole
2062	from all fwmark 0x3e00/0x3f00 unreachable
3001	from all fwmark 0x100/0x3f00 unreachable
3003	from all fwmark 0x300/0x3f00 unreachable
32766	from all lookup main
32767	from all lookup default
```

Here are the respective configuration files:
[firewall-config.txt](https://github.com/openwrt/packages/files/14342911/firewall-config.txt)
[network-config.txt](https://github.com/openwrt/packages/files/14342912/network-config.txt)
[tailscale-config.txt](https://github.com/openwrt/packages/files/14342913/tailscale-config.txt)
[mwan3-config.txt](https://github.com/openwrt/packages/files/14342915/mwan3-config.txt)
[tailscale-up-command.txt](https://github.com/openwrt/packages/files/14343012/tailscale-up-command.txt)

Here are the boot logs for both an example where it was working and one where it failed:
[boot-log-failure.txt](https://github.com/openwrt/packages/files/14342970/boot-log-failure.txt)
[boot-log-success.txt](https://github.com/openwrt/packages/files/14342971/boot-log-success.txt)

Any help is highly appreciated.

qdrop17, 2024-02-20:
The problem continues to occur, even when I disconnect one of the WAN interfaces.

Isn't mwan3 responsible for organizing the routing rules? To be more precise, isn't it the sequence of these rules that appears to be the distinguishing factor between functional and dysfunctional routing?

When we incorporate a "sleep 10" command into the start function of /etc/init.d/tailscale, everything consistently initializes correctly.

brada4, 2024-02-20:
nftables is responsible for assigning fwmarks. Based on those each route is independent from previous in the list.

brada4, 2024-02-20:
i.e. what is the difference between `nft list ruleset` between working and non-working cases.

dc-me, 2024-06-28:
That's because the route rule is not correct, when you restart tailscale it detected mwan3 presense, so the route rule is at the right priority, My guess is that when you reboot it auto starts tailscale but mwan3 is not detected properly, maybe change the tailscale boot order, the default is: 
```
root@e8450r1:/etc/rc.d$ ls | grep tailscale
S80tailscale
root@e8450r1:/etc/rc.d$ ls | grep mwan3
S19mwan3
```
maybe put service tailscale restart at rc.local to see if it works, or manually change the tailscale boot order to 99.

dc-me, 2024-06-28:
Another thing worth mentioning is that, The exit node is also not work, with mwan route rule:
```
1001	from all iif eth1 lookup 1
1003	from all iif eth2 lookup 3
```
When the package comes back from remoute is usally lands on your wan interface, with this rule, the package never reaches to the tailscale0 interface, which makes exit node not working, for now I work around this with manual del this route rule, not sure why mwan3 does this, need futher investigating.