openwrt/packages #23488: daemon.err dnscrypt-proxy[507]

mrusli70, 2024-02-21:
After i install, the dnscrypt-proxy2, there seems to be daemon.err, so i am not that too sure whether the dnscrypt-proxy is in working conditions. But by checking the system logs files it seems that there seems to be a daemon errors. 

Wed Feb 21 06:50:32 2024 daemon.err dnscrypt-proxy[507]: [2024-02-21 06:50:32] [ERROR] Timeout while waiting for network connectivity
Wed Feb 21 06:50:32 2024 daemon.err dnscrypt-proxy[507]: [2024-02-21 06:50:32] [NOTICE] Now listening to 127.0.0.53:53 [UDP]
Wed Feb 21 06:50:32 2024 daemon.err dnscrypt-proxy[507]: [2024-02-21 06:50:32] [NOTICE] Now listening to 127.0.0.53:53 [TCP]
Wed Feb 21 06:50:32 2024 daemon.err dnscrypt-proxy[507]: [2024-02-21 06:50:32] [NOTICE] Source [public-resolvers] loaded
Wed Feb 21 06:50:32 2024 daemon.err dnscrypt-proxy[507]: [2024-02-21 06:50:32] [NOTICE] Source [relays] loaded
Wed Feb 21 06:50:32 2024 daemon.err dnscrypt-proxy[507]: [2024-02-21 06:50:32] [NOTICE] Firefox workaround initialized
Wed Feb 21 06:50:32 2024 daemon.err dnscrypt-proxy[507]: [2024-02-21 06:50:32] [ERROR] Get "https://dns.google/dns-query?dns=yv4BAAABAAAAAAABAAACAAEAACkQAAAAAAAAFAAMABChWRjkGcv7ERDvlbOkfdoH": dial tcp 8.8.8.8:443: connect: network is unreachable
Wed Feb 21 06:50:32 2024 daemon.err dnscrypt-proxy[507]: [2024-02-21 06:50:32] [NOTICE] dnscrypt-proxy is waiting for at least one server to be reachable

and 

Wed Feb 21 06:50:42 2024 daemon.err dnscrypt-proxy[507]: [2024-02-21 06:50:42] [NOTICE] [cloudflare] OK (DoH) - rtt: 87ms
Wed Feb 21 06:50:42 2024 daemon.err dnscrypt-proxy[507]: [2024-02-21 06:50:42] [NOTICE] [google] OK (DoH) - rtt: 6ms
Wed Feb 21 06:50:42 2024 daemon.err dnscrypt-proxy[507]: [2024-02-21 06:50:42] [NOTICE] Sorted latencies:
Wed Feb 21 06:50:42 2024 daemon.err dnscrypt-proxy[507]: [2024-02-21 06:50:42] [NOTICE] -     6ms google
Wed Feb 21 06:50:42 2024 daemon.err dnscrypt-proxy[507]: [2024-02-21 06:50:42] [NOTICE] -    87ms cloudflare
Wed Feb 21 06:50:42 2024 daemon.err dnscrypt-proxy[507]: [2024-02-21 06:50:42] [NOTICE] Server with the lowest initial latency: google (rtt: 6ms)

Can anyone help me out. Thank you.

brada4, 2024-02-21:
The error message clearly says particular connection failed with particular error code at particular moment.
You reaching github indirectly proves your dns works.
You need to check upstream documentation or forums for usage advices.

BKPepe, 2024-02-22:
Yes

closed 2024-02-22 by BKPepe

mrusli70, 2024-02-22:
wait guys, when i install the dnscrypt-proxy2 do i have to change anything or leave it as it is.  like for instance do i have to change anything with the  /etc/config/dhcp? or just install and have it run as it is by default.