openwrt/packages #23495: prometheus-node-exporter-lua: Missing dependency libubus-lua

apirek, 2024-02-21:
Maintainer: @champtar
Environment: ath79/generic, TP-Link EAP245 v1, OpenWrt 23.05.2 r23630-842932a63d / LuCI openwrt-23.05 branch git-23.306.39416-c86c256

Description: prometheus-node-exporter-lua fails to start because it can't find ubus:
```
Wed Feb 21 16:38:43 2024 daemon.err uhttpd[3397]: Error initializing /usr/bin/prometheus-node-exporter-lua Lua handler: /usr/lib/lua/prometheus-collectors/netclass.lua:1: module 'ubus' not found:
Wed Feb 21 16:38:43 2024 daemon.err uhttpd[3397]: 	no field package.preload['ubus']
Wed Feb 21 16:38:43 2024 daemon.err uhttpd[3397]: 	no file './ubus.lua'
Wed Feb 21 16:38:43 2024 daemon.err uhttpd[3397]: 	no file '/usr/share/lua/ubus.lua'
Wed Feb 21 16:38:43 2024 daemon.err uhttpd[3397]: 	no file '/usr/share/lua/ubus/init.lua'
Wed Feb 21 16:38:43 2024 daemon.err uhttpd[3397]: 	no file '/usr/lib/lua/ubus.lua'
Wed Feb 21 16:38:43 2024 daemon.err uhttpd[3397]: 	no file '/usr/lib/lua/ubus/init.lua'
Wed Feb 21 16:38:43 2024 daemon.err uhttpd[3397]: 	no file './ubus.so'
Wed Feb 21 16:38:43 2024 daemon.err uhttpd[3397]: 	no file '/usr/lib/lua/ubus.so'
Wed Feb 21 16:38:43 2024 daemon.err uhttpd[3397]: 	no file '/usr/lib/lua/loadall.so'
Wed Feb 21 16:38:43 2024 daemon.info procd: Instance prometheus-node-exporter-lua::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
```

libubus-lua is missing from the dependencies for the prometheus-node-exporter-lua package. Some of the additional exporters have it but I didn't install any of them, only the base exporter.

Waboath, 2024-04-21:
same problem here

BSthun, 2024-05-27:
Installing `libubus-lua` manually solves the problem.

```
opkg update
opkg install libubus-lua
```

PR #24303 (2024-06-02): prometheus-node-exporter-lua: add missing libubus-lua dependency
Maintainer: me

Fixes #23495
closed 2024-06-02 by champtar