openwrt/packages #23527: mwan3: Multiwan switches to only 1 WAN only

sophipl, 2024-02-24:
Environment: aarch64, AX9000,  OpenWrt 23.05.2 r23630-842932a63d

Description:
MWAN3 does not keep enabled more than one WAN.
At bootup both 2 WANs are Online, I can send pings from them `ping 8.8.8.8 -I vlan-plus` and `ping 8.8.8.8 -I vlan-tplus`
But shortly after one of the WAN is chosen, and another is disabled.
There is no rule which one, it's random.
In config file, there is another WAN, Orange, which is not working yet USB modem failsafe.
Both interfaces keep their IPs

I tried many configurations, none works, replaced legacy xtables and tables to nft versions, i have no legacy packages installed on my system

```
Interface status:
 interface TPLUS is offline 00h:00m:00s, uptime 00h:26m:37s and tracking is active
 interface ORANGE is error (16) and tracking is paused
 interface PLUS is online 00h:08m:40s, uptime 00h:08m:41s and tracking is active
```

Sometimes the logs claims error 16 on one of TPLUS or PLUS too

`cat /etc/config/mwan3`
```
config globals 'globals'
        option mmx_mask '0x3F00'
        list rt_table_lookup '220'
        option logging '1'
        option loglevel 'notice'

config policy 'balanced'
        option last_resort 'unreachable'
        list use_member 'tplus_member'
        list use_member 'plus_member'
        list use_member 'orange_member'

config rule 'https'
        option sticky '1'
        option dest_port '443'
        option proto 'tcp'
        option use_policy 'balanced'

config rule 'default_rule_v4'
        option dest_ip '0.0.0.0/0'
        option use_policy 'balanced'
        option family 'ipv4'
        option proto 'all'
        option sticky '0'

config rule 'default_rule_v6'
        option dest_ip '::/0'
        option use_policy 'balanced'
        option family 'ipv6'

config interface 'TPLUS'
        option initial_state 'online'
        option family 'ipv4'
        option track_method 'ping'
        option reliability '1'
        option count '1'
        option timeout '4'
        option interval '10'
        option failure_interval '5'
        option recovery_interval '5'
        option down '5'
        option up '5'
        option enabled '1'
        list flush_conntrack 'connected'
        list track_ip '8.8.8.8'
        list track_ip '208.67.222.222'
        option size '56'
        option max_ttl '60'

config interface 'ORANGE'
        option enabled '1'
        option initial_state 'offline'
        option family 'ipv4'
        option track_method 'ping'
        option reliability '1'
        option count '1'
        option timeout '4'
        option interval '10'
        option failure_interval '5'
        option recovery_interval '5'
        option down '5'
        option up '5'
        list track_ip '8.8.4.4'
        list track_ip '208.67.220.220'
        option size '56'
        option max_ttl '60'

config interface 'PLUS'
        option enabled '1'
        option initial_state 'online'
        option family 'ipv4'
        option track_method 'ping'
        option reliability '1'
        option count '1'
        option timeout '4'
        option interval '10'
        option failure_interval '5'
        option recovery_interval '5'
        option down '5'
        option up '5'
        list flush_conntrack 'connected'
        list track_ip '8.8.8.8'
        list track_ip '208.67.220.220'
        option size '56'
        option max_ttl '60'

config policy 'orange_only'
        option last_resort 'unreachable'
        list use_member 'orange_member'

config rule 'inverter'
        option family 'ipv4'
        option proto 'all'
        option src_ip '10.234.12.100'
        option sticky '0'
        option use_policy 'orange_only'

config member 'plus_member'
        option interface 'PLUS'
        option metric '1'
        option weight '10'

config member 'tplus_member'
        option interface 'TPLUS'
        option metric '1'
        option weight '1000'

config member 'orange_member'
        option interface 'ORANGE'
        option metric '256'
        option weight '1'

```

part of `cat /etc/config/network`
```
config interface 'TPLUS'
        option proto 'dhcp'
        option device 'vlan-tplus'

config interface 'PLUS'
        option proto 'dhcp'
        option device 'vlan-plus'

config device
        option type '8021q'
        option ifname 'wan'
        option vid '23'
        option name 'vlan-tplus'

config device
        option type '8021q'
        option ifname 'wan'
        option vid '20'
        option name 'vlan-plus'
```

part of log
```
Sun Feb 25 00:35:17 2024 user.info mwan3track[2857]: Check (ping) failed for target "8.8.8.8" on interface TPLUS (vlan-tplus). Current score: 10
Sun Feb 25 00:35:21 2024 user.info mwan3track[2857]: Check (ping) failed for target "208.67.222.222" on interface TPLUS (vlan-tplus). Current score: 10
Sun Feb 25 00:35:21 2024 user.notice mwan3track[2857]: Interface TPLUS (vlan-tplus) is disconnecting
Sun Feb 25 00:35:31 2024 user.info mwan3track[2857]: Check (ping) failed for target "8.8.8.8" on interface TPLUS (vlan-tplus). Current score: 9
Sun Feb 25 00:35:35 2024 user.info mwan3track[2857]: Check (ping) failed for target "208.67.222.222" on interface TPLUS (vlan-tplus). Current score: 9
Sun Feb 25 00:35:44 2024 user.info mwan3track[2857]: Check (ping) failed for target "8.8.8.8" on interface TPLUS (vlan-tplus). Current score: 8
Sun Feb 25 00:35:48 2024 user.info mwan3track[2857]: Check (ping) failed for target "208.67.222.222" on interface TPLUS (vlan-tplus). Current score: 8
Sun Feb 25 00:35:57 2024 user.info mwan3track[2857]: Check (ping) failed for target "8.8.8.8" on interface TPLUS (vlan-tplus). Current score: 7
Sun Feb 25 00:36:01 2024 user.info mwan3track[2857]: Check (ping) failed for target "208.67.222.222" on interface TPLUS (vlan-tplus). Current score: 7
Sun Feb 25 00:36:10 2024 user.info mwan3track[2857]: Check (ping) failed for target "8.8.8.8" on interface TPLUS (vlan-tplus). Current score: 6
Sun Feb 25 00:36:14 2024 user.info mwan3track[2857]: Check (ping) failed for target "208.67.222.222" on interface TPLUS (vlan-tplus). Current score: 6
Sun Feb 25 00:36:14 2024 user.notice mwan3track[2857]: Interface TPLUS (vlan-tplus) is offline
```

sophipl, 2024-02-26:
Sure, keep in mind, I managed to boot Orange USB 3G WAN, it stays up, but it doesn't work, in a sence of wan rule for one device to use it.
I needed to add static route to make it work

```
table inet fw4 {
        chain input {
                type filter hook input priority filter; policy drop;
                iifname "lo" accept comment "!fw4: Accept traffic from loopback"
                ct state established,related accept comment "!fw4: Allow inbound established and related flows"
                tcp flags syn / fin,syn,rst,ack jump syn_flood comment "!fw4: Rate limit TCP syn packets"
                iifname "br-lan" jump input_lan comment "!fw4: Handle lan IPv4/IPv6 input traffic"
                iifname "br-triv" jump input_TrivZone comment "!fw4: Handle TrivZone IPv4/IPv6 input traffic"
                iifname "br-work" jump input_WorkZone comment "!fw4: Handle WorkZone IPv4/IPv6 input traffic"
                iifname "vlan-tplus" jump input_TplusZone comment "!fw4: Handle TplusZone IPv4/IPv6 input traffic"
                iifname "vlan-plus" jump input_PlusZone comment "!fw4: Handle PlusZone IPv4/IPv6 input traffic"
                iifname "3g-ORANGE" jump input_OrangeZone comment "!fw4: Handle OrangeZone IPv4/IPv6 input traffic"
                iifname "br-nowan" jump input_NoWanZone comment "!fw4: Handle NoWanZone IPv4/IPv6 input traffic"
                jump handle_reject
        }

        chain forward {
                type filter hook forward priority filter; policy drop;
                ct state established,related accept comment "!fw4: Allow forwarded established and related flows"
                iifname "br-lan" jump forward_lan comment "!fw4: Handle lan IPv4/IPv6 forward traffic"
                iifname "br-triv" jump forward_TrivZone comment "!fw4: Handle TrivZone IPv4/IPv6 forward traffic"
                iifname "br-work" jump forward_WorkZone comment "!fw4: Handle WorkZone IPv4/IPv6 forward traffic"
                iifname "vlan-tplus" jump forward_TplusZone comment "!fw4: Handle TplusZone IPv4/IPv6 forward traffic"
                iifname "vlan-plus" jump forward_PlusZone comment "!fw4: Handle PlusZone IPv4/IPv6 forward traffic"
                iifname "3g-ORANGE" jump forward_OrangeZone comment "!fw4: Handle OrangeZone IPv4/IPv6 forward traffic"
                iifname "br-nowan" jump forward_NoWanZone comment "!fw4: Handle NoWanZone IPv4/IPv6 forward traffic"
                jump handle_reject
        }

        chain output {
                type filter hook output priority filter; policy accept;
                oifname "lo" accept comment "!fw4: Accept traffic towards loopback"
                ct state established,related accept comment "!fw4: Allow outbound established and related flows"
                oifname "br-lan" jump output_lan comment "!fw4: Handle lan IPv4/IPv6 output traffic"
                oifname "br-triv" jump output_TrivZone comment "!fw4: Handle TrivZone IPv4/IPv6 output traffic"
                oifname "br-work" jump output_WorkZone comment "!fw4: Handle WorkZone IPv4/IPv6 output traffic"
                oifname "vlan-tplus" jump output_TplusZone comment "!fw4: Handle TplusZone IPv4/IPv6 output traffic"
                oifname "vlan-plus" jump output_PlusZone comment "!fw4: Handle PlusZone IPv4/IPv6 output traffic"
                oifname "3g-ORANGE" jump output_OrangeZone comment "!fw4: Handle OrangeZone IPv4/IPv6 output traffic"
                oifname "br-nowan" jump output_NoWanZone comment "!fw4: Handle NoWanZone IPv4/IPv6 output traffic"
        }

        chain prerouting {
                type filter hook prerouting priority filter; policy accept;
                iifname "br-lan" jump helper_lan comment "!fw4: Handle lan IPv4/IPv6 helper assignment"
                iifname "br-triv" jump helper_TrivZone comment "!fw4: Handle TrivZone IPv4/IPv6 helper assignment"
                iifname "br-work" jump helper_WorkZone comment "!fw4: Handle WorkZone IPv4/IPv6 helper assignment"
                iifname "br-nowan" jump helper_NoWanZone comment "!fw4: Handle NoWanZone IPv4/IPv6 helper assignment"
        }

        chain handle_reject {
                meta l4proto tcp reject with tcp reset comment "!fw4: Reject TCP traffic"
                reject comment "!fw4: Reject any other traffic"
        }

        chain syn_flood {
                limit rate 25/second burst 50 packets return comment "!fw4: Accept SYN packets below rate-limit"
                drop comment "!fw4: Drop excess packets"
        }

        chain input_lan {
                jump accept_from_lan
        }

        chain output_lan {
                jump accept_to_lan
[... 391 lines cut ...]
        }

        chain mwan3_policy_balanced {
                meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 60 bytes 10495 xt target "MARK"
        }

        chain mwan3_policy_orange_only {
                meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 0 bytes 0 xt target "MARK"
        }

        chain mwan3_policy_plus_only {
                meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 0 bytes 0 xt target "MARK"
        }

        chain mwan3_policy_tplus_only {
                meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 0 bytes 0 xt target "MARK"
        }

        chain mwan3_rule_https {
                meta mark & 0x00003f00 == 0x00000000 counter packets 0 bytes 0 jump mwan3_policy_balanced
                meta mark & 0x0000fc00 != 0x0000fc00 counter packets 0 bytes 0 xt target "SET"
                meta mark & 0x0000fc00 != 0x0000fc00 counter packets 0 bytes 0 xt target "SET"
        }
}
```

brada4, 2024-02-26:
xt marks and nft marks dont see eachother. @jow- might be better versed which module got skipped.