openwrt/packages #23546: tor: Version 4.7.10 in 22.03 is considered insecure by the network

peci1, 2024-02-27:
Maintainer: @rsalvaterra @BKPepe @neheb 
Environment: Turris Omnia, OpenWRT 22.03

Description: I received this email from eol-relays@torproject.org:

> You are running a Tor relay, which is great

> However, that relay's Tor version is obsolete, and because of old bugs, we will soon cut relays and bridges running that version out of the network. Please consider upgrading!

Can I ask for backporting a version that is not considered insecure by the network?

BKPepe, 2024-02-27:
Sure, I will look into it. Thanks for reporting. Just ask Turris guys to run a new build for you and release it into HBT branch.

peci1, 2024-02-27:
Thanks! 

> Just ask Turris guys to run a new build

When should I ask them? After you do the release? Or now?

BKPepe, 2024-02-27:
Backported. Hopefully, it would be enough.
You can see it here: https://github.com/openwrt/packages/commits/openwrt-22.03/?since=2024-02-27&until=2024-02-27

Tor is now synced with all OpenWrt stable releases (OpenWrt 22.03, OpenWrt 23.05 and OpenWrt master).

closed 2024-02-27 by BKPepe

peci1, 2024-02-27:
Great, that was lightning-fast! Thanks a lot.

BKPepe, 2024-02-27:
You are welcome. I am ready to help, if I can and if there is time.