openwrt/packages #23561: libc: BPI R3 issue on repo

iz4tow, 2024-02-28:
Hi,
I'm using a BananaPi R3 with openwrt last image (23.05) downloaded from openwrt firmware selector.
During zabbix-agentd installation, `opkg install zabbix-agentd` I discovered that it failed because "libc" had no candidates for my architecture.
But....very strange....Browsing manually the official repo I noticed that "libc" actually exists for BPI R3 arch.
So using these commands I was able to install libc, then zabbix agent
```
wget <official repo>/libc(...).ipk
opkg install ./libc(...).ipk
opkg install zabbix-agentd
```

Could be an indexing issue on the repo I think

brada4, 2024-02-29:
libc is installed since you are able to run busybox shell and opkg.  Likely you upgraded with one of partial images and you have mixed userland from various versions.

iz4tow, 2024-02-29:
No, I downloaded this image [23.05.2](https://downloads.openwrt.org/releases/23.05.2/targets/mediatek/filogic/openwrt-23.05.2-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz) then:
```
opkg update
opkg install libc
```

brada4, 2024-03-01:
why would you ever reinstall preinstalled library? 
```
usr/lib/opkg/info# cat libc.control
          Package: libc                                             
    Version: 1.2.4-4
     Architecture: aarch64_cortex-a53                              Installed-Size: 280548                                        Description:  C library
```

iz4tow, 2024-03-01:
Because when I tried to install zabbix-agentd
```
opkg install zabbix-agentd
```
The error was "dependency libc is not available for your arch".

Anyway in the afternoon I will reinstall my bpi3 and I'll send u the full commands and errors.

brada4, 2024-03-01:
The opkg database contains `./libc_1.2.4-4_aarch64_cortex-a53.ipk` preinstalled, it is kind of main system entrypoint that every program, including dropbear, busybox ash shell and opkg depend on.
Please check /usr/lib/opkg/info, frankly no idea what removed local package database....