openwrt/packages #23567: https-dns-proxy: After reboot, a backup dnsmasq configuration is loaded

rakhard, 2024-03-01:
Maintainer: @stangri
Environment: mt7621, 23.05.2

After reboot:

```
# logread -e https-dns-proxy
Fri Mar  1 15:43:20 2024 user.notice https-dns-proxy: Starting service instances on_boot ✓
Fri Mar  1 15:43:20 2024 user.notice https-dns-proxy: Updating dnsmasq config ✓
Fri Mar  1 15:43:26 2024 user.notice https-dns-proxy: Restarting dnsmasq on_config_update ✓
Fri Mar  1 15:43:31 2024 user.notice https-dns-proxy: Stopping service on_boot ✓
```

```
# service https-dns-proxy status
running
```

```
# logread -e dnsmasq
Fri Mar  1 15:43:11 2024 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Mar  1 15:43:11 2024 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Mar  1 15:43:11 2024 daemon.info dnsmasq[1]: started, version 2.89 cachesize 1000
Fri Mar  1 15:43:11 2024 daemon.info dnsmasq[1]: DNS service limited to local subnets
Fri Mar  1 15:43:11 2024 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-nftset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
Fri Mar  1 15:43:11 2024 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Fri Mar  1 15:43:11 2024 daemon.info dnsmasq[1]: using nameserver 9.9.9.9#53
Fri Mar  1 15:43:11 2024 daemon.info dnsmasq[1]: using nameserver 149.112.112.112#53
Fri Mar  1 15:43:11 2024 daemon.info dnsmasq[1]: using only locally-known addresses for test
Fri Mar  1 15:43:11 2024 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Fri Mar  1 15:43:11 2024 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Fri Mar  1 15:43:11 2024 daemon.info dnsmasq[1]: using only locally-known addresses for local
Fri Mar  1 15:43:11 2024 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Fri Mar  1 15:43:11 2024 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Fri Mar  1 15:43:11 2024 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Fri Mar  1 15:43:11 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Fri Mar  1 15:43:11 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 names
Fri Mar  1 15:43:20 2024 user.notice https-dns-proxy: Updating dnsmasq config ✓
Fri Mar  1 15:43:21 2024 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Fri Mar  1 15:43:24 2024 daemon.info dnsmasq[1]: started, version 2.89 cachesize 1000
Fri Mar  1 15:43:24 2024 daemon.info dnsmasq[1]: DNS service limited to local subnets
Fri Mar  1 15:43:24 2024 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-nftset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
Fri Mar  1 15:43:24 2024 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Fri Mar  1 15:43:24 2024 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 12h
Fri Mar  1 15:43:24 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5053
Fri Mar  1 15:43:24 2024 daemon.info dnsmasq[1]: using only locally-known addresses for test
Fri Mar  1 15:43:24 2024 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Fri Mar  1 15:43:24 2024 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Fri Mar  1 15:43:24 2024 daemon.info dnsmasq[1]: using only locally-known addresses for local
Fri Mar  1 15:43:24 2024 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Fri Mar  1 15:43:24 2024 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Fri Mar  1 15:43:24 2024 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Fri Mar  1 15:43:24 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Fri Mar  1 15:43:24 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Fri Mar  1 15:43:24 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c.2105 - 0 names
Fri Mar  1 15:43:24 2024 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 0 names
Fri Mar  1 15:43:24 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Fri Mar  1 15:43:26 2024 user.notice https-dns-proxy: Restarting dnsmasq on_config_update ✓
Fri Mar  1 15:43:28 2024 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan)
Fri Mar  1 15:43:28 2024 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan)
Fri Mar  1 15:43:28 2024 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Fri Mar  1 15:43:31 2024 daemon.info dnsmasq[1]: started, version 2.89 cachesize 1000
Fri Mar  1 15:43:31 2024 daemon.info dnsmasq[1]: DNS service limited to local subnets
Fri Mar  1 15:43:31 2024 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-nftset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
Fri Mar  1 15:43:31 2024 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Fri Mar  1 15:43:31 2024 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 12h
Fri Mar  1 15:43:31 2024 daemon.info dnsmasq[1]: using nameserver 9.9.9.9#53
Fri Mar  1 15:43:31 2024 daemon.info dnsmasq[1]: using nameserver 149.112.112.112#53
Fri Mar  1 15:43:31 2024 daemon.info dnsmasq[1]: using only locally-known addresses for test
Fri Mar  1 15:43:31 2024 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Fri Mar  1 15:43:31 2024 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Fri Mar  1 15:43:31 2024 daemon.info dnsmasq[1]: using only locally-known addresses for local
Fri Mar  1 15:43:31 2024 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Fri Mar  1 15:43:31 2024 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Fri Mar  1 15:43:31 2024 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Fri Mar  1 15:43:31 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Fri Mar  1 15:43:31 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Fri Mar  1 15:43:31 2024 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 0 names
Fri Mar  1 15:43:31 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
...
```

https-dns-proxy restart:
```
Fri Mar  1 16:07:29 2024 user.notice https-dns-proxy: Updating dnsmasq config ✓
Fri Mar  1 16:07:29 2024 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Fri Mar  1 16:07:33 2024 daemon.info dnsmasq[1]: started, version 2.89 cachesize 1000
Fri Mar  1 16:07:33 2024 daemon.info dnsmasq[1]: DNS service limited to local subnets
Fri Mar  1 16:07:33 2024 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-nftset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
Fri Mar  1 16:07:33 2024 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Fri Mar  1 16:07:33 2024 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 12h
Fri Mar  1 16:07:33 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5053
Fri Mar  1 16:07:33 2024 daemon.info dnsmasq[1]: using only locally-known addresses for test
Fri Mar  1 16:07:33 2024 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Fri Mar  1 16:07:33 2024 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Fri Mar  1 16:07:33 2024 daemon.info dnsmasq[1]: using only locally-known addresses for local
Fri Mar  1 16:07:33 2024 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Fri Mar  1 16:07:33 2024 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Fri Mar  1 16:07:33 2024 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Fri Mar  1 16:07:33 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Fri Mar  1 16:07:33 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Fri Mar  1 16:07:33 2024 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 0 names
Fri Mar  1 16:07:33 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Fri Mar  1 16:07:33 2024 user.notice https-dns-proxy: Restarting dnsmasq on_config_update ✓
```

stangri, 2024-03-01:
Thank you for providing logs. I'm failing to see what the issue is.

rakhard, 2024-03-01:
This relates to the [comment](https://github.com/openwrt/packages/issues/23469#issuecomment-1951305529) made in another issue regarding the functioning of https-dns-proxy, which only seems to work after manual restart.

rakhard, 2024-03-01:
> Thank you for providing logs. I'm failing to see what the issue is.

Here's the problem: `resolveip` in not listed as a dependency. [is_resolver_working() { resolveip -t 3 one.one.one.one >/dev/null 2>&1; }](https://github.com/openwrt/packages/blob/04ac8c177d9ac662790ef5fec9b66cab038c341f/net/https-dns-proxy/files/etc/init.d/https-dns-proxy#L38)

stangri, 2024-03-01:
Are you saying you don't have resolveip on your OpenWrt install?

stangri, 2024-03-01:
> This relates to the [comment](https://github.com/openwrt/packages/issues/23469#issuecomment-1951305529) made in another issue regarding the functioning of https-dns-proxy, which only seems to work after a manual restart.

In the non-working-after-restart state, please provide the output of:
```
logread -ehttps
service https-dns-proxy status
```

rakhard, 2024-03-01:
> Are you saying you don't have resolveip on your OpenWrt install?

Yes, that's correct. I didn't have the package. After installing it, everything works fine now.

rakhard, 2024-03-01:
> Are you saying you don't have resolveip on your OpenWrt install?

Should it be there? This package is not listed in the manifest file.

stangri, 2024-03-01:
> > Are you saying you don't have resolveip on your OpenWrt install?
> 
> Should it be there? This package is not listed in the manifest file.

My understanding is that it's included with all systems, but you're right, there should be a dependency, I'll push a fix this weekend!

Thank you for pretty much solving the issue and presenting the solution!

closed 2024-03-04 by stangri