openwrt/packages #23582: kmod-iwl-legacy and kmod-iwl3945 packages missing in 23.05.2

Manfred-Knick, 2024-03-04:
Maintainer: Felix Fietkau @\<nbd@openwrt.org> (?)

Environment: arch x86, generic, OpenWrt 23.05.2

Description:

Their absence disables eg. ThinkPad T60 etc.

Both missing packages, kmod-iwl-legacy and kmod-iwl3945,
had been maintained up to OpenWrt 22.03.6

They can be found in .../packages:
- [kmod-iwl-legacy_5.10.201+5.15.92-1-1_i386_pentium4.ipk](https://downloads.openwrt.org/releases/22.03.6/targets/x86/generic/packages/kmod-iwl-legacy_5.10.201+5.15.92-1-1_i386_pentium4.ipk)
- [kmod-iwl3945_5.10.201+5.15.92-1-1_i386_pentium4.ipk](https://downloads.openwrt.org/releases/22.03.6/targets/x86/generic/packages/kmod-iwl3945_5.10.201+5.15.92-1-1_i386_pentium4.ipk)

Ironically, the name seems to indicate a backport from kernel 5.15
which is the kernel line selected for OpenWrt 23.05.2 ?

Reference:
This issue has been reported in the OpenWrt Forum -> [Installing and Using OpenWrt](https://forum.openwrt.org/c/general/6) on 6. / 7. Feb.;
latest: [No more Intel 3945 Golan wifi driver on openwrt 23.05.2](https://forum.openwrt.org/t/no-more-intel-3945-golan-wifi-driver-on-openwrt-23-05-2/187133)

BKPepe, 2024-03-04:
This is packages feed, not main repository. So, you need to report it elsewhere.

closed 2024-03-04 by BKPepe

Manfred-Knick, 2024-03-04:
"elsewhere" is very helpful, indeed -
especially after consulting [ https://openwrt.org/bugs ] ...

Thanks.