openwrt/packages #23588: p910nd: drivers are missing when hotpluging

loss-and-quick, 2024-03-05:
Maintainer: @pkerling
Environment: `ramips/mt7621`, `TP-Link Archer C6U v1`, `OpenWrt 23.05.0`

Description:
Recently encountered a problem when connecting the printer: 
`daemon.err p910nd hotplug: Missing driver file: /opt/p910nd_drivers/FUJI_XEROX_DocuPrint_CP105_b_0550_00f5.bin for /dev/usb/lp0 [ 0550/00f5 ] (please upload it).`

It is solved by restarting the p910nd, but there was no such problem before.
How to solve this problem normally?

brada4, 2024-03-05:
What do you mean connecting? Like you connect usb and it asks bin file?
Hotplug?
i.e can you detail your setup and explain what does not work and how you wanted it to work.

systemcrash, 2024-03-05:
@loss-and-quick The very important text: `please upload it`. 

hotplug sets up the environment for your printer - you supply the blob. Does your printer need a blob loading to it before it will print? Or do you just plug it in and dump data to `lp0` and it prints?

If not, please paste the output of the following commands:

```sh
ls /sys/class/usbmisc/lp0/device/
cat /sys/class/usbmisc/lp0/device/ieee1284_id
cat /sys/class/usbmisc/lp0/device/bInterfaceClass
cat /sys/class/usbmisc/lp0/device/bInterfaceProtocol
cat /sys/class/usbmisc/lp0/device/bInterfaceSubClass
cat /sys/class/usbmisc/lp0/device/bNumEndpoints
```

Might need a bit more data later.

loss-and-quick, 2024-03-05:
> What do you mean connecting?

I just turn the printer on, it's always off if I'm not using it.

loss-and-quick, 2024-03-05:
Temporarily I made this script for hotplug.d/usb:
```
#!/bin/sh
if [ "$PRODUCT" = "0550/00f5/100" ]
then
case "$ACTION" in
        add)
        sleep 11s
        /etc/init.d/p910nd restart
        ;;
esac
fi
```

systemcrash, 2024-03-05:
> Temporarily I made this script for hotplug.d/usb:

please paste the output of the following commands
```sh
ls /sys/class/usbmisc/lp0/device/
cat /sys/class/usbmisc/lp0/device/ieee1284_id
cat /sys/class/usbmisc/lp0/device/bInterfaceClass
cat /sys/class/usbmisc/lp0/device/bInterfaceProtocol
cat /sys/class/usbmisc/lp0/device/bInterfaceSubClass
cat /sys/class/usbmisc/lp0/device/bNumEndpoints
```

loss-and-quick, 2024-03-05:
> ```shell
> ls /sys/class/usbmisc/lp0/device/
> cat /sys/class/usbmisc/lp0/device/ieee1284_id
> cat /sys/class/usbmisc/lp0/device/bInterfaceClass
> cat /sys/class/usbmisc/lp0/device/bInterfaceProtocol
> cat /sys/class/usbmisc/lp0/device/bInterfaceSubClass
> cat /sys/class/usbmisc/lp0/device/bNumEndpoints
> ```

`ls /sys/class/usbmisc/lp0/device/`:
```

authorized            bNumEndpoints         subsystem
bAlternateSetting     driver                supports_autosuspend
bInterfaceClass       ep_01                 uevent
bInterfaceNumber      ep_82                 usbmisc
bInterfaceProtocol    ieee1284_id
bInterfaceSubClass    modalias
```

eee1284_id:
```
MFG:FUJI XEROX;CMD:PJL,RASTER,DOWNLOAD,HBPL;MDL:DocuPrint CP105 b;DES:FUJI XEROX DocuPrint CP105 b;CLS:PRINTER;STS:AAAWAAD/AP8AAAAAAAIAAWQDZANkA2QDAAAAAGQDQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAD/
/w==;
```

bInterfaceClass:
`07`

bInterfaceProtocol:
`02`

bInterfaceSubClass
`01`

loss-and-quick, 2024-03-05:
> hotplug sets up the environment for your printer - you supply the blob. Does your printer need a blob loading to it before it will print? Or do you just plug it in and dump data to `lp0` and it prints?

Sorry,I don't know

loss-and-quick, 2024-03-05:
> i.e can you detail your setup and explain what does not work and how you wanted it to work.

I have a Phaser 6000 that has been converted to Docu Print 105b
When I need it I just turn it on and print files, but recently I got a missing drivers error when I turn the printer on, I solve this problem by restarting the p910nd( either manually or with a script).

loss-and-quick, 2024-03-05:
> Rephrase: do you just turn your printer on and it prints without having to do anything other than click 'print'?

Yes, i just turn on printer and click `print`

systemcrash, 2024-03-05:
I note `        sleep 11s` from your script - why 11 seconds?

loss-and-quick, 2024-03-05:
> I note ` sleep 11s` from your script - why 11 seconds?

if i just restart without `sleep` it will throw error

systemcrash, 2024-03-05:
Does `sleep 1` work equally well?

loss-and-quick, 2024-03-05:
> Does `sleep 1` work equally well?

I lied, I've checked it now and it works without delay (I just changed the script without restarting the router.)

brada4, 2024-03-05:
You can stop and start p910nd on USB hotplug, i.e 5s after printer connects. Print server will be offline when printer is of - kind of elegant.
Some need to be filled with paper to set up on windows first time.

loss-and-quick, 2024-03-05:
> You can stop and start p910nd on USB hotplug, i.e 5s after printer connects. Print server will be offline when printer is of - kind of elegant.
> Some need to be filled with paper to set up on windows first time.

I'm already doing it

brada4, 2024-03-05:
Some printers try to get fw update from windows driver when powered on, cups has quirks table like eject CD, wait 10s then know it is up. Otherwise you get all pages printed with one character in rapid sequence.

loss-and-quick, 2024-03-05:
> Some printers try to get fw update from windows driver when powered on, cups has quirks table like eject CD, wait 10s then know it is up. Otherwise you get all pages printed with one character in rapid sequence.

I don't think so

systemcrash, 2024-03-05:
> > Does `sleep 1` work equally well?
> 
> I lied, I've checked it now and it works without delay (I just changed the script without restarting the router.) I kind of tried it without delay and the error was

OK, I think I have a fix.

systemcrash, 2024-03-05:
@loss-and-quick please test the following. 

Remove your script in `hotplug.d/usb` and modify the script `hotplug.d/usbmisc/20-p910nd`

```diff
@@ -335,10 +335,9 @@ send_driver() {
                        logger -t "$DAEMON_HOTPLUG" -p "$DAEMON_ERR" "Sending driver to $CHAR_DEV [ $THIS_USB_VIDPID ] failed for some reason."
                else
                        logger -t "$DAEMON_HOTPLUG" -p "$DAEMON_INFO" "Sent $DRIVER_FILE to $CHAR_DEV [ $THIS_USB_VIDPID ]."
-                       daemon_restart
                fi
        else
-               logger -t "$DAEMON_HOTPLUG"  -p "$DAEMON_ERR" "Missing driver file: $DRIVER_FILE for $CHAR_DEV [ $THIS_USB_VIDPID ] (please upload it)."
+               logger -t "$DAEMON_HOTPLUG"  -p "$DAEMON_INFO" "No driver file: $DRIVER_FILE for $CHAR_DEV [ $THIS_USB_VIDPID ] (upload it if your printer needs a driver loading)."
        fi
 }

@@ -370,6 +369,7 @@ case "$ACTION" in
                                [ "$DEBUG" ] && echo "THIS_USB_VIDPID match for $DAEMON device $THIS_USB_VIDPID."

                                send_driver
+                               daemon_restart

                        else
                                [ "$DEBUG" ] && echo "No THIS_USB_VIDPID match."
```

And power off/on your printer as usual.

systemcrash, 2024-03-06:
OK, thanks for confirming. PR is all set then.

closed 2024-03-12 by BKPepe