openwrt/packages #23596: cloudflared: Needs to be built via Go with Cloudflare experimental patches

MoetaYuko, 2024-03-06:
Maintainer: @1715173329
Environment: OpenWrt 23.05.2 (r23630-842932a63d), any arch

Description:
cloudflared fails to make quic connection and finally fallbacks to http2
```
Wed Mar  6 15:32:28 2024 daemon.err cloudflared[14145]: 2024-03-06T07:32:28Z ERR Failed to create new quic connection error="failed to dial to edge with quic: INTERNAL_ERROR (local): tls: CurvePreferences includes unsupported curve" connIndex=0 event=0 ip=198.41.192.57
Wed Mar  6 15:32:28 2024 daemon.err cloudflared[14145]: 2024-03-06T07:32:28Z INF Retrying connection in up to 2s connIndex=0 event=0 ip=198.41.192.57
Wed Mar  6 15:32:30 2024 daemon.err cloudflared[14145]: 2024-03-06T07:32:30Z ERR Failed to create new quic connection error="failed to dial to edge with quic: INTERNAL_ERROR (local): tls: CurvePreferences includes unsupported curve" connIndex=0 event=0 ip=198.41.192.107
Wed Mar  6 15:32:30 2024 daemon.err cloudflared[14145]: 2024-03-06T07:32:30Z INF Retrying connection in up to 4s connIndex=0 event=0 ip=198.41.192.107
Wed Mar  6 15:32:33 2024 daemon.err cloudflared[14145]: 2024-03-06T07:32:32Z ERR Failed to create new quic connection error="failed to dial to edge with quic: INTERNAL_ERROR (local): tls: CurvePreferences includes unsupported curve" connIndex=0 event=0 ip=198.41.192.67
Wed Mar  6 15:32:33 2024 daemon.err cloudflared[14145]: 2024-03-06T07:32:32Z INF Retrying connection in up to 8s connIndex=0 event=0 ip=198.41.192.67
Wed Mar  6 15:32:35 2024 daemon.err cloudflared[14145]: 2024-03-06T07:32:35Z ERR Failed to create new quic connection error="failed to dial to edge with quic: INTERNAL_ERROR (local): tls: CurvePreferences includes unsupported curve" connIndex=0 event=0 ip=198.41.192.37
Wed Mar  6 15:32:35 2024 daemon.err cloudflared[14145]: 2024-03-06T07:32:35Z INF Retrying connection in up to 16s connIndex=0 event=0 ip=198.41.192.37
Wed Mar  6 15:32:47 2024 daemon.err cloudflared[14145]: 2024-03-06T07:32:47Z ERR Failed to create new quic connection error="failed to dial to edge with quic: INTERNAL_ERROR (local): tls: CurvePreferences includes unsupported curve" connIndex=0 event=0 ip=198.41.192.47
Wed Mar  6 15:32:47 2024 daemon.err cloudflared[14145]: 2024-03-06T07:32:47Z INF Retrying connection in up to 32s connIndex=0 event=0 ip=198.41.192.47
Wed Mar  6 15:33:04 2024 daemon.err cloudflared[14145]: 2024-03-06T07:33:04Z ERR Failed to create new quic connection error="failed to dial to edge with quic: INTERNAL_ERROR (local): tls: CurvePreferences includes unsupported curve" connIndex=0 event=0 ip=198.41.192.57
```

the solution is to build it via [Go with Cloudflare experimental patches](https://github.com/cloudflare/go) according to https://github.com/cloudflare/cloudflared/issues/1132#issuecomment-1885154712 https://github.com/cloudflare/cloudflared/issues/1151#issuecomment-1888738119

1715173329, 2024-03-06:
If cloudflared somehow does not work nice with official Go toolchain, it is not a issue related to us. Cloudflare should work on addressing it.

We will not pack a vendor-patched go toolchain for only single package.

MoetaYuko, 2024-03-06:
> Cloudflare should work on addressing it.

ig they'll try to upstream their patches.

However, assuming their patches are accepted in a future golang version (say, 1.XY), go 1.XY won't get backported to previous openwrt releases according to the policy of not bumping major version, so cloudflared will be broken for openwrt released w/ go < 1.XY forever.

Hence, there are two solutions from my point of view:
1. package patched go for cloudflared
2. downgrade cloudflared to the one that doesn't depend on patched go

1715173329, 2024-03-06:
> package patched go for cloudflared

it's impossible unless they upstreamed those patches.

cc @stokito any suggestion?

MoetaYuko, 2024-03-06:
FYI nix uses patched go https://github.com/NixOS/nixpkgs/commit/f13a5874518b046bf390fa593b01da0b08d06b84

stokito, 2024-03-06:
I agree with @1715173329 and we shouldn't change anything in go just for the non critical package to fix non critical problem. Maybe it's possible to just disable the QUIC from a build until the problem resolved.

stokito, 2024-03-06:
@MoetaYuko you may build own package with the patched go and upload for example here in attachment. If someone really needs it they can install it manually

MoetaYuko, 2024-03-06:
I just created a project to compile cloudflared with cfgo for all openwrt architectures https://github.com/MoetaYuko/openwrt-cloudflared

1715173329, 2024-03-06:
> Maybe it's possible to just disable the QUIC from a build until the problem resolved.

looks like cloudflared does not offer such build-time option. is it possible to disable it in configuration?

> downgrade cloudflared to the one that doesn't depend on patched go

btw the version that does not require patched go and works with go1.21 does not exist.

MoetaYuko, 2024-03-06:
> > downgrade cloudflared to the one that doesn't depend on patched go
> 
> btw the version that does not require patched go and works with go1.21 does not exist.

indeed. according to cf dev https://github.com/cloudflare/cloudflared/issues/1151#issuecomment-1888738119, introducing dependence on cfgo is part of the requirement to support go1.21

BKPepe, 2024-03-06:
Since I agree with @1715173329 and as well with @stokito, there isn't anything what we can do except to disable QUIC. We are not going to include some experimental patches, etc. This should be closed as not planned. Cloudflare should upstream their patches.

closed 2024-03-08 by BKPepe

alex9434, 2024-04-11:
I suggested in https://github.com/openwrt/packages/issues/23852#issuecomment-2048908443 to add a warning to the package description to make users aware of the issue.

swg0101, 2024-09-09:
Note that disabling QUIC will also disable UDP and ICMP support. This means if you run tunnels over TLS only, only TCP packets would be allowed - all other packets directed to the tunnel will be dropped.

PunkFleet, 2024-12-23:
Is this problem solved? I'm installing cloudflared on my NAS, and setting up a VM as openwrt at the same time, and I'm trying to point both the NAS gateway and DNS to the openwrt address, but I'm running into the cloudflare error!
```
20024-12-23 04:45:38.253644+00:002024-12-23T04:45:38Z INF Starting tunnel tunnelID=d652c760-3ccf-45fd-b637-8be5cbf34702
2024-12-23 04:45:38.253683+00:002024-12-23T04:45:38Z INF Version 2024.12.2 (Checksum cb61fcb41380efff4e2ef0a1eeea23bcddff360e8703f99dbb5c24da1de937b9)
2024-12-23 04:45:38.253689+00:002024-12-23T04:45:38Z INF GOOS: linux, GOVersion: go1.22.5-devel-cf, GoArch: amd64
2024-12-23 04:45:38.253691+00:002024-12-23T04:45:38Z INF Settings: map[no-autoupdate:true]
2024-12-23 04:45:38.253696+00:002024-12-23T04:45:38Z INF Environmental variables map[TUNNEL_TOKEN:*****]
2024-12-23 04:45:38.253838+00:002024-12-23T04:45:38Z INF Generated Connector ID: cd8ce6e8-42f4-47ef-a353-165ef2189970
2024-12-23 04:45:39.045169+00:002024-12-23T04:45:39Z INF Initial protocol quic
2024-12-23 04:45:39.059692+00:002024-12-23T04:45:39Z INF ICMP proxy will use 172.16.5.2 as source for IPv4
2024-12-23 04:45:39.059853+00:002024-12-23T04:45:39Z INF ICMP proxy will use ::1 in zone lo as source for IPv6
2024-12-23 04:45:39.654816+00:002024-12-23T04:45:39Z INF ICMP proxy will use 172.16.5.2 as source for IPv4
2024-12-23 04:45:39.655078+00:002024-12-23T04:45:39Z INF ICMP proxy will use ::1 in zone lo as source for IPv6
2024-12-23 04:45:39.655223+00:002024-12-23T04:45:39Z INF Starting metrics server on 127.0.0.1:20241/metrics
2024-12-23 04:45:39.660388+00:002024/12/23 04:45:39 failed to sufficiently increase receive buffer size (was: 208 kiB, wanted: 7168 kiB, got: 416 kiB). See https://github.com/quic-go/quic-go/wiki/UDP-Buffer-Sizes for details.
2024-12-23 04:45:44.663102+00:002024-12-23T04:45:44Z ERR Failed to dial a quic connection error="failed to dial to edge with quic: timeout: no recent network activity" connIndex=0 event=0 ip=198.41.192.107
2024-12-23 04:45:44.663253+00:002024-12-23T04:45:44Z INF Retrying connection in up to 2s connIndex=0 event=0 ip=198.41.192.107
2024-12-23 04:45:50.933009+00:002024-12-23T04:45:50Z ERR Failed to dial a quic connection error="failed to dial to edge with quic: timeout: no recent network activity" connIndex=0 event=0 ip=198.41.192.37
2024-12-23 04:45:50.933142+00:002024-12-23T04:45:50Z INF Retrying connection in up to 4s connIndex=0 event=0 ip=198.41.192.37
2024-12-23 04:45:56.094671+00:002024-12-23T04:45:56Z ERR Failed to dial a quic connection error="failed to dial to edge with quic: timeout: no recent network activity" connIndex=0 event=0 ip=198.41.200.193
```