openwrt/packages #23607: _testcapi | ipq40xx generic | ModuleNotFoundError when extra C flags used

knuxyl, 2024-03-07:
Maintainer: not sure it's relevant, this package may not be intentionally included
Environment: armv7-a | GL.iNet GL-A1300 | ipq40xx/generic | master

Anytime a kernel C flag or target C flag is added in menuconfig it results in this error. I've tried several combinations of them with no change in reults. Anytime there is something custom there, it cannot find this python module _testcapi. This error doesn't show up until near the end of compilation, and all 10 times I have tested it this is the steps I take

make dirclean
make menuconfig / save
make -j`nproc` V=sc (make V=sc didn't make a difference)

I don't even know where to being with this. The flags I was trying to use were different combinations/ommissions of these
```
-mtune=cortex-a7
-mfpu=neon-vfpv4
-mfloat-abi=hard
-O2
-O3
-funsafe-math-optimizations (tried after it was failing)
```
The target c flags were default to `-O2 -pipe`, so I left that there or change -O2 to -O3.

The build works perfectly fine as is without the extra C flags added. I've attached my working configuration file.
[config.buildinfo.zip](https://github.com/openwrt/packages/files/14524439/config.buildinfo.zip)

Error
```
0:02:55 load avg: 3.62 [37/44] test_sqlite3
Failed to import test module: test.test_sqlite3.test_dbapi
Traceback (most recent call last):
  File "/home/knuxyl/Mount/4tb/system/git/openwrt/build_dir/hostpkg/Python-3.11.7/Lib/unittest/loader.py", line 419, in _find_test_path
    module = self._get_module_from_name(name)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/knuxyl/Mount/4tb/system/git/openwrt/build_dir/hostpkg/Python-3.11.7/Lib/unittest/loader.py", line 362, in _get_module_from_name
    __import__(name)
  File "/home/knuxyl/Mount/4tb/system/git/openwrt/build_dir/hostpkg/Python-3.11.7/Lib/test/test_sqlite3/test_dbapi.py", line 37, in <module>
    from _testcapi import INT_MAX, ULLONG_MAX
ModuleNotFoundError: No module named '_testcapi'

Failed to import test module: test.test_sqlite3.test_dump
Traceback (most recent call last):
  File "/home/knuxyl/Mount/4tb/system/git/openwrt/build_dir/hostpkg/Python-3.11.7/Lib/unittest/loader.py", line 419, in _find_test_path
    module = self._get_module_from_name(name)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/knuxyl/Mount/4tb/system/git/openwrt/build_dir/hostpkg/Python-3.11.7/Lib/unittest/loader.py", line 362, in _get_module_from_name
    __import__(name)
  File "/home/knuxyl/Mount/4tb/system/git/openwrt/build_dir/hostpkg/Python-3.11.7/Lib/test/test_sqlite3/test_dump.py", line 5, in <module>
    from .test_dbapi import memory_database
  File "/home/knuxyl/Mount/4tb/system/git/openwrt/build_dir/hostpkg/Python-3.11.7/Lib/test/test_sqlite3/test_dbapi.py", line 37, in <module>
    from _testcapi import INT_MAX, ULLONG_MAX
ModuleNotFoundError: No module named '_testcapi'

Failed to import test module: test.test_sqlite3.test_hooks
Traceback (most recent call last):
  File "/home/knuxyl/Mount/4tb/system/git/openwrt/build_dir/hostpkg/Python-3.11.7/Lib/unittest/loader.py", line 419, in _find_test_path
    module = self._get_module_from_name(name)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/knuxyl/Mount/4tb/system/git/openwrt/build_dir/hostpkg/Python-3.11.7/Lib/unittest/loader.py", line 362, in _get_module_from_name
    __import__(name)
  File "/home/knuxyl/Mount/4tb/system/git/openwrt/build_dir/hostpkg/Python-3.11.7/Lib/test/test_sqlite3/test_hooks.py", line 29, in <module>
    from test.test_sqlite3.test_dbapi import memory_database, cx_limit
  File "/home/knuxyl/Mount/4tb/system/git/openwrt/build_dir/hostpkg/Python-3.11.7/Lib/test/test_sqlite3/test_dbapi.py", line 37, in <module>
    from _testcapi import INT_MAX, ULLONG_MAX
ModuleNotFoundError: No module named '_testcapi'

Failed to import test module: test.test_sqlite3.test_regression
Traceback (most recent call last):
  File "/home/knuxyl/Mount/4tb/system/git/openwrt/build_dir/hostpkg/Python-3.11.7/Lib/unittest/loader.py", line 419, in _find_test_path
    module = self._get_module_from_name(name)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/knuxyl/Mount/4tb/system/git/openwrt/build_dir/hostpkg/Python-3.11.7/Lib/unittest/loader.py", line 362, in _get_module_from_name
    __import__(name)
  File "/home/knuxyl/Mount/4tb/system/git/openwrt/build_dir/hostpkg/Python-3.11.7/Lib/test/test_sqlite3/test_regression.py", line 31, in <module>
    from test.test_sqlite3.test_dbapi import memory_database, cx_limit
  File "/home/knuxyl/Mount/4tb/system/git/openwrt/build_dir/hostpkg/Python-3.11.7/Lib/test/test_sqlite3/test_dbapi.py", line 37, in <module>
    from _testcapi import INT_MAX, ULLONG_MAX
ModuleNotFoundError: No module named '_testcapi'

Failed to import test module: test.test_sqlite3.test_transactions
Traceback (most recent call last):
  File "/home/knuxyl/Mount/4tb/system/git/openwrt/build_dir/hostpkg/Python-3.11.7/Lib/unittest/loader.py", line 419, in _find_test_path
    module = self._get_module_from_name(name)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/knuxyl/Mount/4tb/system/git/openwrt/build_dir/hostpkg/Python-3.11.7/Lib/unittest/loader.py", line 362, in _get_module_from_name
    __import__(name)
  File "/home/knuxyl/Mount/4tb/system/git/openwrt/build_dir/hostpkg/Python-3.11.7/Lib/test/test_sqlite3/test_transactions.py", line 28, in <module>
    from test.test_sqlite3.test_dbapi import memory_database
  File "/home/knuxyl/Mount/4tb/system/git/openwrt/build_dir/hostpkg/Python-3.11.7/Lib/test/test_sqlite3/test_dbapi.py", line 37, in <module>
    from _testcapi import INT_MAX, ULLONG_MAX
ModuleNotFoundError: No module named '_testcapi'

Failed to import test module: test.test_sqlite3.test_userfunctions
Traceback (most recent call last):
  File "/home/knuxyl/Mount/4tb/system/git/openwrt/build_dir/hostpkg/Python-3.11.7/Lib/unittest/loader.py", line 419, in _find_test_path
    module = self._get_module_from_name(name)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/knuxyl/Mount/4tb/system/git/openwrt/build_dir/hostpkg/Python-3.11.7/Lib/unittest/loader.py", line 362, in _get_module_from_name
    __import__(name)
  File "/home/knuxyl/Mount/4tb/system/git/openwrt/build_dir/hostpkg/Python-3.11.7/Lib/test/test_sqlite3/test_userfunctions.py", line 35, in <module>
    from test.test_sqlite3.test_dbapi import cx_limit
  File "/home/knuxyl/Mount/4tb/system/git/openwrt/build_dir/hostpkg/Python-3.11.7/Lib/test/test_sqlite3/test_dbapi.py", line 37, in <module>
    from _testcapi import INT_MAX, ULLONG_MAX
ModuleNotFoundError: No module named '_testcapi'

0:02:55 load avg: 3.62 [38/44] test_statistics -- test_sqlite3 failed (uncaught exception)
```

brada4, 2024-03-07:
Python 3.11 requires floating point NaN support.
You need to make fresh build tree after intentional miscompilation.

knuxyl, 2024-03-08:
according to
https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem
make dirclean wipes out everything built, even the toolchain.
so what do you mean a fresh build tree?

brada4, 2024-03-08:
ccache for example.

knuxyl, 2024-03-08:
are you saying i should or should not use ccache? i believe ccache is for speeding up builds by caching older one and comparing. i feel like you're just saying random stuff to sound smart because you're providing absolutely no valuable information in solving this problem.

brada4, 2024-03-08:
Just build toolset with generic parameters so that emitted target platform code is  later compiled correctly.