openwrt/packages #23621: PF_RING-8.0.0 error: invalid application of 'sizeof' to incomplete type 'char[]'

JimMatthew, 2024-03-11:
Environment: x86 snapshot from master (seems to be all targets?)

Description:
Unable to build libpfring

See logs:

https://downloads.openwrt.org/snapshots/faillogs/x86_64/packages/libpfring/

make[4]: Entering directory '/builder/shared-workdir/build/sdk/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.80'
  CC [M]  /builder/shared-workdir/build/sdk/build_dir/target-x86_64_musl/linux-x86_64/PF_RING-8.4.0/kernel/pf_ring.o
/builder/shared-workdir/build/sdk/build_dir/target-x86_64_musl/linux-x86_64/PF_RING-8.4.0/kernel/pf_ring.c: In function 'ring_bind':
/builder/shared-workdir/build/sdk/build_dir/target-x86_64_musl/linux-x86_64/PF_RING-8.4.0/kernel/pf_ring.c:5574:21: error: invalid application of 'sizeof' to incomplete type 'char[]'
 5574 |     char name[sizeof(sa->sa_data)+1];
      |                     ^
In file included from ./include/linux/string.h:293,
                 from ./include/linux/bitmap.h:11,
                 from ./include/linux/cpumask.h:12,
                 from ./arch/x86/include/asm/cpumask.h:5,
                 from ./arch/x86/include/asm/msr.h:11,
                 from ./arch/x86/include/asm/processor.h:22,
                 from ./arch/x86/include/asm/timex.h:5,
                 from ./include/linux/timex.h:67,
                 from ./include/linux/time32.h:13,
                 from ./include/linux/time.h:60,
                 from ./include/linux/stat.h:19,
                 from ./include/linux/module.h:13,
                 from /builder/shared-workdir/build/sdk/build_dir/target-x86_64_musl/linux-x86_64/PF_RING-8.4.0/kernel/pf_ring.c:79:
/builder/shared-workdir/build/sdk/build_dir/target-x86_64_musl/linux-x86_64/PF_RING-8.4.0/kernel/pf_ring.c:5579:37: error: invalid application of 'sizeof' to incomplete type 'char[]'
 5579 |     memcpy(name, sa->sa_data, sizeof(sa->sa_data));
      |                                     ^
./include/linux/fortify-string.h:457:48: note: in definition of macro '__fortify_memcpy_chk'
  457 |         const size_t __fortify_size = (size_t)(size);                   \
      |                                                ^~~~
/builder/shared-workdir/build/sdk/build_dir/target-x86_64_musl/linux-x86_64/PF_RING-8.4.0/kernel/pf_ring.c:5579:5: note: in expansion of macro 'memcpy'
 5579 |     memcpy(name, sa->sa_data, sizeof(sa->sa_data));
      |     ^~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-x86_64_musl/linux-x86_64/PF_RING-8.4.0/kernel/pf_ring.c:5574:10: error: unused variable 'name' [-Werror=unused-variable]
 5574 |     char name[sizeof(sa->sa_data)+1];
      |          ^~~~
```

remittor, 2024-03-11:
Patch for libpfring:
```
--- a/kernel/pf_ring.c
+++ b/kernel/pf_ring.c
@@ -5571,12 +5571,12 @@
* Check legality
*/
if (addr_len == sizeof(struct sockaddr)) {
- char name[sizeof(sa->sa_data)+1];
+ char name[sizeof(sa->sa_data_min)+1];

if (sa->sa_family != PF_RING)
return(-EINVAL);

- memcpy(name, sa->sa_data, sizeof(sa->sa_data));
+ memcpy(name, sa->sa_data, sizeof(sa->sa_data_min));

/* Add trailing zero if missing */
name[sizeof(name)-1] = '\0'; 
```

alimirjamali, 2024-03-12:
I can confirm that the patch works on bcm47xx sdk

marcinmajsc, 2024-03-13:
For me works pertect!

PF_RING-8.4.0
OpenWrt SNAPSHOT r25529+1-1d3d6ef826
mvebu/cortexa9
Kernel 6.1.81
device wrt1200ac

marcinmajsc, 2024-03-13:
> Patch for libpfring:
> 
> ```
> --- a/kernel/pf_ring.c
> +++ b/kernel/pf_ring.c
> @@ -5562,12 +5562,12 @@ static int ring_bind(struct socket *sock, struct sockaddr *sa, int addr_len)
>     * Check legality
>     */
>    if (addr_len == sizeof(struct sockaddr)) {
> -    char name[sizeof(sa->sa_data)+1];
> +    char name[sizeof(sa->sa_data_min)+1];
>  
>      if (sa->sa_family != PF_RING)
>        return(-EINVAL);
>  
> -    memcpy(name, sa->sa_data, sizeof(sa->sa_data));
> +    memcpy(name, sa->sa_data, sizeof(sa->sa_data_min));
>  
>      /* Add trailing zero if missing */
>      name[sizeof(name)-1] = '\0';
> ```

For me works pertect!

PF_RING-8.4.0
OpenWrt SNAPSHOT r25529+1-1d3d6ef826
mvebu/cortexa9
Kernel 6.1.81
device wrt1200ac

taylorkline, 2024-03-16:
Quick question, why does this not fail for the build bot that builds `mediatek/filogic`, but it fails for me when I am using the same config file downloaded from:

`wget https://downloads.openwrt.org/snapshots/targets/mediatek/filogic/config.buildinfo -O .config`

db260179, 2024-03-17:
To add to this 23.05 is broken as well, this patch fixes this

```
--- a/kernel/pf_ring.c
+++ b/kernel/pf_ring.c
@@ -5605,7 +5605,7 @@ static int packet_ring_bind(struct sock
 static int ring_bind(struct socket *sock, struct sockaddr *sa, int addr_len)
 {
   struct sock *sk = sock->sk;
-  char name[sizeof(sa->sa_data)+1];
+  char name[sizeof(sa->sa_data_min)+1];
 
   debug_printk(2, "ring_bind() called\n");
 
@@ -5617,7 +5617,7 @@ static int ring_bind(struct socket *sock
   if(sa->sa_family != PF_RING)
     return(-EINVAL);
 
-  memcpy(name, sa->sa_data, sizeof(sa->sa_data));
+  memcpy(name, sa->sa_data, sizeof(sa->sa_data_min));
 
   /* Add trailing zero if missing */
   name[sizeof(name)-1] = '\0';
```

taylorkline, 2024-03-20:
It's also fixed in `libpfring` versions 8.6.0 and above, so #23612 should fix it.

grobian, 2024-03-26:
yes, but 23.05 is on 8.0.0 so https://github.com/openwrt/packages/pull/23612 doesn't apply

remittor, 2024-03-26:
@grobian 
Yesterday, to successfully build many 23.05.3 firmwares, I added the following patch to the `libpfring` directory:
```
--- a/kernel/pf_ring.c
+++ b/kernel/pf_ring.c
@@ -5562,18 +5562,18 @@ static int ring_bind(struct socket *sock, struct sockaddr *sa, int addr_len)
 {
   struct sock *sk = sock->sk;
-  char name[sizeof(sa->sa_data)+1];
+  char name[sizeof(sa->sa_data_min)+1];
 
   debug_printk(2, "ring_bind() called\n");
 
   /*
    * Check legality
    */
   if(addr_len != sizeof(struct sockaddr))
     return(-EINVAL);
   if(sa->sa_family != PF_RING)
     return(-EINVAL);
 
-  memcpy(name, sa->sa_data, sizeof(sa->sa_data));
+  memcpy(name, sa->sa_data, sizeof(sa->sa_data_min));
 
   /* Add trailing zero if missing */
   name[sizeof(name)-1] = '\0';
```

grobian, 2024-03-26:
Yes, thank you.  Building already, and it allows to continue.  The solution(s) in https://github.com/openwrt/openwrt/issues/14841 did not work for me.

The patch mentioned here multiple times can be added to the `feeds/packages/libs/libpfring/patches/` directory, e.g. as `999-issue-23621.patch`.

Leo-PL, 2024-03-30:
This hit me today as well. I wonder about one thing: why on earth this package is built despite not being selected at all in my .config and having no direct dependencies?

ntzb, 2024-03-30:
tried to search this error for days now.
the title should be `ERROR: package/feeds/packages/libpfring failed to build.`

> This hit me today as well. I wonder about one thing: why on earth this package is built despite not being selected at all in my .config and having no direct dependencies?

I wonder this as well, seems very fishy

taylorkline, 2024-03-31:
@Leo-PL @ntzb Do you have `CONFIG_ALL_KMODS=y` in your config?

Leo-PL, 2024-03-31:
@taylorkline I do.

ingamedeo, 2024-04-21:
I am confused, I checked out v23.05.3 and I get the same issue, how does this build on the buildbot without the patch above?

Leo-PL, 2024-04-22:
Isn't that related to GCC version as well?
Just in case, the (def)config used by buildbot is here: https://downloads.openwrt.org/releases/23.05.3/targets/ath79/generic/config.buildinfo - the same is true for respective subtargets, and `CONFIG_ALL_KMODS=y` is definitely there. Other question is, how all other symbols are expanded from that.

dudeitssm, 2024-06-03:
Worked for me! Thanks!