openwrt/packages #23626: transmission-daemon missing

plasticassius, 2024-03-12:
Maintainer: @dangowrt
Environment: kirkwood, seagate_goflexnet, 23.05.2
Description:
transmission-daemon not found immediately after sysupgrade.
```
sysupgrade -n -v openwrt-23.05.2-kirkwood-generic-seagate_goflexnet-squashfs-sysupgrade.bin
.
.
.
root@OpenWrt:~# opkg update
Downloading https://downloads.openwrt.org/releases/23.05.2/targets/kirkwood/generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.2/targets/kirkwood/generic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/arm_xscale/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/arm_xscale/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/arm_xscale/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/arm_xscale/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/arm_xscale/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/arm_xscale/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/arm_xscale/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/arm_xscale/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/arm_xscale/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/arm_xscale/telephony/Packages.sig
Signature check passed.
root@OpenWrt:~# opkg install transmission-daemon
Unknown package 'transmission-daemon'.
Collected errors:
 * opkg_install_cmd: Cannot install package transmission-daemon.
root@OpenWrt:~# 
```

BKPepe, 2024-03-12:
This is caused by libdeflate. It is missing: https://downloads.openwrt.org/releases/faillogs-23.05/arm_xscale/packages/libdeflate/compile.txt

Snip:
```
/builder/shared-workdir/build/sdk/build_dir/target-arm_xscale_musl_eabi/libdeflate-1.18/lib/../common_defs.h:446:1: error: inlining failed in call to 'always_inline' 'get_unaligned_le16': target specific option mismatch
  446 | get_unaligned_le16(const u8 *p)
      | ^~~~~~~~~~~~~~~~~~
/builder/shared-workdir/build/sdk/build_dir/target-arm_xscale_musl_eabi/libdeflate-1.18/lib/arm/crc32_impl.h:220:23: note: called from here
  220 |                 crc = __crc32h(crc, get_unaligned_le16(p));
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:41: /builder/shared-workdir/build/sdk/build_dir/target-arm_xscale_musl_eabi/libdeflate-1.18/.built] Error 1
```

plasticassius, 2024-08-08:
I still see the issue in 23.05.4 but it's fine in SNAPSHOT r27100-21a5b3b540

silverStSt, 2024-09-30:
I still see the issue in 23.05.5

dtechsrv, 2024-10-25:
There is probably some GCC-related bug in libdeflate v1.18, because I tried replacing it to the current version (v1.22), and the transmission package was successfully built for the kirkwood platform (ZyXEL NSA325) using OpenWrt v23.05.5 sources.

silverStSt, 2024-10-26:
I tried with 23.05.0 and the same problem was already there.

dtechsrv, 2024-10-28:
Until then, I made the missing binaries available for v23.05.5:

https://openwrt.dtech.hu/.upload/openwrt-23.05.5-kirkwood-transmission/

plasticassius, 2024-10-30:
@dtechsrv works for me, thanks.

dtechsrv, 2026-02-21:
Was this issue still open?

@shvetsnikita Please open a new one, because this thread was already solved, but it wasn't closed.

@plasticassius Please close this old issue.