openwrt/packages #23645: unbound failed to build with python

beelze, 2024-03-14:
Maintainer: @BKPepe
Environment: (mips_24kc_musl/ath79_generic, OpenWrt 23.05)

Unbound-1.19.1 failed to build with `PACKAGE_libunbound_pythonmodule=y`

Seems that one issue is in Makefile:
```
 $(if $(CONFIG_PACKAGE_libunbound_python),--with-pythonmodule,) \
```
Here expected to be `PACKAGE_libunbound_pythonmodule` instead of `CONFIG_PACKAGE_libunbound_python`

Issuing `make package/unbound/configure V=sc` I've got:
```
checking for python... /mnt/bulk/openwrt-build/openwrt/staging_dir/host/bin/python
checking for the sysconfig Python module... yes
checking for Python include path... -I/usr/include/python3.11
checking for Python library path... -L/usr/lib64 -L/usr/lib/python3.11 -L. -lpython3.11
checking for Python site-packages path... /usr/lib/python3.11/site-packages
checking consistency of all components of python development environment... no
configure: error: 
  Could not link test program to Python. Maybe the main Python library has been
  installed in some non-standard library path. If so, pass it to configure,
  via the LDFLAGS environment variable.
  Example: ./configure LDFLAGS="-L/usr/non-standard-path/python/lib"
```
seems `configure` is looking for python stuff outside \<buildroot>. Here's configure log: [build.log](https://github.com/openwrt/packages/files/14596184/build.log)

BKPepe, 2024-03-14:
I am not the maintainer. Feel free to propose pull request or ask @EricLuehrsen , the maintainer of unbound here.

beelze, 2024-03-14:
@EricLuehrsen , is there any chance to resolve an issue? Tried to fix it myself but no luck

brada4, 2024-03-14:
You see here it picks up host python:
> checking for Python site-packages path... /usr/lib/python3.11/site-packages

beelze, 2024-03-15:
I'm not sure if the host python is needed for build process, but unbound itself (with pythonmodule) should be linked against libpython. I believe the proper lib is `./staging_dir/target-mips_24kc_musl/usr/lib/libpython3.11.so.1.0` but not `/usr/lib64/libpython3.11.so.1.0`, so `-L/usr/lib64` is obviously wrong

beelze, 2024-03-15:
Additionally, because of swig is needed for building unbound with python, Makefile needed something like
`PKG_BUILD_DEPENDS:=PACKAGE_libunbound_pythonmodule:swig/host`
Idk about proper syntax, though...

EricLuehrsen, 2024-03-16:
This custom option was built a long time ago. I haven't worked with unbound+python in awhile, so I am unclear on what dependencies may have shifted.

beelze, 2024-03-18:
@EricLuehrsen Yes, there is a few changes that should be done (not only python-related). Currently I'm working on it.

beelze, 2024-03-19:
@EricLuehrsen I've managed to successfully build and run unbound with pythonmodule (along with other small fixes), Are you interested in it? Just in case, here's a [patch](https://gist.github.com/beelze/9c140972a0694855fc6e979ff861791c)

A little explanation about minor changes:
1. `--with-user` is not supported by configure, but `--with-username` is
2. `PACKAGE_libunbound_ipset` is `n` because of openwrt nowadays uses nftables instead iptables and ipset userspace utility (along with kernel module) isn't installed by default
3. Building unbound with pythonmodule requires host `swig`, so I've put a dependency

brada4, 2024-03-19:
musl?

beelze, 2024-03-19:
Yes, I've biult unbound with target-mips_24kc_musl