openwrt/packages #23651: ERROR: package/feeds/packages/perl [host] failed to build.

uosbox, 2024-03-14:
编译在新版的op报错，之前没遇到这情况
perl版本滚回到5.28就能正常编译通过

`  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/aarch64-linux/Digest/SHA.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Digest/base.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Digest/file.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Net/Ping.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Net/protoent.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Net/SMTP.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Net/hostent.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Net/FTP.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Net/NNTP.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Net/libnetFAQ.pod
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Net/servent.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Net/POP3.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Net/Config.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Net/Netrc.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Net/Time.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Net/Cmd.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Net/Domain.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Net/netent.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Net/FTP/E.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Net/FTP/I.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Net/FTP/L.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Net/FTP/A.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Net/FTP/dataconn.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Test/Tester.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Test/Harness.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Test/More.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Test/Tutorial.pod
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Test/Builder.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Test/Simple.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Test/Tester/Delegate.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Test/Tester/CaptureRunner.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Test/Tester/Capture.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Test/Builder/Tester.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Test/Builder/TodoDiag.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Test/Builder/Module.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Test/Builder/Formatter.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Test/Builder/IO/Scalar.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Test/Builder/Tester/Color.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Test/use/ok.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Perl/OSType.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/HTTP/Tiny.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Attribute/Handlers.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Text/Abbrev.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Text/ParseWords.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Text/Balanced.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Text/Tabs.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Text/Wrap.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Memoize/Storable.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Memoize/NDBM_File.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Memoize/Expire.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Memoize/SDBM_File.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Memoize/AnyDBM_File.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Archive/Tar.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Archive/Tar/File.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Archive/Tar/Constant.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Locale/Maketext.pod
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Locale/Maketext.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Locale/Maketext/Guts.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Locale/Maketext/Cookbook.pod
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Locale/Maketext/GutsLoader.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Locale/Maketext/TPJ13.pod
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/Locale/Maketext/Simple.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/aarch64-linux/IPC/SysV.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/aarch64-linux/IPC/Msg.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/IPC/Open3.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/IPC/Open2.pm
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/aarch64-linux/IPC/SharedMem.pm
[... 328 lines cut ...]
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/pod/perlboot.pod
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/pod/perlbot.pod
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/pod/perlrepository.pod
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/pod/perltodo.pod
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/pod/perltooc.pod
  /root/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38.2/pod/perltoot.pod
install -d -m0755 /root/openwrt/staging_dir/hostpkg/usr/bin/
cp -fpR /root/openwrt/build_dir/hostpkg/perl/perl-5.38.2/generate_uudmap /root/openwrt/staging_dir/hostpkg/usr/bin/
# Link any possibly installed static extension in
make -C /root/openwrt/build_dir/hostpkg/perl/perl-5.38.2/relink clean || true
make[4]: Entering directory '/root/openwrt/build_dir/hostpkg/perl/perl-5.38.2/relink'
make[4]: *** No rule to make target 'clean'.  Stop.
make[4]: Leaving directory '/root/openwrt/build_dir/hostpkg/perl/perl-5.38.2/relink'
( cd /root/openwrt/build_dir/hostpkg/perl/perl-5.38.2/relink && /root/openwrt/staging_dir/hostpkg/usr/bin/perl Makefile.PL )
bash: line 1: 3834328 Segmentation fault      (core dumped) /root/openwrt/staging_dir/hostpkg/usr/bin/perl Makefile.PL
make[3]: *** [Makefile:151: /root/openwrt/staging_dir/hostpkg/stamp/.perl_installed] Error 139
make[3]: Leaving directory '/root/openwrt/feeds/packages/lang/perl'
time: package/feeds/packages/perl/host-compile#222.86#26.52#248.33
    ERROR: package/feeds/packages/perl [host] failed to build.
make[2]: *** [package/Makefile:127: package/feeds/packages/perl/host/compile] Error 1
make[2]: Leaving directory '/root/openwrt'
make[1]: *** [package/Makefile:123: /root/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/openwrt'
make: *** [/root/openwrt/include/toplevel.mk:233: world] Error 2
`

brada4, 2024-03-14:
You have to build as unprivileged user.

uosbox, 2024-03-15:
> You have to build as unprivileged user.

不用root用户还是一样的结果！！
[image]

mu-b, 2024-03-31:
Same here.
Host: arm64

`cp -fpR /mnt/data/source/openwrt/build_dir/hostpkg/perl/perl-5.38.2/generate_uudmap /mnt/data/source/openwrt/staging_dir/hostpkg/usr/bin/
# Link any possibly installed static extension in
make -C /mnt/data/source/openwrt/build_dir/hostpkg/perl/perl-5.38.2/relink clean || true
make[4]: Entering directory '/mnt/data/source/openwrt/build_dir/hostpkg/perl/perl-5.38.2/relink'
make[4]: *** No rule to make target 'clean'.  Stop.
make[4]: Leaving directory '/mnt/data/source/openwrt/build_dir/hostpkg/perl/perl-5.38.2/relink'
( cd /mnt/data/source/openwrt/build_dir/hostpkg/perl/perl-5.38.2/relink && /mnt/data/source/openwrt/staging_dir/hostpkg/usr/bin/perl Makefile.PL )
bash: line 1: 2818844 Segmentation fault      /mnt/data/source/openwrt/staging_dir/hostpkg/usr/bin/perl Makefile.PL
make[3]: *** [Makefile:151: /mnt/data/source/openwrt/staging_dir/hostpkg/stamp/.perl_installed] Error 139
make[3]: Leaving directory '/mnt/data/source/openwrt/feeds/packages/lang/perl'
time: package/feeds/packages/perl/host-compile#3.00#2.86#13.36
    ERROR: package/feeds/packages/perl [host] failed to build.
make[2]: *** [package/Makefile:127: package/feeds/packages/perl/host/compile] Error 1
make[2]: Leaving directory '/mnt/data/source/openwrt'
make[1]: *** [package/Makefile:123: /mnt/data/source/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/mnt/data/source/openwrt'
make: *** [/mnt/data/source/openwrt/include/toplevel.mk:233: world] Error 2`

`[~/source/openwrt]
[2008:0]-16:38:13 mu-b@files $ cd /mnt/data/source/openwrt/build_dir/hostpkg/perl/perl-5.38.2/relink
[/mnt/data/source/openwrt/build_dir/hostpkg/perl/perl-5.38.2/relink]
[2009:0]-16:39:08 mu-b@files $ /mnt/data/source/openwrt/staging_dir/hostpkg/usr/bin/perl Makefile.PL
Segmentation fault
[/mnt/data/source/openwrt/build_dir/hostpkg/perl/perl-5.38.2/relink]
[2010:0]-16:39:17 mu-b@files $ gdb /mnt/data/source/openwrt/staging_dir/hostpkg/usr/bin/perl
GNU gdb (Debian 10.1-1.7) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /mnt/data/source/openwrt/staging_dir/hostpkg/usr/bin/perl...
(No debugging symbols found in /mnt/data/source/openwrt/staging_dir/hostpkg/usr/bin/perl)
(gdb) r Makefile.PL
Starting program: /mnt/data/source/openwrt/staging_dir/hostpkg/usr/bin/perl Makefile.PL
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x0000aaaaab295444 in Perl_re_op_compile ()
(gdb) bt
#0  0x0000aaaaab295444 in Perl_re_op_compile ()
#1  0x0000aaaaab234c08 in Perl_pmruntime ()
#2  0x0000aaaaab2754bc in Perl_yyparse ()
#3  0x0000aaaaab32ca70 in S_doeval_compile ()
#4  0x0000aaaaab32e518 in Perl_pp_require ()
#5  0x0000aaaaab2e9aa0 in Perl_runops_standard ()
#6  0x0000aaaaab23b954 in Perl_call_sv ()
#7  0x0000aaaaab23e718 in Perl_call_list ()
#8  0x0000aaaaab224c6c in S_process_special_blocks ()
#9  0x0000aaaaab233cfc in Perl_newATTRSUB_x ()
#10 0x0000aaaaab236f6c in Perl_utilize ()
#11 0x0000aaaaab274374 in Perl_yyparse ()
#12 0x0000aaaaab242d94 in perl_parse ()
#13 0x0000aaaaab0f21dc in main ()
(gdb) q
A debugging session is active.

	Inferior 1 [process 2818861] will be killed.

Quit anyway? (y or n) y`

Reverting the recent bump for perl to perl-5.38.2 fixes the build.

mu-b, 2024-03-31:
Confirmed, reverting 0142d0659822413a3ac5ed4c76d74cbcf5a1ca7e and 24847159857ccd64d5d823d109a03b1908b25bd0 fixes the build.

predators46, 2024-03-31:
i see the new build in master no problem

https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9/packages/perl_5.38.2-r1_arm_cortex-a9.ipk

If you still experience this problem, please create a PR and rollback to Perl 5.28.x. thanks

esaaprillia, 2024-06-17:
@uosbox @mu-b 

i didn't get this error when building perl-5.40.0

mu-b, 2024-06-17:
Recently updated build machine to bullseye from buster, build failure solved.

closed 2024-06-25 by neheb