openwrt/packages #23653: unbound-daemon: init script causes DNS request over WAN but means to use localhost as destination

bughunter2, 2024-03-14:
Maintainer: @EricLuehrsen
Environment: x86_64, generic, 23.05.2

Description:

When dnsmasq is enabled, and Unbound is configured to listen at port 53, the Unbound init script makes a DNS request which can leave your WAN interface (visible in Wireshark, see below).

In the function ```resolv_setup()```, there's the bug ...:
```
  nslookup localhost 127.0.0.1#53 >/dev/null 2>&1
```

... when fixed, it becomes ...:
```
  nslookup localhost 127.0.0.1:53 >/dev/null 2>&1
```

The original code might have always caused DNS requests to leave WAN, unless some older version of busybox (nslookup) perhaps did accept the '#' syntax to denote the port, but I don't know whether that's the case.

[image]

brada4, 2024-03-14:
The port is not even necessary, default 53 is compiled-in...

bughunter2, 2024-03-14:
> The port is not even necessary, default 53 is compiled-in...

Agreed. Guess I left it in as I was playing around with the syntax, but indeed!

brada4, 2024-03-14:
Original source file has hashmark which is wrong by all means. You found it broken.

EricLuehrsen, 2024-03-16:
Well that's not good... :facepalm:

EricLuehrsen, 2024-03-16:
That's not good... :facepalm: