openwrt/packages #23655: ERROR: package/feeds/packages/samba4 failed to build.

uosbox, 2024-03-15:
[image]

`s' 
Dynconfig[LOCKDIR]:                                                               : '/var/lock' 
Dynconfig[PIDDIR]:                                                                : '/var/run' 
Dynconfig[STATEDIR]:                                                              : '/var/lib/samba' 
Dynconfig[CACHEDIR]:                                                              : '/var/cache/samba' 
Dynconfig[LOGFILEBASE]:                                                           : '/var/log' 
Dynconfig[SOCKET_DIR]:                                                            : '/var/run/samba' 
Dynconfig[PRIVILEGED_SOCKET_DIR]:                                                 : '/var/lib/samba' 
Dynconfig[WINBINDD_SOCKET_DIR]:                                                   : '/var/run/samba/winbindd' 
Dynconfig[NMBDSOCKETDIR]:                                                         : '/var/run/samba/nmbd' 
Dynconfig[NTP_SIGND_SOCKET_DIR]:                                                  : '/var/lib/samba/ntp_signd' 
Dynconfig[NCALRPCDIR]:                                                            : '/var/run/samba/ncalrpc' 
Dynconfig[CONFIGFILE]:                                                            : '/etc/samba/smb.conf' 
Dynconfig[LMHOSTSFILE]:                                                           : '/etc/samba/lmhosts' 
Dynconfig[SMB_PASSWD_FILE]:                                                       : '/etc/samba/smbpasswd' 
Checking for getrandom                                                            : ok 
Checking for GnuTLS >= 3.4.7                                                      : yes 
Checking for gnutls_pkcs7_get_embedded_data_oid in gnutls                         : ok 
Checking for gnutls_set_default_priority_append in gnutls                         : ok 
Checking for gnutls_pbkdf2 in gnutls                                              : ok 
Checking for gnutls_aead_cipher_encryptv2 in gnutls                               : ok 
Checking for gnutls fips mode support                                             : not found 
Checking for header gnutls/gnutls.h                                               : yes 
Checking value of GNUTLS_CIPHER_AES_128_CFB8                                      : 29 
Checking value of GNUTLS_MAC_AES_CMAC_128                                         : 203 
Checking for 'zlib'                                                               : yes 
Checking for library z                                                            : yes 
Checking for inflateInit2 in z                                                    : not found 
Checking for strsignal                                                            : ok 
Checking for longjmp                                                              : ok 
Checking for siglongjmp                                                           : ok 
Checking for system popt                                                          : yes 
Checking for header popt.h                                                        : yes 
Checking for library popt                                                         : yes 
Checking for poptGetContext in popt                                               : ok 
Checking for python version >= 3.6.0                                              : 3.11.7 
python headers                                                                    : using cache 
Checking linker accepts -Wl,-no-undefined                                         : yes 
Checking linker accepts ['-undefined', 'dynamic_lookup']                          : no 
Checking for header sys/auxv.h                                                    : yes 
Checking for getauxval                                                            : ok 
Checking linker accepts -Wl,-no-undefined                                         : yes 
Checking linker accepts ['-undefined', 'dynamic_lookup']                          : no 
Checking for python version >= 3.6.0                                              : 3.11.7 
python headers                                                                    : using cache 
Checking for system pytalloc-util.cpython-311-aarch64-linux-gnu (>=2.4.0)         : not found 
ERROR: System library pytalloc-util.cpython-311-aarch64-linux-gnu of version 2.4.0 not found, and bundling disabled
make[3]: *** [Makefile:470: /root/openwrt/build_dir/target-aarch64_generic_musl/samba-4.18.8/.configured_b43d749860bdab7d5a011c3f741992f6] Error 1
make[3]: Leaving directory '/root/openwrt/feeds/packages/net/samba4'
time: package/feeds/packages/samba4/compile#15.48#6.52#20.95
    ERROR: package/feeds/packages/samba4 failed to build.
make[2]: *** [package/Makefile:129: package/feeds/packages/samba4/compile] Error 1
make[2]: Leaving directory '/root/openwrt'
make[1]: *** [package/Makefile:123: /root/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/openwrt'
make: *** [/root/openwrt/include/toplevel.mk:233: world] Error 2
`

uosbox, 2024-03-15:
[image]
[image]

takanashi-tetsuya, 2025-05-19:
there's a same error on my pc linuxmint21.3 ...

sunowsir, 2025-07-22:
there's a same error on my pc ubuntu 24.04