openwrt/packages #23662: P910nd always bedirectional

zebraonpc, 2024-03-15:
OpenWrt 22.03.5 r20134 ,p910nd whatever i do after reboot options Bidirectional mode
always 1 (always on). is it a bug ?
i have tried form Luci also /etc/configs/p910nd
 after reboot it is always biderectinal mode 1

brada4, 2024-03-15:
https://github.com/openwrt/packages/blob/d359fa04eda29638b9326c194490685c1177fd49/net/p910nd/files/p910nd.init#L40
check `ps w|grep 910`  for -b parameter being absent.
If config is not permanent most likely cause is /overlay full.

zebraonpc, 2024-03-16:
it is   **append_bool "$section" bidirectional "-b"** in my init.d in my init.d
wdym  /overlay  is full
i have 128Kb left on my storage

```
`
root@OpenWrt:~# df -h
    Filesystem                Size      Used Available Use% Mounted on
    /dev/root                 3.8M      3.8M         0    100% /rom
     tmpfs                    28.1M    384.0K     27.8M      1% /tmp
     /dev/mtdblock6            2.1M      1.9M    168.0K     92% /overlay
      overlayfs:/overlay        2.1M      1.9M    168.0K     92% /
      tmpfs                   512.0K         0    512.0K      0% /dev
.`
```

brada4, 2024-03-16:
> have 128Kb

at this point jffs2 becomes read-only.
if you reboot youll wonder your edits revert.

just reinstall and after installing some functionality use auc to move packages to squashfs and free overlay space

ivanmara555, 2024-03-28:
> OpenWrt 22.03.5 r20134 ,p910nd whatever i do after reboot options Bidirectional mode always 1 (always on). is it a bug ? i have tried form Luci also /etc/configs/p910nd after reboot it is always biderectinal mode 1

this is probably an echo of this issue: https://github.com/openwrt/packages/issues/23774