openwrt/packages #23673: mpd: ldd /usr/bin/mpd indicates /usr/lib/libFLAC.so.8 is missing while /usr/lib/libFLAC.so exists

davidandreoletti, 2024-03-15:
Maintainer: @neheb  - _Adding you since Makefile does not indicate the maintainer and you are the last / most often contributor to this package_

Environment:
- x86-generic / OpenWRT 23.05

Issue Description:

`/usr/bin/mpd` expects `/usr/lib/libFLAC.so.8` and does not find it. However `/usr/lib/libFLAC.so` exists.

```
$ ldd /usr/bin/mpd
        /lib/ld-musl-x86_64.so.1 (0x7fd3a533e000)
        libfmt.so.9 => /usr/lib/libfmt.so.9 (0x7fd3a5327000)
        libz.so.1 => /usr/lib/libz.so.1 (0x7fd3a5312000)
        libcurl.so.4 => /usr/lib/libcurl.so.4 (0x7fd3a52ac000)
        libavutil.so.57 => /usr/lib/libavutil.so.57 (0x7fd3a50d8000)
        libavformat.so.59 => /usr/lib/libavformat.so.59 (0x7fd3a4e51000)
        libavcodec.so.59 => /usr/lib/libavcodec.so.59 (0x7fd3a3cb2000)
        libmms.so.0 => /usr/lib/libmms.so.0 (0x7fd3a3ca3000)
        libyajl.so.2 => /usr/lib/libyajl.so.2 (0x7fd3a3c97000)
        libasound.so.2 => /usr/lib/libasound.so.2 (0x7fd3a3bc3000)
        libsoxr.so.0 => /usr/lib/libsoxr.so.0 (0x7fd3a3b62000)
        libid3tag.so.0 => /usr/lib/libid3tag.so.0 (0x7fd3a3b46000)
        libpulse.so.0 => /usr/lib/libpulse.so.0 (0x7fd3a3afc000)
        libshout.so.3 => /usr/lib/libshout.so.3 (0x7fd3a3ae2000)
Error loading shared library libFLAC.so.8: No such file or directory (needed by /usr/bin/mpd)
        libfaad.so.2 => /usr/lib/libfaad.so.2 (0x7fd3a3ac5000)
        libmad.so.0.16.3 => /usr/lib/libmad.so.0.16.3 (0x7fd3a3aa4000)
        libopus.so.0 => /usr/lib/libopus.so.0 (0x7fd3a3a58000)
        libogg.so.0 => /usr/lib/libogg.so.0 (0x7fd3a3a4f000)
        libvorbis.so.0 => /usr/lib/libvorbis.so.0 (0x7fd3a3a28000)
        libvorbisenc.so.2 => /usr/lib/libvorbisenc.so.2 (0x7fd3a397e000)
        libmp3lame.so.0 => /usr/lib/libmp3lame.so.0 (0x7fd3a3912000)
        libexpat.so.1 => /usr/lib/libexpat.so.1 (0x7fd3a38f2000)
        libnpupnp.so.5 => /usr/lib/libnpupnp.so.5 (0x7fd3a38a2000)
        libmpdclient.so.2 => /usr/lib/libmpdclient.so.2 (0x7fd3a3884000)
        libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x7fd3a36d8000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7fd3a36c0000)
        libc.so => /lib/ld-musl-x86_64.so.1 (0x7fd3a533e000)
        libnghttp2.so.14 => /usr/lib/libnghttp2.so.14 (0x7fd3a3698000)
        libmbedtls.so.14 => /usr/lib/libmbedtls.so.14 (0x7fd3a367c000)
        libmbedx509.so.1 => /usr/lib/libmbedx509.so.1 (0x7fd3a366b000)
        libmbedcrypto.so.7 => /usr/lib/libmbedcrypto.so.7 (0x7fd3a360e000)
        libbz2.so.1.0 => /usr/lib/libbz2.so.1.0 (0x7fd3a3600000)
        libgnutls.so.30 => /usr/lib/libgnutls.so.30 (0x7fd3a33fe000)
        libswresample.so.4 => /usr/lib/libswresample.so.4 (0x7fd3a33df000)
        libfdk-aac.so.2 => /usr/lib/libfdk-aac.so.2 (0x7fd3a3359000)
        libpulsecommon-14.2.so => /usr/lib/pulseaudio/libpulsecommon-14.2.so (0x7fd3a32e6000)
        libdbus-1.so.3 => /usr/lib/libdbus-1.so.3 (0x7fd3a329d000)
        libssl.so.3 => /usr/lib/libssl.so.3 (0x7fd3a31f8000)
        libcrypto.so.3 => /usr/lib/libcrypto.so.3 (0x7fd3a2dc1000)
        libvorbisidec.so.1 => /usr/lib/libvorbisidec.so.1 (0x7fd3a2da6000)
        libmicrohttpd.so.12 => /usr/lib/libmicrohttpd.so.12 (0x7fd3a2d83000)
        libnettle.so.8 => /usr/lib/libnettle.so.8 (0x7fd3a2d35000)
        libhogweed.so.6 => /usr/lib/libhogweed.so.6 (0x7fd3a2cee000)
        libgmp.so.10 => /usr/lib/libgmp.so.10 (0x7fd3a2c88000)
        libsndfile.so.1 => /usr/lib/libsndfile.so.1 (0x7fd3a2c27000)

```

Workaround

Symlinking `/usr/lib/libFLAC.so` to `/usr/lib/libFLAC.so.8` resolves the problem.

neheb, 2024-03-15:
Probably an old build.

davidandreoletti, 2024-03-15:
> Probably an old build.

```
opkg update
opkg upgrade mpd
opkg upgrade libflac
```
The error persists.

At first glance, it does not look like old build artefact.

How can I confirm this ?

davidandreoletti, 2024-03-16:
@neheb Compiled:
- OpenWRT 's mpd + libflac from source -> the issue persists.
- OpenWRT x86 image (from prebuilt sdk) 

The issue persists. Keeping this open.

davidandreoletti, 2024-03-17:
Turns out the `dl` cache created by the OpenWRT sdk builder was out of date.

No issue.

closed 2024-03-17 by davidandreoletti