openwrt/packages #23688: samba4: The FAT/exFAT file system shared by samba is writing abnormally under macOS when "Enable macOS compatible shares"

jjm2473, 2024-03-18:
Maintainer: @miska 
Environment:
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.6'
DISTRIB_REVISION='r20265-f85a79bcb4'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 22.03.6 r20265-f85a79bcb4'
DISTRIB_TAINTS=''
```
samba4 version (`smbd -V`): `4.18.8`
macOS version: `12.7.2`

Description:
Same as closed issue https://github.com/openwrt/packages/issues/11369

Related configuration files:

/etc/config/samba4:
```ini

config samba
        option workgroup 'WORKGROUP'
        option charset 'UTF-8'
        option description 'Samba on OpenWRT'
        option macos '1'

config sambashare
        option name 'test'
        option path '/mnt/vfat'
        option read_only 'no'
        option force_root '1'
        option guest_ok 'yes'
        option inherit_owner 'yes'
        option create_mask '0666'
        option dir_mask '0777'

```

/etc/samba/smb.conf:
```ini
[global]
        netbios name = OpenWrt 
        interfaces = br-lan 
        server string = Samba on OpenWRT
        unix charset = UTF-8
        workgroup = WORKGROUP

        ## This global parameter allows the Samba admin to limit what interfaces on a machine will serve SMB requests.
        bind interfaces only = yes

        ## time for inactive connections to-be closed in minutes
        deadtime = 15

        ## disable core dumps
        enable core files = no

        ## set security (auto, user, domain, ads)
        security = user

        ## This parameter controls whether a remote client is allowed or required to use SMB encryption.
        ## It has different effects depending on whether the connection uses SMB1 or SMB2 and newer:
    ## If the connection uses SMB1, then this option controls the use of a Samba-specific extension to the SMB protocol introduced in Samba 3.2 that makes use of the Unix extensions.
        ## If the connection uses SMB2 or newer, then this option controls the use of the SMB-level encryption that is supported in SMB version 3.0 and above and available in Windows 8 and newer. 
        ## (default/auto,desired,required,off)
        #smb encrypt = default

        ## set invalid users
        invalid users = root
[... 119 lines cut ...]
```

macOS did not report an error when copying a folder, and the files in the folder were written successfully:
[image]

But at this time smbd still generated an error log:
```
Mon Mar 18 18:28:00 2024 daemon.err smbd[8495]: [2024/03/18 18:28:00.321461,  0] ../../source3/modules/vfs_fruit.c:2693(fruit_pwrite_meta_stream)
Mon Mar 18 18:28:00 2024 daemon.err smbd[8495]:   fruit_pwrite_meta_stream: On-demand create [mtk/MT7531_Reference_Manual_for_Development_Board.pdf:AFP_AfpInfo] in write failed: No such file or directory
Mon Mar 18 18:28:00 2024 daemon.err smbd[8495]: [2024/03/18 18:28:00.389464,  0] ../../source3/modules/vfs_fruit.c:2693(fruit_pwrite_meta_stream)
Mon Mar 18 18:28:00 2024 daemon.err smbd[8495]:   fruit_pwrite_meta_stream: On-demand create [mtk/mtk_wi-fi_softap_software_programming_guide_v4.6.pdf:AFP_AfpInfo] in write failed: No such file or directory
Mon Mar 18 18:28:00 2024 daemon.err smbd[8495]: [2024/03/18 18:28:00.393867,  0] ../../source3/modules/vfs_fruit.c:2693(fruit_pwrite_meta_stream)
Mon Mar 18 18:28:00 2024 daemon.err smbd[8495]:   fruit_pwrite_meta_stream: On-demand create [mtk:AFP_AfpInfo] in write failed: No such file or directory
Mon Mar 18 18:28:00 2024 daemon.err smbd[8495]: [2024/03/18 18:28:00.400523,  0] ../../source3/modules/vfs_fruit.c:2693(fruit_pwrite_meta_stream)
Mon Mar 18 18:28:00 2024 daemon.err smbd[8495]:   fruit_pwrite_meta_stream: On-demand create [mtk:AFP_AfpInfo] in write failed: No such file or directory
```

`chmod 0777 /mnt/vfat` does not change anything:
```
chmod 0777 /mnt/vfat
/etc/init.d/samba4 stop
/etc/init.d/samba4 start
```

This problem can be solved by unchecked the "Enable macOS compatible shares", but this means that macOS will create a lot of files starting with `._`.

brada4, 2024-03-18:
> streams_xattr

fat does not support that, you need ext4 or xfs to support that.

jjm2473, 2024-03-18:
@brada4 But `streams_xattr` is generated by openwrt. In addition, the developers also mentioned `streams_xattr` supports FAT https://github.com/openwrt/packages/issues/11369#issuecomment-788770703

birdmanhj, 2025-05-14:
I have same error as yours, have u resolved it?