openwrt/packages #23698: I tried to compile iperf3-ssl, but the libiperf.so.* libraries were not found.

bauw2008, 2024-03-20:
The following is part of the log

OpenWrt-libtool: install: chmod 644 /home/bauw/openwrt6.6/build_dir/target-x86_64_musl/iperf-ssl/iperf-3.16/ipkg-install/usr/lib/libiperf.a
OpenWrt-libtool: install: x86_64-openwrt-linux-musl-gcc-ranlib /home/bauw/openwrt6.6/build_dir/target-x86_64_musl/iperf-ssl/iperf-3.16/ipkg-install/usr/lib/libiperf.a
libtool: warning: remember to run 'libtool --finish /usr/lib'
 /usr/bin/mkdir -p '/home/bauw/openwrt6.6/build_dir/target-x86_64_musl/iperf-ssl/iperf-3.16/ipkg-install/usr/bin'

mkdir -p /home/bauw/openwrt6.6/bin/targets/x86/64/packages /home/bauw/openwrt6.6/build_dir/target-x86_64_musl/iperf-ssl/iperf-3.16/ipkg-x86_64/libiperf3/CONTROL /home/bauw/openwrt6.6/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /home/bauw/openwrt6.6/build_dir/target-x86_64_musl/iperf-ssl/iperf-3.16/ipkg-x86_64/libiperf3/usr/lib
cp -fpR /home/bauw/openwrt6.6/build_dir/target-x86_64_musl/iperf-ssl/iperf-3.16/ipkg-install/usr/lib/libiperf.so.* /home/bauw/openwrt6.6/build_dir/target-x86_64_musl/iperf-ssl/iperf-3.16/ipkg-x86_64/libiperf3/usr/lib
cp: cannot stat '/home/bauw/openwrt6.6/build_dir/target-x86_64_musl/iperf-ssl/iperf-3.16/ipkg-install/usr/lib/libiperf.so.*': No such file or directory
make[3]: *** [Makefile:114: /home/bauw/openwrt6.6/bin/packages/x86_64/packages/libiperf3_3.16-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/bauw/openwrt6.6/feeds/packages/net/iperf3'
time: package/feeds/packages/iperf3/ssl/compile#15.47#9.31#25.05
    ERROR: package/feeds/packages/iperf3 failed to build (build variant: ssl).

brada4, 2024-03-20:
check here:
https://github.com/openwrt/packages/blob/063b2393cbc3e5aab9d2b40b2911cab1c3967c59/net/iperf3/Makefile#L42
looks like -ssl dependency on atomic has to be swapped to iperf?

closed 2024-03-21 by bauw2008

smallprogram, 2024-08-23:
How can I solve this problem?