openwrt/packages #23700: htop: compiling with support for cpu temp does not add libsensors5 as a dep

graysky2, 2024-03-20:
Maintainer: @champtar 
Environment: flogic/glinet_gl-mt6000, latest snapshot

Description:  Building `htop` with `Compile Htop with lm-sensors support` does not add a dependency of `libsensors5` which is needed to get the functionality.

brada4, 2024-03-20:
Existing behaviour is documented upstream.

BKPepe, 2024-03-21:
@brada4 Maybe, it would be great if you add reference to it, so we dont need to look at it and waste our time to find that information on the Internet.

graysky2, 2024-03-21:
I think I figured out the syntax to add a depends with a conditional.  See https://github.com/openwrt/packages/pull/23704

Closing this.

brada4, 2024-03-21:
Reference : sensors support opens sensors lib if that is installed on target. Additional dependency is excessive.
https://github.com/htop-dev/htop?tab=readme-ov-file#linux

PR #23704 (2024-03-31): htop: build with CONFIG_HTOP_LMSENSORS=y
Default CONFIG_HTOP_LMSENSORS to y so users just need to install libsensors, no need to build your own, but no need to pay the size price when you don't use it.  Closes #23700.

Maintainer: @champtar
commit 4948049d (2024-03-31): htop: build with CONFIG_HTOP_LMSENSORS=y

Default CONFIG_HTOP_LMSENSORS to y so users just need to install
libsensors, no need to build your own, but no need to pay the
size price when you don't use it.  Closes #23700.
closed 2024-03-31 by champtar